pbatard/libwdi #42: 2.1.1 Hangs when listing hubs/composite parents

oysteinkrog, 2015-01-14:
Using 2.1.1 the application hangs/deadlocks every time I:
1. Check "Options - List all devices"
2. Uncheck "Options - ignore hubs and composite parents"

Let me know what information you need from me :)

pbatard, 2015-01-14:
This looks like a duplicate of #41.
This is not an issue I can replicate these issues, so I will need your help to investigate:
1. What version of Windows are you using?
2. Can you try to follow [this guide](https://github.com/pbatard/libwdi/wiki/Compiling-and-debugging-libwdi-or-Zadig) and, if there is an application crash, let me know where in the code it occurs. Or if it's a lock, you can try to pause the program execution after Zadig seems to be locked in the debugger (using what looks like the pause button on the Visual Studio frontend), and then the call stack should give a better idea where to look.

oysteinkrog, 2015-01-14:
Hmm ok I will take a look at that tomorrow.
I did quickly fire up windbg to see what the threads were doing:
```
0:001> ~
   0  Id: 1db0.2a5c Suspend: 1 Teb: 00000000`7ffdb000 Unfrozen
.  1  Id: 1db0.1480 Suspend: 1 Teb: 00000000`7ffd8000 Unfrozen
0:001> ~0 kb
RetAddr           : Args to Child                                                           : Call Site
00000000`77ce1fff : 00000023`77d7cb3c 00000000`00000023 00000000`00000000 00000000`00000000 : wow64cpu!CpupSyscallStub+0x2
00000000`77d0219a : 00000000`00000000 00000000`77ce1574 00000000`00000000 00000000`77d02380 : wow64cpu!ReadWriteFileFault+0x31
00000000`77d020d2 : 00000000`00000000 00000000`00000000 00000000`0008fd30 00000000`0008f830 : wow64!RunCpuSimulation+0xa
00007fff`6f263e39 : 0066f000`547ba700 00000000`77d01f60 00000000`7ffdf000 00000000`00000000 : wow64!Wow64LdrpInitialize+0x172
00007fff`6f2434e8 : 00007fff`6f1a0000 00000000`00000000 00000000`7ffdf000 00000000`00000000 : ntdll!LdrpInitializeProcess+0x1591
00007fff`6f1ba65e : 00000000`0008f830 00000000`00000000 00000000`7ffdf000 00000000`00000000 : ntdll!_LdrpInitialize+0x88e38
00000000`00000000 : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : ntdll!LdrInitializeThunk+0xe
0:001> ~1 kb
RetAddr           : Args to Child                                                           : Call Site
00007fff`6f260544 : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : ntdll!DbgBreakPoint
00007fff`6f2103d1 : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : ntdll!DbgUiRemoteBreakin+0x34
00000000`00000000 : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : ntdll!RtlUserThreadStart+0x41

```

oysteinkrog, 2015-01-14:
I forgot to mention that it does not hang with 2.1.0

ms-jdow, 2015-02-07:
For reference this has happened for me just recently when I installed a USB 3.0 PCIe card.from SUNIX model 2300 with a NEC chip.

{^_^}

mfleisz, 2015-04-27:
I have got the same issue - I tracked it down to this line in logging.c (109):

```
WriteFile(logger_wr_handle, buffer, (DWORD)(size1+size2+1), &junk, NULL);
```

After reading the comment above I changed LOGGER_PIPE_SIZE to (2 * 8192) and the hang was gone. Seems that logging exceeds the pipe's buffer size and causes WriteFile to deadlock the process.

pbatard, 2015-04-27:
@mfleisz,

Thanks for looking into this! I was hoping that someone who experiences the issue would take the time to try to investigate it, as I haven't been able to replicate it on any of the machines I tried with.

The interesting thing of course, is that right before the line you mention, I had commented:
```
	// http://msdn.microsoft.com/en-us/library/aa365150%28VS.85%29.aspx:
	// "if your specified buffer size is too small, the system will grow the
	//  buffer as needed, but the downside is that the operation will block
	//  until the (existing) data is read from the pipe."
	// Existing pipe data should have produced a notification, but if the pipe
	// is left to fill without readout, we might run into blocking log calls.
	// TODO: address this potential issue if it is reported
```
So I guess some combination of hardware can create log messages that exceed the default buffer size, and while I had vaguely anticipated that this might be an issue, I thought it would manifest itself in a much more obvious way.

I expect that I should be able to reproduce the issue on my end, by using a very small buffer size, which should help with making sure that this gets fixed once and for all, so I'll try that when I get a chance. I'll probably release a new version of Zadig once that's done, as I know this problem has been plaguing way too many people.

Can't thank you enough for doing the hard work and pinpointing the probable cause! :smiley:

mfleisz, 2015-04-28:
@pbatard You'r welcome - glad I could help out with this one! Looking forward to a new version fixing this issue :) :+1:

pbatard, 2015-04-30:
I was able to reproduce the issue when reducing `LOGGER_PIPE_SIZE` and I have now applied a fix. I have also uploaded a new version of Zadig at [the usual location](http://zadig.akeo.ie/downloads/).

I'm going to close the issue (and wait a little bit before announcing the new release). If you or anyone else sees an issue with this new release, please let me know.

commit 55a96538 (2015-04-30): [core] fix application freezout when log messages exceed the pipe size

* This was reported by many people and the actual issue was pinpointed by Martin Fleisz
* Closes #42
closed 2015-04-30 by pbatard