pbatard/libwdi #43: Driver Installation: FAILED (Resource already exists)

MarkKharitonov, 2015-01-30:
The log contains the following messages:

    libwdi:debug [process_message] switching timeout to infinite
    libwdi:debug [installer process] Installing driver for USB\VID_04E8&PID_685D&REV_7<7<&Modem - please wait...
    libwdi:debug [process_message] switching timeout back to finite
    libwdi:debug [installer process] more recent driver was found (force option required)
    libwdi:debug [process_message] installer process completed
    Driver Installation: FAILED (Resource already exists)

Inspecting the source code reveals that on the surface this error should not occur, because the code does pass the `INSTALLFLAG_FORCE` to the `UpdateDriverForPlugAndPlayDevices` function. However, according to http://stackoverflow.com/questions/11474317/updatedriverforplugandplaydevices-error-is-telling-me-im-not-doing-something this is not enough - could it be that the generated INF file is incomplete?

pbatard, 2015-02-01:
Have you tried uninstalling the existing driver?
In the vast majority of cases, I don't think I want Zadig to replace a more recent driver with an older one, if that's the issue.
If a user really wants to do that, then they should first uninstall the existing driver and try again.

MarkKharitonov, 2015-02-01:
I am trying to root my Galaxy Tab 2 10.1 following the guide here - http://wiki.cyanogenmod.org/w/Install_CM_for_p5110

The original Samsung driver is only visible when the tablet is connected by USB and the instructions clearly say to replace the driver while the tablet is connected.

Next, I have noticed that the zadig source code does attempt to enforce the replacement of the current driver, even if it is newer than the replacement. From which I conclude that this should work, even if discouraged.

Or am I missing something?

pbatard, 2015-02-02:
> The original Samsung driver is only visible when the tablet is connected by USB and the instructions clearly say to replace the driver while the tablet is connected.

How does that prevent you from trying to uninstall the driver, as I recommended? If you are in a position to try to replace a driver then you also are in a position to uninstall it. Again, have you tried uninstalling the driver first.

> From which I conclude that this should work, even if discouraged.

I seem to recall that I added the flag to solve another issue (Microsoft APIs are tricky, and clearly the flag does not work in the manner you interpret it, as it didn't replace your driver) rather than force a replacement of the driver always, but I might be wrong.

Again, I have to ask, have you tried uninstalling the existing driver first?

pbatard, 2015-05-30:
No update from OP &rarr; Closing.

closed 2015-05-30 by pbatard