pbatard/libwdi #45: ibwdi Zadig: The driver installation fail.

pradhr, 2015-04-13:
Hi, I am new to libwdi. 
I have posted the same issue for inf-wizard to make it simple but the same issue with Zadig.
I have just compiled lidwdi example on Win 7 with Visual Studio Community Edition 2013 and use libusb0 driver. I can compile the project successfully and created Zadig.exe file.
When I run the Zadig.exe file then at the end it says "The driver installation fail".

If I use the original Zadig.exe then it installs driver successfully.
But when install newly compiled Zadig.exe then it gives this error.

Please guide me, what could be the problem?

I have modified the config.h file for only using libusb0:

#ifndef DDK_DIR
//#define DDK_DIR "C:/Program Files (x86)/Windows Kits/8.1"
#endif

/* 64bit subdirectory for WinUSB redist files ("x64" or "amd64") */
#define X64_DIR "x64"

/* embed libusb0 driver files from the following location */
#ifndef LIBUSB0_DIR
#define LIBUSB0_DIR "D:/temp/libusb-win32"
#endif

/* embed libusbK driver files from the following location */
#ifndef LIBUSBK_DIR
//#define LIBUSBK_DIR "D:/libusbK/bin"
#endif

/* embed user defined driver files from the following location */
#ifndef USER_DIR
// #define USER_DIR "C:/signed-driver"
#endif

pradhr, 2015-04-13:
Here is the logfile generated in Zadig:
Zadig 2.1.1.675
Windows 7 SP1 32 bit
ini file 'zadig.ini' not found - default parameters will be used
'WinUSB' driver is not available
falling back to 'libusb-win32' for default driver
1 device found.
Using inf name: PDT4000_Ver_1.0.inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:debug [installer process] got parameter PDT4000_Ver_1.0.inf
libwdi:debug [installer process] got device_id: 'USB\VID_045B&PID_0512\P001'
libwdi:debug [installer process] got hardware_id: 'USB\VID_045B&PID_0512&REV_0200'
libwdi:debug [installer process] got user_sid: 'S-1-5-21-812045857-3654368192-2273473021-1000'
libwdi:debug [installer process] using syslog 'C:\Windows\inf\setupapi.dev.log'
libwdi:debug [installer process] sylog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [installer process] Installing driver for USB\VID_045B&PID_0512&REV_0200 - please wait...
libwdi:debug [syslog] <Garbled data>
libwdi:debug [syslog] >>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - USB\VID_045B&PID_0512&REV_0200]
libwdi:debug [syslog] >>>  Section start 2015/04/13 15:21:39.946
libwdi:debug [syslog]       cmd: "C:\Users\prashant\usb_driver\installer_x86.exe" "PDT4000_Ver_1.0.inf"
libwdi:debug [syslog]      dvi: Set selected driver complete.
libwdi:debug [syslog]      dvi: {Build Driver List} 15:21:39.973
libwdi:debug [syslog]      dvi:      Searching for hardware ID(s):
libwdi:debug [syslog]      dvi:           usb\vid_045b&pid_0512&rev_0200
libwdi:debug [syslog]      dvi:           usb\vid_045b&pid_0512
libwdi:debug [syslog]      dvi:      Searching for compatible ID(s):
libwdi:debug [syslog]      dvi:           usb\class_ff&subclass_ff&prot_ff
libwdi:debug [syslog]      dvi:           usb\class_ff&subclass_ff
libwdi:debug [syslog]      dvi:           usb\class_ff
libwdi:debug [syslog]      cpy:      Policy is set to make all digital signatures equal.
libwdi:debug [syslog]      dvi:      Processing a single INF: 'c:\users\prashant\usb_driver\pdt4000_ver_1.0.inf'
libwdi:debug [syslog]      inf:      Opened INF: 'c:\users\prashant\usb_driver\pdt4000_ver_1.0.inf' ([strings])
libwdi:debug [syslog]      sig:      {_VERIFY_FILE_SIGNATURE} 15:21:39.991
libwdi:debug [syslog]      sig:           Key      = pdt4000_ver_1.0.inf
libwdi:debug [syslog]      sig:           FilePath = c:\users\prashant\usb_driver\pdt4000_ver_1.0.inf
libwdi:debug [syslog]      sig:           Catalog  = c:\users\prashant\usb_driver\PDT4000_Ver_1.0.cat
libwdi:debug [syslog] !    sig:           Verifying file against specific (valid) catalog failed! (0x800b0109)
libwdi:debug [syslog] !    sig:           Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.
libwdi:debug [syslog]      sig:      {_VERIFY_FILE_SIGNATURE exit(0x800b0109)} 15:21:40.082
libwdi:debug [syslog]      sig:      {_VERIFY_FILE_SIGNATURE} 15:21:40.082
libwdi:debug [syslog]      sig:           Key      = pdt4000_ver_1.0.inf
libwdi:debug [syslog]      sig:           FilePath = c:\users\prashant\usb_driver\pdt4000_ver_1.0.inf
libwdi:debug [syslog]      sig:           Catalog  = c:\users\prashant\usb_driver\PDT4000_Ver_1.0.cat
libwdi:debug [syslog]      sig:           Success: File is signed in Authenticode(tm) catalog.
libwdi:debug [syslog]      sig:           Error 0xe0000241: The INF was signed with an Authenticode(tm) catalog from a trusted publisher.
libwdi:debug [syslog]      sig:      {_VERIFY_FILE_SIGNATURE exit(0xe0000241)} 15:21:40.248
libwdi:debug [syslog]      dvi:      Created Driver Node:
libwdi:debug [syslog]      dvi:           HardwareID   - USB\VID_045B&PID_0512
libwdi:debug [syslog]      dvi:           InfName      - c:\users\prashant\usb_driver\pdt4000_ver_1.0.inf
libwdi:debug [syslog]      dvi:           DevDesc      - PDT4000 Ver 1.0
libwdi:debug [syslog]      dvi:           DrvDesc      - PDT4000 Ver 1.0
libwdi:debug [syslog]      dvi:           Provider     - libusb-win32
libwdi:debug [syslog]      dvi:           Mfg          - Hitachi, Ltd
libwdi:debug [syslog]      dvi:           ModelsSec    - Devices.NT
libwdi:debug [syslog]      dvi:           InstallSec   - LIBUSB_WIN32_DEV.NT
libwdi:debug [syslog]      dvi:           ActualSec    - LIBUSB_WIN32_DEV.NT
libwdi:debug [syslog]      dvi:           Rank         - 0x00ff0001
libwdi:debug [syslog]      dvi:           Signer       - USB\VID_045B&PID_0512 (libwdi autogenerated)
libwdi:debug [syslog]      dvi:           Signer Score - Authenticode
libwdi:debug [syslog]      dvi:           DrvDate      - 02/11/2014
libwdi:debug [syslog]      dvi:           Version      - 1.2.6.0
libwdi:debug [syslog]      dvi: {Build Driver List - exit(0x00000000)} 15:21:40.266
libwdi:debug [syslog]      dvi: {DIF_SELECTBESTCOMPATDRV} 15:21:40.267
libwdi:debug [syslog]      dvi:      No class installer for 'PDT4000 Ver 1.0  '
libwdi:debug [syslog]      dvi:      No CoInstallers found
libwdi:debug [syslog]      dvi:      Default installer: Enter 15:21:40.271
libwdi:debug [syslog]      dvi:           {Select Best Driver}
libwdi:debug [syslog]      dvi:                Selected driver installs from section [LIBUSB_WIN32_DEV.NT] in 'c:\users\prashant\usb_driver\pdt4000_ver_1.0.inf'.
[... 354 lines cut ...]
libwdi:debug [syslog]      ump: {Plug and Play Service: Device Install exit(00000002)}
[... cut ...]
libwdi:debug [syslog]      dvi:                {Restarting Devices} 15:21:51.834
libwdi:debug [syslog]      dvi:                     Restart: USB\VID_045B&PID_0512\P001
libwdi:debug [syslog]      dvi:                     Restart complete.
libwdi:debug [installer process] the system can not find the file specified (C:\Users\prashant\usb_driver\PDT4000_Ver_1.0.inf)
Driver Installation: FAILED (Requested resource not found)
1 device found.
1 device found.
Using 'C:\Users\prashant\usb_driver' as extraction directory.

pradhr, 2015-04-13:
Newly created wdi-simple.exe is also behaving the same way.
Actually I need wdi-simple.exe which will be used with NSIS script to install the driver.
If some can please provide me link to download wdi-simple.exe and if that works, then it will solve my problem.
As ready made Zadig.exe and inf-wizard.exe both successfully install the driver, I think wdi-simple.exe will also work.

Thanks.

pradhr, 2015-04-14:
I have done little bit more progress.
Currently when we use wdi-simple.exe which is compiled by the Visual Studio Community Edition 2013 on my computer and use this exe to create the inf etc driver files
and then install on computer to which PDT4000 USB Device is connected, it is giving error that "Resource not found" or in Zadig "The driver installation fail".

But if we use the inf etc driver files created by original inf-wizard.exe or Zadig.exe and then manually select the driver, it will install the driver successfully.

I don't mind using the inf etc driver files created by original Zadig.exe, but I found that wdi-simple.exe always create inf etc files, it does not just pick from specified folder. If it is possible for wdi-simple.exe to just use the driver from the specified folder, please do let me know.

Thank you for all your help.

pbatard, 2015-04-17:
1. What does the embedder step of the Visual Studio compilation say? With libusb0.sys alone, you should see something like this during compilation:
```
>Project not selected to build for this solution configuration 
4>  detected change for 'D:\libwdi\Win32\Debug\helper\installer_x86.exe'
4>    EMBED  D:\libusb-win32\bin\x86\libusb0_x86.dll (2012.08.02)
4>    EMBED  D:\libusb-win32\bin\x86\install-filter.exe (2012.08.02)
4>    EMBED  D:\libusb-win32\bin\x86\libusb0.sys (2012.08.02)
4>    EMBED  D:\libusb-win32\bin\amd64\libusb0.dll (2012.08.02)
4>    EMBED  D:\libusb-win32\bin\amd64\libusb0.sys (2012.08.02)
4>    EMBED  D:\libusb-win32\bin\amd64\install-filter.exe (2012.08.02)
4>    EMBED  D:\libusb-win32\installer_license.txt (2012.08.02)
4>    EMBED  D:\libwdi\Win32\Debug\helper\installer_x86.exe (2014.12.31)
4>    EMBED  D:\libwdi\x64\Debug\helper\installer_x64.exe (2014.12.31)
4>    EMBED  D:\libwdi\libwdi\winusb.inf.in (2014.12.21)
4>    EMBED  D:\libwdi\libwdi\libusb0.inf.in (2014.12.21)
4>    EMBED  D:\libwdi\libwdi\libusbk.inf.in (2014.12.21)
4>    EMBED  D:\libwdi\libwdi\winusb.cat.in (2014.12.21)
4>    EMBED  D:\libwdi\libwdi\libusb0.cat.in (2014.12.21)
4>    EMBED  D:\libwdi\libwdi\libusbk.cat.in (2014.12.21)
```
Especially, if you don't see the `libusb0.sys` and `libusb0.dll` mentioned, for bot the `x86` and `amd64` versions. you're never going to be able to install the libusb-win32 driver. The fact that you get a `system cannot find the file specified` error during installation would tend to indicate that you are missing some files.

pbatard, 2015-05-30:
No update from OP &rarr; Closing.

closed 2015-05-30 by pbatard

eng-Robert, 2017-02-07:
I am not the OP, but this issue is happening to me too.
More details is that :  `UpdateDriverForPlugAndPlayDevicesW` returns the value `2`  `ERROR_FILE_NOT_FOUND`, while the INF file already exists in the file system.

eng-Robert, 2017-02-08:
Please re-open this issue and see a related question on [stackoverflow](http://stackoverflow.com/questions/42091483)

pbatard, 2017-02-08:
Unless your issue is that `the system can not find the file specified (C:\Users\prashant\usb_driver\PDT4000_Ver_1.0.inf)`, this is not the same issue, so __please__ create a new issue and make sure to provide either your full debug log from Zadig or the relevant section from `C:\Windows\INF\setupapi.dev.log`.

Also, since you seem to be able to debug, you may want to have a look at [this](https://github.com/pbatard/libwdi/wiki/Compiling-and-debugging-libwdi-or-Zadig) and add a breakpoint in `UpdateDriverForPlugAndPlayDevicesU()` so that you can check the values of `wFullInfPath` and `FullInfPath` at the time of the call.

eng-Robert, 2017-02-08:
I think it is the same issue. Off course the filename and windows user name are different but the cause is the same.

If you don't define `LIBUSBK_DIR`  , the generated inf file contains a wrong placeholder for its files.
`libusb0.dll, libusb0_x86.dll #LK_COMMA# #LK_DLL# #LK_COMMA# #LK_X86_DLL#`
This makes windows search for these placeholder files.

I solved it by removing the placeholders if K DLLs are not defined in `wdi_prepare_driver()`

```
if ((driver_type == WDI_LIBUSB0)){
    // If the target is libusb-win32 and we have the K DLLs, add them to the inf
    if ((wdi_is_driver_supported(WDI_LIBUSBK, NULL))) {
        wdi_info("K driver available - adding the libusbK DLLs to the libusb-win32 inf");
        static_strcpy(inf_entities[LK_COMMA].replace, ",");
        static_strcpy(inf_entities[LK_DLL].replace, "libusbk.dll");
        static_strcpy(inf_entities[LK_X86_DLL].replace, "libusbk_x86.dll");
        static_strcpy(inf_entities[LK_EQ_X86].replace, "= 1,x86");
        static_strcpy(inf_entities[LK_EQ_X64].replace, "= 1,amd64");
    }
    else {
        // issue #45 - remove placeholders.
        static_strcpy(inf_entities[LK_COMMA].replace, " ");
	static_strcpy(inf_entities[LK_DLL].replace, " ");
	static_strcpy(inf_entities[LK_X86_DLL].replace, " ");
	static_strcpy(inf_entities[LK_EQ_X86].replace, " ");
	static_strcpy(inf_entities[LK_EQ_X64].replace, " ");
    }
}
```

pbatard, 2017-02-08:
I'm still not seeing a log from you or any of the details I asked. As the people from stackoverflow say, we are not mind readers. If you don't or won't provide the data we ask for, don't expect much help. Also if you want to report an issue for the `LIBUSBK_DIR` __please__ create a new issue. It's very problematic for __other people who experience the same problem as you__ if they have to wade through an unrelated issue to find some answers, and that is why specific independent problems should be logged separately in the issue tracker.