pbatard/libwdi #47: Repacking  and installing extracted driver

diabolo38, 2015-07-02:
When driver is extracted (extract but do no install) from GUI 
[image]
The folder contain  2 x86 and x64 installer exe  but that want install anything 
when installer exe are started in command line we can see the error message 

"This application can not be run from the command line.
Please use your initial installer application if you want to install the driver.this one want install the driver "
how could the full installer  be repacked  and driver installed from as a single exe ? 
What is the installer applacition ? dpinst ?
Actualy i have customized the generated inf file to install for multiple interface of composite 
Having that capabilty in teh GUUI woudl be nice  also option to store/save the installer exe for future use would be nice.

pbatard, 2015-07-04:
The full installer application is Zadig. As the message indicates, the installer executables are not meant to be used from the command line. If you select _Extract Files_, which is an advanced feature, then I assume that you know how to install drivers by pointing to the `.inf` file (using the Device Manager or something else). If not, then you should not use _Extract Files_.

> how could the full installer be repacked and driver installed from as a single exe ? 

That single exe __is__ Zadig. If you find it limited for your usage, then you will have to read the [documentation from libwdi](https://github.com/pbatard/libwdi/wiki#Documentation), which is the library used by Zadig for the driver installing, and learn to either recompile your own libwdi-based application or use one of the other samples, such as [`wdi-simple`](https://github.com/pbatard/libwdi/blob/master/examples/wdi-simple.c) (which you will need to recompile yourself).

diabolo38, 2015-07-06:
Thanks for the inputs 
likely simple installer would do the works 
Unfortunately i was not able to quickly get the liwdi to build with libusbk  
yet while trying i  found that libusbk inf wizard was doing exactly what i need so will be using that.

pbatard, 2015-07-06:
OK, I'll close this issue then.

closed 2015-07-06 by pbatard