pbatard/libwdi #50: Windows 10

cpldcpu, 2015-08-03:
Not a request - rather a question:

It seems that windows 10 will require all USB drivers to be signed starting in about two months:

http://blogs.msdn.com/b/windows_hardware_certification/archive/2015/04/01/driver-signing-changes-in-windows-10.aspx

Hows does this affect Zadig, and will it continue to work?

pbatard, 2015-08-04:
libwdi/Zadig simply __repackages__ drivers. So the makers of these drivers (Microsoft for WinUSB, or the libusb-win32 people for the other ones) comply with the Windows 10 signing requirements, then it's a non issue - once these compatible drivers are published, I'll just rebuild Zadig by pointing to those drivers.

Also, in the case of WinUSB, it's likely that it's a non-issue, since Windows 10 __already__ comes with WinUSB, and Zadig (or rather the Windows driver installation subsystem) will not replace that driver.

This being said, because my time is very limited, and libwdi/Zadig are one of my lowest priority right now, I haven't tested, and it's unlikely I will do so for some time (months?... or more?).

So if you have a question that can easily be answered by testing, I would strongly encourage you to do so and report your findings and any issue you may find, rather than expect somebody else to do that work for you... ;)

cpldcpu, 2015-08-05:
Thank you for responding, but I am a bit confused by the answer. The readme of libwdi states:

> - Automated catalog file creation and signing, using autogenerated certificate

So my understanding so far was, that libwdi actually creates the signatures on the fly by generating a cat file from the driver dll and inf files? 

My understanding from the msdn link above is, that basically the signing files would have to be registered with microsoft starting in about 60 days. That would render on-the-fly signing invalid? Or am I missing something?

diabolo38, 2015-08-05:
Driver cat is not resign for sure
Maybe the installer is for some sort of editor and avoid annoying warning when install "windows cannot check. ...install anyway" 
Do you known about libusbk drivers availability or plan  for win 10th?

cpldcpu, 2015-08-05:
Zadig and libusb work fine under windows 10 right now, i tested it. The concern was about what happens after 90 days have passed. 

However, maybe this is not an issue for non-kernelmode drivers.

pbatard, 2015-08-05:
> So my understanding so far was, that libwdi actually creates the signatures on the fly by generating a cat file from the driver dll and inf files? 

libwdi does __NOT__ sign the drivers. It only signs the driver __package__ (`.cat`). Those are completely different things, and the latter has not changed in Windows 10 AFAIK.

And yes, signing the driver __package__ is to avoid unwanted warnings during driver installation. Please see [the FAQ](https://github.com/pbatard/libwdi/wiki/FAQ#What_are_these_USBVID_PID_MI__Autogenerated_certificates_that_libwdi_installs_in_the_Trusted_certificate_stores) for more info.

For info about libusbK driver, please post on the libusb-win32 mailing list. I am not involved in libusbK development so I don't know what the plans are.

pbatard, 2015-10-21:
I will close this issue as I have answered the questions I could answer.

closed 2015-10-21 by pbatard