pbatard/libwdi #52: Zadig 2.1.2 fails to install winusb on Windows 7 with Error 0x800b0109

ichnaea-coder, 2015-12-11:
I tried to install winusb for use with RTL-SDR on a Windows 7 machine. I get the following errors. Full log is attached.

libwdi:debug [syslog]      sig:                          Catalog  = C:\windows\System32\DriverStore\Temp\{2da18a78-c925-49e7-1951-d8158f700f19}\RTL2838UHIDIR_(Composite_Parent).cat
libwdi:debug [installer process] unhandled error 0x422 (1058)
libwdi:debug [syslog] !    sig:                          Verifying file against specific (valid) catalog failed! (0x800b0109)
libwdi:debug [installer process] [1058] The service cannot be started, either because it is disabled or because it has no enabled devices associated with it.
libwdi:debug [process_message] installer process completed
libwdi:debug [syslog] !    sig:                          Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.
Driver Installation: FAILED (Other error)

[zadig.txt](https://github.com/pbatard/libwdi/files/59821/zadig.txt)

pbatard, 2015-12-12:
Nah, `0x800b0109` is a standard benign __warning__, that everybody gets, but that Microsoft somehow reports as a (non fatal) error.

On the other hand, from previous reports I had, `1058` is most likely due to the _Windows Update_ service not being started on your computer. Please make sure that your service is 'Running', as per the picture below, and try again.

[image]

ichnaea-coder, 2015-12-15:
It seems like Windows Update was the problem. Just out of professional curiosity, why does Zadig need the Windows Update process to be running?

pbatard, 2015-12-15:
It's not Zadig that needs it. It's the native Windows driver installation process.

All Zadig does, really, is tell Windows: _"Hey, I'd like you to install this driver. Here are the files you need"_, and then Windows takes over, using a process that only Microsoft engineers know the specifics of, and reports any failure it encounters. So I don't have more information than you have on why the Windows driver installation process requires the Update service to be running.

Since you have confirmed that enabling the Windows Update service fixed your problem, I will close this issue.

closed 2015-12-15 by pbatard