pbatard/libwdi #53: Distinguish libwdi driver name from previous one.

mirhl, 2016-01-09:
I read in your [FAQ](https://github.com/pbatard/libwdi/wiki/FAQ#What_are_these_USBVID_PID_MI__Autogenerated_certificates_that_libwdi_installs_in_the_Trusted_certificate_stores) that you disregard custom names because that may be used improperly and all, but I was wondering: couldn't you just include a switch to make libwdi "tag" the driver name?

I mean, I often have to switch between this and normal Windows driver, and it's always guessing considering they look identical in update driver dialog. nefarius/ScpServer#124

pbatard, 2016-01-09:
> it's always guesswork

It doesn't have to be. Just right click the device in Device Manager &rarr; Properties &rarr; Details tab and look for the _Service_ property. There you will find if it's `libusbk` (libwdi installed driver) or something else.

Also the device description you provided in libwdi (which can be changed by the user during installation) should appear somewhere in the properties, and I'm not sure what exactly you believe can be changed by libwdi, when a lot of what appears in the Device Manager is decided by Windows on its own, according to the USB device descriptors. There's no Windows API allowing you to "tag" a driver, and the best you will get as a "tag" is what I described above.

I will now close this issue.

closed 2016-01-09 by pbatard

mirhl, 2016-01-09:
Mhh, I guess I couldn't explain good enough. 
I'm talking of this window. 
[image]
And when I said "tag" I just meant something in the name so that I don't have to, as I said, guess.

pbatard, 2016-01-09:
Have you tried modifying the proposed name in Zadig/libwdi?

I am not sure how you install your libusbK driver - libwdi is only a __library__ that applications can use to install drivers. It already offers the possibility to change the driver name. If you use [Zadig](http://zadig.akeo.ie) , which is the user facing application I provide, you will have a field that you can edit to set the name, and I would expect this to be reflected in the dialog you show above. But if you don't use Zadig, then you will have to contact whoever created the libwdi based driver installer application you use, because it'll be up to them to allow you to edit the name. And, no, I will not add default driver name tagging in libwdi, because __very few__ people need to differentiate 2 drivers, and it would be out of place for libwdi to add its mark here and there as it could break applications that rely on looking up the default driver name.

Again, please try to uninstall your libusbK driver completely (make sure the driver files are deleted) and then reinstall it using [Zadig](http://zadig.akeo.ie) while making sure you edit the driver name in the Zadig interface when you install it. Then check what you get in the dialog you showed above.

mirhl, 2016-01-10:
I didn't ask for *default* driver name tagging, and I don't see how applications would look for a driver instead of the OS interface, but thanks for replying. 
I'll ask whoever in charge.