pbatard/libwdi #54: Zadig crash when trying to install or replase driver.

09th, 2016-01-14:
Win 7 x86. In windows even log I found this report
<EventData>
  <Data>zadig_2.1.2.exe</Data> 
  <Data>2.1.2.677</Data> 
  <Data>5542a807</Data> 
  <Data>CRYPT32.dll</Data> 
  <Data>6.1.7601.18277</Data> 
  <Data>52506fc6</Data> 
  <Data>c0000005</Data> 
  <Data>000107e0</Data> 
  <Data>450</Data> 
  <Data>01d14e9286a85823</Data> 
  <Data>D:\programms\SDR\zadig_2.1.2.exe</Data> 
  <Data>C:\Windows\system32\CRYPT32.dll</Data> 
  <Data>cd3a8cd0-ba85-11e5-9358-8c89a5dcb9f1</Data> 
  </EventData>

I also trying to install it on another PC with win xp, but get the similar error.

pbatard, 2016-01-18:
Is there any way you could follow [this](https://github.com/pbatard/libwdi/wiki/Compiling-and-debugging-libwdi-or-Zadig) on your Windows 7 platform, to see exactly where the crash occurs?

Also, are you sure you are using the XP version of Zadig on XP, and the non XP version of Zadig on Windows 7? be mindful that there are [2 versions of Zadig](http://zadig.akeo.ie/) that you can download, and you must use the one that is relevant to your platform.

09th, 2016-01-19:
Yes, I used appropriate different versions for XP and 7

call stack
```
 	crypt32.dll!752c07e0()	Нет данных
 	[Указанные ниже фреймы могут быть неверны и (или) отсутствовать, символы для crypt32.dll не загружены]	
 	[Внешний код]	
>	zadig.exe!DeletePrivateKey(const _CERT_CONTEXT * pCertContext)Строка 783	C
 	zadig.exe!SelfSignFile(const char * szFileName, const char * szCertSubject)Строка 919	C
 	zadig.exe!wdi_prepare_driver(wdi_device_info * device_info, const char * path, const char * inf_name, wdi_options_prepare_driver * options)Строка 1256	C
 	zadig.exe!install_driver()Строка 377	C
 	zadig.exe!main_callback(HWND__ * hDlg, unsigned int message, unsigned int wParam, long lParam)Строка 1722	C
 	[Внешний код]	
 	zadig.exe!subclass_callback(HWND__ * hDlg, unsigned int message, unsigned int wParam, long lParam)Строка 1297	C
 	[Внешний код]	
 	zadig.exe!WinMain(HINSTANCE__ * hInstance, HINSTANCE__ * hPrevInstance, char * lpCmdLine, int nShowCmd)Строка 1936	C
 	[Внешний код]	

```

pbatard, 2016-01-19:
Thanks. Can you please also provide the program ouput from when it crashes. If you are running it in Visual Studio, you should see stull like:
```
libwdi:debug [wdi_create_list] Compatible ID: USB\MS_COMP_XUSB10
libwdi:debug [wdi_create_list] Driver version: 10.0.10586.0
libwdi:debug [wdi_create_list] xusb22 USB device (18): USB\VID_045E&PID_028E\1F3251C
libwdi:debug [wdi_create_list] Device description: 'Controller'
```
In the Ouput window of Visual Studio. Or, you can also try to enable Advanced Mode in Zadig (See the 'Options' menu) to see the same messages displayed in the interface (but if the application crashes, you won't be able to copy/paste these messages).

I suspect there are warning messages before the crash, and I would like to see what they are if there are any...

09th, 2016-01-20:
```
Zadig 2.1.2.682
Windows 7 SP1 32 bit
ini file 'zadig.ini' not found - default parameters will be used
default driver set to 'WinUSB'
"zadig.exe" (Win32). Загружено "C:\Windows\System32\wintrust.dll". Невозможно найти или открыть PDB-файл.
"zadig.exe" (Win32). Загружено "C:\Windows\System32\crypt32.dll". Невозможно найти или открыть PDB-файл.
"zadig.exe" (Win32). Загружено "C:\Windows\System32\msasn1.dll". Невозможно найти или открыть PDB-файл.
0 devices found.
4 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_03F0&PID_0314&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_07&SubClass_01&Prot_02
libwdi:debug [wdi_create_list] Driver version: 6.1.7600.16385
libwdi:debug [wdi_create_list] usbprint USB device (2): USB\VID_03F0&PID_0314\SG4A728024
libwdi:debug [wdi_create_list] Device description: 'hp designjet 130'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_046D&PID_C05B&REV_5400
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_02
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.18199
libwdi:debug [wdi_create_list] HidUsb USB device (3): USB\VID_046D&PID_C05B\6&230D7757&0&2
libwdi:debug [wdi_create_list] Device description: 'USB Optical Mouse'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0BDA&PID_2838&REV_0100&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Upper filter: RTL2832UBDA
libwdi:debug [wdi_create_list] Driver version: 86.1.521.2012
libwdi:debug [wdi_create_list] RTL2832UUSB USB device (5): USB\VID_0BDA&PID_2838&MI_00\7&38A180A5&0&0000
libwdi:debug [wdi_create_list] Device description: 'Bulk-In, Interface (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0BDA&PID_2838&REV_0100&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Driver version: 8664.1.119.2012
libwdi:debug [wdi_create_list] RTL2832U_IRHID USB device (6): USB\VID_0BDA&PID_2838&MI_01\7&38A180A5&0&0001
libwdi:debug [wdi_create_list] Device description: 'Bulk-In, Interface (Interface 1)'
Поток 0x12e8 завершился с кодом 0 (0x0).
Using inf name: Bulk-In_Interface_(Interface_0).inf
"zadig.exe" (Win32). Загружено "C:\Windows\System32\cryptsp.dll". Невозможно найти или открыть PDB-файл.
"zadig.exe" (Win32). Загружено "C:\Windows\System32\rsaenh.dll". Невозможно найти или открыть PDB-файл.
"zadig.exe" (Win32). Загружено "C:\Windows\System32\bcrypt.dll". Невозможно найти или открыть PDB-файл.
"zadig.exe" (Win32). Загружено "C:\Windows\System32\bcryptprimitives.dll". Невозможно найти или открыть PDB-файл.
"zadig.exe" (Win32). Загружено "C:\Windows\System32\mssign32.dll". Невозможно найти или открыть PDB-файл.
"zadig.exe" (Win32). Загружено "C:\Windows\System32\ncrypt.dll". Невозможно найти или открыть PDB-файл.
Вызвано исключение по адресу 0x752C07E0 (crypt32.dll) в zadig.exe: 0xC0000005: нарушение прав доступа при чтении по адресу 0x00000010.
```

pbatard, 2016-01-20:
Thank you very much!

Can you please try the following test version of Zadig 2.2 [here](http://zadig.akeo.ie/test/zadig_2.2.exe) and let me know if it works better? At the very least, I'm hoping that I have fixed the crash now.

Note that this is the Vista or later version of Zadig so it will not work on XP.

Thanks.

09th, 2016-01-20:
Yes, this version works! Thank you!

closed 2016-01-20 by 09th

pbatard, 2016-01-20:
I'm very pleased to hear that! Thank you very much for the testing and debug output - it really helped! :smile:

commit 0b2ae4ad (2016-01-20): [pki] fix a possible crash when deleting the private key

* Closes #54