pbatard/libwdi #55: winusb will not install using ZADIG

gusepalfa, 2016-04-08:
Trying to install driver for microchip Real ICE device.. Appears to be a winusb driver.
On other windows 7 64bit PCs specific driver copied and will install easily.
On my particular machine win 7 64 bit will not.
I have used this Real ICE device on my PC with older MPLAB ide software with another driver OK
New MPLAB X requires a newer driver    Not OK
[zadig2.txt](https://github.com/pbatard/libwdi/files/210016/zadig2.txt)
There is a registry block?????
Yes, have been using administrator rights

Log might unlock mystery

pbatard, 2016-04-08:
Most likely this is because your Windows Update service is not started:
```
  Error 1058: The service cannot be started, either because it is disabled or because it has no enabled devices associated with it.
```

Please see [this comment](https://github.com/pbatard/libwdi/issues/52#issuecomment-164085818) in issue #52.

gusepalfa, 2016-04-09:
Hello Pete

 

Perfect

 

Thank you very much

You know your stuff

 

Thank You,

Joseph Beninca

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

pbatard, 2016-04-09:
Glad I could help. I will close this issue then.

closed 2016-04-09 by pbatard