pbatard/libwdi #56: why always being winusb(v0.0.0.0)?

Haze97, 2016-05-11:
Why does my zadig can't show the version number of all drivers?
here is the log:
Zadig 2.1.2.677
Windows 10 64-bit
ini file 'zadig.ini' not found - default parameters will be used
default driver set to 'WinUSB'
0 devices found.
libwdi:warning [get_version_info] unable to allocate buffer for version info

pbatard, 2016-05-11:
The issue is this:
```
libwdi:warning [get_version_info] unable to allocate buffer for version info
```
Which means there is an issue on your machine when using the default Windows APIs to get the version data.

I'm running Windows 10 64 bit too, but I am unable to replicate your problem, which will make it exceedingly difficult to troubleshoot. Are you running a Windows 10 preview build by any chance?

Haze97, 2016-05-21:
Thank you, I changed my laptop to a Desktop PC and the problem was solved.
PS:I'm not running a Windows 10 preview build either on my laptop or Desktop PC

pbatard, 2016-05-21:
Okay. I'm going to close this issue then.

closed 2016-05-21 by pbatard