pbatard/libwdi #57: Windows 10 WUP-028 Driver Installation: FAILED (Could not allocate resource)

Sorzan, 2016-05-13:
Title pretty much sums it all up but heres the log.

Zadig 2.2.689
Windows 7 SP1 64-bit
ini file 'zadig.ini' not found - default parameters will be used
default driver set to 'WinUSB'
0 devices found.
13 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_046D&PID_082D&REV_0011&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_0e&SubClass_03&Prot_00
libwdi:debug [wdi_create_list] Driver version: 13.40.840.0
libwdi:debug [wdi_create_list] LVUVC64 USB device (3): USB\VID_046D&PID_082D&MI_00\7&890C442&0&0000
libwdi:debug [wdi_create_list] Device description: 'HD Pro Webcam C920 (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_046D&PID_082D&REV_0011&MI_02
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_01&SubClass_02&Prot_00
libwdi:debug [wdi_create_list] Upper filter: lvrs64
libwdi:debug [wdi_create_list] Driver version: 13.40.840.0
libwdi:debug [wdi_create_list] usbaudio USB device (4): USB\VID_046D&PID_082D&MI_02\7&890C442&0&0002
libwdi:debug [wdi_create_list] Device description: 'HD Pro Webcam C920 (Interface 2)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_057E&PID_0337&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.18199
libwdi:debug [wdi_create_list] HidUsb USB device (5): USB\VID_057E&PID_0337\15/07/2014
libwdi:debug [wdi_create_list] Device description: 'WUP-028'
libwdi:debug [wdi_create_list] Hardware ID: USB\Vid_1038&Pid_1000&Rev_0100&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.18199
libwdi:debug [wdi_create_list] HidUsb USB device (8): USB\VID_1038&PID_1000&REV_0100&MI_00\2&2B7F9E1&0&0000
libwdi:debug [wdi_create_list] Device description: 'Steelseries Keyboard (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\Vid_1038&Pid_1000&Rev_0100&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.18199
libwdi:debug [wdi_create_list] HidUsb USB device (9): USB\VID_1038&PID_1000&REV_0100&MI_01\2&2B7F9E1&0&0001
libwdi:debug [wdi_create_list] Device description: 'Steelseries Consumer (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\Vid_1038&Pid_1000&Rev_0100&MI_02
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_02
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.18199
libwdi:debug [wdi_create_list] HidUsb USB device (10): USB\VID_1038&PID_1000&REV_0100&MI_02\2&2B7F9E1&0&0002
libwdi:debug [wdi_create_list] Device description: 'Steelseries Mouse (Interface 2)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_1038&PID_1202&REV_0038&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 2.4.2.5
libwdi:debug [wdi_create_list] HidUsb USB device (12): USB\VID_1038&PID_1202&MI_00\8&2E94EF&0&0000
libwdi:debug [wdi_create_list] Device description: 'SteelSeries Apex Gaming Keyboard (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_1038&PID_1202&REV_0038&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 2.3.2.0
libwdi:debug [wdi_create_list] hidkmdf USB device (13): USB\VID_1038&PID_1202&MI_01\8&2E94EF&0&0001
libwdi:debug [wdi_create_list] Device description: 'SteelSeries Apex Gaming Keyboard (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_1038&PID_1202&REV_0038&MI_02
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 2.4.2.5
libwdi:debug [wdi_create_list] HidUsb USB device (14): USB\VID_1038&PID_1202&MI_02\8&2E94EF&0&0002
libwdi:debug [wdi_create_list] Device description: 'SteelSeries Apex Gaming Keyboard (Interface 2)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_1B1C&PID_1B22&REV_0107&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_02
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.18199
libwdi:debug [wdi_create_list] HidUsb USB device (16): USB\VID_1B1C&PID_1B22&MI_00\8&36795CC6&0&0000
libwdi:debug [wdi_create_list] Device description: 'Corsair Gaming KATAR Mouse (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_1B1C&PID_1B22&REV_0107&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.18199
libwdi:debug [wdi_create_list] HidUsb USB device (17): USB\VID_1B1C&PID_1B22&MI_01\8&36795CC6&0&0001
libwdi:debug [wdi_create_list] Device description: 'Corsair Gaming KATAR Mouse (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_1B1C&PID_1B22&REV_0107&MI_02
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.18199
libwdi:debug [wdi_create_list] HidUsb USB device (18): USB\VID_1B1C&PID_1B22&MI_02\8&36795CC6&0&0002
libwdi:debug [wdi_create_list] Device description: 'Corsair Gaming KATAR Mouse (Interface 2)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_1B1C&PID_1B22&REV_0107&MI_03
[... 96 lines cut ...]
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.18199
libwdi:debug [wdi_create_list] HidUsb USB device (18): USB\VID_1B1C&PID_1B22&MI_02\8&36795CC6&0&0002
libwdi:debug [wdi_create_list] Device description: 'Corsair Gaming KATAR Mouse (Interface 2)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_1B1C&PID_1B22&REV_0107&MI_03
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.18199
libwdi:debug [wdi_create_list] HidUsb USB device (19): USB\VID_1B1C&PID_1B22&MI_03\8&36795CC6&0&0003
libwdi:debug [wdi_create_list] Device description: 'Corsair Gaming KATAR Mouse (Interface 3)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_8087&PID_8000&REV_0005
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_09&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.18328
libwdi:debug [wdi_create_list] usbhub USB device (20): USB\VID_8087&PID_8000\5&18088FC2&0&1
20 devices found.
libwdi:debug [wdi_create_list] Device description: 'Generic USB Hub'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_8087&PID_8008&REV_0005
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_09&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.18328
libwdi:debug [wdi_create_list] usbhub USB device (21): USB\VID_8087&PID_8008\5&49593B2&0&1
libwdi:debug [wdi_create_list] Device description: 'Generic USB Hub'
Using inf name: WUP-028.inf
Could not extract files
Driver Installation: FAILED (Could not allocate resource)
libwdi:error [extract_binaries] failed to create file 'C:\Users\Moon Cricket\usb_driver\.\installer_x64.exe' ([#00000020] The process cannot access the file because it is being used by another process.)
Saved file as 'C:\Users\Moon Cricket\Downloads\zadig.log'

TLonk, 2016-05-20:
It's not helpful, but the same thing happened to me and I have windows 10. I'm going to reinstall everything, and correct me if I'm wrong, but WUP-028 is a Wii U GC adapter, right?

pbatard, 2016-08-24:
I suspect that this error can be averted by issuing a reboot and trying to run Zadig again. I will close it for now. Feel free to reopen if that's not the case.

closed 2016-08-24 by pbatard