pbatard/libwdi #58: WinUSB driver not available

Pedro147, 2016-05-28:
I hope it is OK to post this problem as an issue here and if not I apologise. I am using Zadig to replace a driver for an STM32 device, a quadcopter flight control board. [Here](http://alienflight.com/firmware-flashing-f3/) is the recommended procedure which I have followed to the letter. So as instructed I install the Dfuse package, run the requested exe file which I assume installs the STM driver for this device. I then open Zadig to replace the STM driver with the WinUSB one as required. I get a notification that this has been successfully done but then when I go to Windows device manager the device is listed under unknown devices which to me indicates that there is no driver available to it. I then tried to go through the Zadig driver replacement procedure again but it now shows that the WinUSB driver is the one available for this device. I have contacted one of the developer's of this flight board @mj666 but be does not seem able to offer any solution so that is why I am here asking for help. There is definitely no problem with the flight board as it connects correctly on another PC of mine running the same OS etc. Another friend of mine is experiencing exactly the same issues with his flight board too. Is there any other way that I can make the WinUSB driver available to this device please as I have exhausted all other avenues of help. Thank you 
[image]
[image]
[image]
[image]
[image]
[image]
[image]

MJ666, 2016-05-28:
What I can see on your screen shots is that the STM VCP driver did not install corectly for any reason. This has noting to do with the DFU mode driver. You shoul be able the flash the FC like this but will not be able to comunication via the configurator. Did you try to uninstall and reinstall the the STM driver and always reboot in between? It is definitely an STM driver issue.

Pedro147, 2016-05-28:
@mj666 how do I uninstall the STM driver please? Do I just delete uninstall it via the properties dialogue box in device manager and then just run the dpinst_x##.exe file in the the STMicroelectronics folder in Program files on C Drive to reinstall that STM driver . You say that "You shoul be able the flash the FC like this but will not be able to comunication via the configurator." but the only way I can flash the firmware is by using the configurator isn't it? This is what shows up in Cleanflight with the FC in DFU mode
[image]

diabolo38, 2016-05-28:
ISn't ST VCP device showing cause your f/w is based on "CDC/ACM"  f/w class with default ST VID/PID ?
if so you should change thems to your own VID/PID , so that ST VCP will not associate to your device.
this driver is commonly installed as part of stlink installer  (nucleo also include a vcp serial).
  
to un-install a driver yes press uninstall in the property dialog, in next dialog you can select "delete" tick box to delete all file form system in a more radical way so that driver will re-associate in next plug .
[image]

if device is not plug or listed follow
 this  https://support.microsoft.com/en-us/kb/315539 to get "hiden" device" to show.

Pedro147, 2016-05-28:
@diabolo38 you are obviously a lot better than me at all this but this reminds me that I recently used an ST-link V2 dongle for this [project](http://www.rcgroups.com/forums/showthread.php?t=2512604) that used these [drivers](http://www.st.com/content/st_com/en/products/embedded-software/development-tool-software/stsw-link009.html). Do you think that this could possibly be causing a conflict with what I am trying to do with this flight board? If this is a possibilty how do I reslove this conflict if possible. Thanks for your help here

MJ666, 2016-05-28:
It will be sufficient to uninstall the driver in the device manger only, like @diabolo38 has described. No need for uninstalling the complete software. You should reboot after uninstalling and installing the STM driver. If this does not work there will be an conflict on your PC. Unfortunatey in various STM forums you can see lots of reports like this.

The screenshot you send is telling me the configurator can see the FC in DFU mode. The configurator can flash the FC when in DFU mode and the zadig driver is installed. You will see DFU in the window were you usuay see the com port.

On some PC's we have seen issues that the com port is only detected after an complete disconnect of the FC. This does not look to be related to an specific version of windows.

pbatard, 2016-05-28:
Just so you know, if you go to _Options_ &rarr; _Advanced Mode_, you can get the full log from the driver installation, which usually contains __very detailed__ data as to what problems Windows may have encountered during the driver installation.

Or on Vista or later, you can have a look at `C:\Windows\INF\setupapi.dev.log`, which contains the same data (including data from your previous installation). If you are using XP, there's a similar file, but I can't remember what it's called (XP has been retired for 2 years already - please move on)...

Pedro147, 2016-05-29:
When I have access to my work PC again (where this problem occurs) I will uninstall the driver, reinstall it and reboot the PC. Do I need to reboot between the uninstall and the reinstall or just after the reinstall please. @mj666 can you recommend the best STM forum to go to if there is indeed a conflict on the PC that is causing this which is what I suspect. So you are saying that the FC can be put into DFU mode and flashed, but because it will not connect to the configurator via serial, I can therefore not configure any settings for the FC in Cleanflight and if that is correct then that is of not much use is it :-)

@pbatard I will look at this log thank you but from what mj666 is saying Zadig has replaced the STM driver with the WinUSB driver although I cannot understand why there are no details for the driver showing in Device Manager but as I said I am pretty clueless in regard to all this compared to you guys. If it is OK I will leave this issue open until I can check the things that you three gents have shown me although it may be a few days seeing as I am off work sick for the rest of the week.

MJ666, 2016-06-01:
@Pedro147 yes you should reboot twice! There is no specific STM forum. There are various and i also would need to use Google again to find places were there are similar problems than yours. Beside the STM resources you can also find information in the related firmware threads (CleanFlight, BetaFlight, RaceFlight). Not sure why this is not working on some PC's and some users have problems. The STM driver looks to be a bit tricky. For me personally i could fix all problems i had in the past with the steps I described here (and at other places already). You need to fix your driver problem or in the worst case use an different computer to connect to the FC to configure it. This can affect all FC wich are using the build in STM VCP port and is not really specific to the AlienFlight FC only.

Pedro147, 2016-06-01:
Thank you once again @mj666 and I am only pursuing this in a variety of forums because that is the only way to potentially get help for this annoying issue. I also am aware that this is a problem not specific to AlienFlight FC's but more generally to the STM32 processor used. I will close this issue not because it is solved but because it seems that what I have been told here is the extent of the advice that you good gents can offer me, and for this I thank you.

closed 2016-06-01 by Pedro147