pbatard/libwdi #60: 2in1 Joystick Constant Vibration

wafflefaust, 2016-06-14:
Hello, I've been trying to use Zadig to get the libusb drivers installed for my knock-off PS3 DualSchock3 controller so that I can use ScpToolKit to have correct button mapping. The problem I'm having is that the controller will not stop vibrating, and sends no key data after installing the libusb0, libusbK or WinUSB.. I've tried them all...

Now when I go to use Zadig to try to change the driver again or extract files it gives me this error: "The driver installation failed."

I'm running version 2.2 of Zadig and the latest ScpToolKit version. Not too sure what's going on here, but I really want to use this controller and not be forced to exchange it for an xbox 360 one.

pbatard, 2016-06-19:
Please run Zadig in advanced mode (Options &rarr; Advanced Mode) and report what the log says.

pbatard, 2016-08-24:
No update from user &rarr; Closing

closed 2016-08-24 by pbatard