pbatard/libwdi #61: Ignoring changes to inbox device ClassInstall32 section: Windows8.1 x64

jon452, 2016-06-20:
Hello!
Zadig Version 2.2.689 (Build 0)
Zadig(WinUSB) for Windows7 x64 all work great!

When I try use zadig WinUSB for Windows8.1 x64,I get Exception:
> USB error 12: Unable to open USB device: Operation not supported or unimplemented on this platform

When I install zadig WinUSB for Windows8.1 x64 in log file Zadig:

libwdi:debug [syslog] !    sig:                Verifying file against specific (valid) catalog failed! (0x800b0109)
libwdi:debug [syslog] !    sig:                Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.
...
libwdi:debug [syslog]      sig:                Error 0xe0000241: The INF was signed with an Authenticode(tm) catalog from a trusted publisher.
...
libwdi:debug [syslog] !    sig:                          Verifying file against specific (valid) catalog failed! (0x800b0109)
libwdi:debug [syslog] !    sig:                          Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.
...
libwdi:debug [syslog] !    sto:                     Ignoring changes to inbox device class {88bae032-5a81-49f0-bc3d-a4ff138216d6} through ClassInstall32 section.
...
libwdi:debug [syslog] !    idb:                          Skipping update of existing inbox class {88bae032-5a81-49f0-bc3d-a4ff138216d6}.

I try and it not supporting:
1.)install WinUSB zadig as Administrator
2.)Test Mode Windows8.1 ON
3.)Code signing for device drivers:Disable.
4.)service Windows update run

I don't understand why zadig(WinUSB)  Windows 7 work great and zadig(WinUSB) Windows 8.1 not working...
Full log:
[zadig2.txt](https://github.com/pbatard/libwdi/files/323004/zadig2.txt)

pbatard, 2016-06-20:
`Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.` is normal. What matters is this:
```
libwdi:debug [syslog]      dvi:                Install Device: Starting device completed. 00:44:03.030
libwdi:debug [syslog]      ndv:           {Core Device Install - exit(0x00000000)} 00:44:03.032
libwdi:debug [syslog]      ump:      {Plug and Play Service: Device Install exit(00000000)}
libwdi:debug [syslog]      ndv: {Update Device Driver - exit(00000000)}
libwdi:debug [syslog] <<<  Section end 2016/06/20 00:44:03.051
libwdi:debug [syslog] <<<  [Exit status: SUCCESS]
```
which indicates that the WinUSB driver was installed successfully.

Your `USB error 12: Unable to open USB device: Operation not supported or unimplemented on this platform` seems to come from your own application, not Zadig, so, since Zadig reports that the driver was installed successfully, I don't think this is something I can help you with. You'll have to contact the developers of the application that produces `USB error 12`.

jon452, 2016-06-20:
But this app work great on Windows 7.
And this app work great on Debian.
I change only Windows7 to Windows8.1.

pbatard, 2016-06-20:
The fact that your app works fine on another platform does not mezaThere are many differences that your OS can have

jon452, 2016-06-20:
Thanks pbatard.
I developer this app.
I test platforms and I think it not trouble libUSB because deb and win7 work.
I get this Exception when I try function libUSB OpenDevice().
I was hoping that you can help me.
Ok thanks.

closed 2016-06-20 by jon452