pbatard/libwdi #62: STM32F3 In DFU mode with read protection

fedorcomander, 2016-07-13:
When STM32F3 has read protection enabled, it fails to register in DFU mode. 

[image]

Left windows - device with read protection enabled, Right window - device without read protection.

fedorcomander, 2016-07-13:
I am trying to flash KISSFC STM32 based board using DFU from Chrome app using chrome USB api...
The driver fails to register device, when STM32 chip is in read protect mode. If i remove read protect, using DFuseDemo and StmTube drivers, i se that device driver is properly initialized.

Any ideas what can be done for read protected device?

pbatard, 2016-07-13:
Sorry, but this rather looks like a device specific issue rather than a libwdi one.

Also, when you say "it fails to register", what do you mean exactly?
Do you mean that the driver installation using libwdi/Zadig fails?
If so, what is the content of your `C:\Windows\inf\setupapi.dev.log` after the driver installation against the DFU device?
If not, how is that related to libwdi (i.e. in which place are you actually using libwdi)?

pbatard, 2016-08-24:
No user input &rarr; Closing.

closed 2016-08-24 by pbatard