pbatard/libwdi #63: In Windows 10 Zadig stops witrh message, cancelled by user.

MarBo72, 2016-07-14:
The log file shows an error " Error 87: The parameter is incorrect" prior to issuing the screen message.
This all came about trying to update SDRSHARP to work on Win 10. I cannot see where the block/interrupt starts.. Can anyone help. Thanks.
Here is a copy of the error log where this error appears. 
libwdi:debug [syslog]      sig:                          Key      = rtl2838uhidir.inf
libwdi:debug [syslog]      sig:                          FilePath = C:\WINDOWS\System32\DriverStore\Temp\{0d55b086-a339-ab48-b5bd-111dfd83ef53}\rtl2838uhidir.inf
libwdi:debug [syslog]      sig:                          Catalog  = C:\WINDOWS\System32\DriverStore\Temp\{0d55b086-a339-ab48-b5bd-111dfd83ef53}\RTL2838UHIDIR.cat
libwdi:debug [syslog] !    sig:                          Verifying file against specific (valid) catalog failed! (0x00000057)
libwdi:debug [syslog] !    sig:                          Error 87: The parameter is incorrect.
libwdi:debug [syslog]      sig:                     {_VERIFY_FILE_SIGNATURE exit(0x00000057)} 14:24:46.345
libwdi:debug [syslog]      sig:                     {_VERIFY_FILE_SIGNATURE} 14:24:46.345
libwdi:debug [syslog]      sig:                          Key      = rtl2838uhidir.inf
libwdi:debug [syslog]      sig:                          FilePath = C:\WINDOWS\System32\DriverStore\Temp\{0d55b086-a339-ab48-b5bd-111dfd83ef53}\rtl2838uhidir.inf
libwdi:debug [syslog]      sig:                          Catalog  = C:\WINDOWS\System32\DriverStore\Temp\{0d55b086-a339-ab48-b5bd-111dfd83ef53}\RTL2838UHIDIR.cat
libwdi:debug [syslog] !    sig:                          Verifying file against specific Authenticode(tm) catalog failed! (0x800b0100)
libwdi:debug [syslog] !    sig:                          Error 0x800b0100: No signature was present in the subject.
libwdi:debug [syslog]      sig:                     {_VERIFY_FILE_SIGNATURE exit(0x800b0100)} 14:24:46.345
libwdi:debug [syslog] !!!  sig:                     Driver package catalog file does not contain a signature, and Code Integrity is enforced.
libwdi:debug [syslog] !!!  sig:                     Driver package failed signature validation. Error = 0xE0000247
libwdi:debug [syslog]      sto:                {DRIVERSTORE IMPORT VALIDATE: exit(0xe0000247)} 14:24:46.345
libwdi:debug [syslog] !!!  sig:                Driver package failed signature verification. Error = 0xE0000247
libwdi:debug [syslog] !!!  sto:                Failed to import driver package into Driver Store. Error = 0xE0000247
libwdi:debug [syslog]      sto:           {Stage Driver Package: exit(0xe0000247)} 14:24:46.361
libwdi:debug [syslog]      sto:      {Setup Import Driver Package - exit (0xe0000247)} 14:24:46.361
libwdi:debug [syslog] !!!  ndv:      Driver package import failed for device.
libwdi:debug [syslog] !!!  ndv:      Error 0xe0000247: A problem was encountered while attempting to add the driver to the store.
libwdi:debug [syslog]      ndv: {Update Device Driver - exit(e0000247)}
libwdi:debug [syslog] !!!  ndv: Failed to install device instance USB\VID_0BDA&PID_2838\00000001 (Error 0xe0000247)
libwdi:debug [process_message] switching timeout back to finite
libwdi:debug [installer process] operation cancelled by the user
libwdi:debug [process_message] installer process completed
Driver Installation: Cancelled by User
2 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0AC8&PID_C349&REV_0015&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_0e&SubClass_03&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.10240.16384
libwdi:debug [wdi_create_list] usbvideo USB device (0): USB\VID_0AC8&PID_C349&MI_00\7&2F62A5B2&1&0000
libwdi:debug [wdi_create_list] Device description: 'WebCam SC-03FFL11739P (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0BDA&PID_2838&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\DevClass_00&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Upper filter: RTL2832UBDA
libwdi:debug [wdi_create_list] Driver version: 64.1.304.2009
libwdi:debug [wdi_create_list] RTL2832UUSB USB device (2): USB\VID_0BDA&PID_2838\00000001
libwdi:debug [wdi_create_list] Device description: 'RTL2838UHIDIR'

pbatard, 2016-08-24:
Can you upload a copy of your driver files? You should find them in C:\Users\<Your user>\drivers

pbatard, 2016-11-01:
No update &rarr; Closed.

closed 2016-11-01 by pbatard

dl9sec, 2019-01-09:
I suspect you have UEFI secure boot activated on your Win10 PC.
This prevents the driver to install. Although the driver is signed, Windows allows only to install trusted signatures. Similar problems exists with e.g. TUN/TAP drivers on Win10 (as a workaround they install old driver versions before a explicite date which allows installation without trusted signature)...

dl9sec, 2019-01-09:
Could also be a "Driver signature enforcement", see https://windowsreport.com/driver-signature-enforcement-windows-10/...

pbatard, 2019-01-09:
Nah, given the error reported in the `setupapi.dev.log` (which Zadig simply reads from during driver install - You can usually find the original at `C:\Windows\INF\setupapi.dev.log`), this is most likely due to OP having configured their machine in a manner that prevents `.cat` files from being signed.

Most likely, they got an earlier message of:

```
libwdi:warning [wdi_prepare_driver] could not sign cat file
```

But since they chose to truncate their log (that is why it is __crucial__ to always copy/paste the FULL log), we don't get to see that earlier issue and we are left wasting time looking for probable causes.

Still, the `Driver package catalog file does not contain a signature, and Code Integrity is enforced.` is pretty explicit, and this looks exactly like #149. So my bet is that OP's system cannot sign `.cat` files for whatever reason (which OP will have to figure out), which of course, prevents newer version of Windows from installing the driver, since they require a signed `.cat`.

Note that, since the error reported by Zadig (_Cancelled by user_ when it's really a Cat file signing issue) is not accurate, I have already applied an enhancement [8123b41ad0eb0baf98e6b94e35ad7bf3affe938c] to provide a better error message. But again, there's not much I can do if the system has been altered to make Cat file signature fail, as all libwdi/Zadig is doing there is using standard system calls that are expected to work (and do work just fine on vanilla Windows installations).

dl9sec, 2019-01-11:
Hmmm. Thanks for the hints and explanations. 
It  seems, that (in my case) the self signed certificate(s) can not be added to "TrustedPublisher" (see attachment). Could it be a workaround to import these certificate(s) manually to "TrustedPublishers" and then try to install? Maybe the certificate addition will fail, but if the certificates are there the cat-file could be verified...