pbatard/libwdi #66: Mouse still not working after reinstalling System driver

ItsMeNiV, 2016-10-04:
I accidentally replaced the driver of my mouse and it stopped working. I already tried the method in the FAQ and it still doesn't work. What else could be the problem? My old mouse works and the affected mouse works on other PCs

pbatard, 2016-10-04:
I'm afraid that the only advice I can give is "you're not doing it properly".

I regularly get report from people indicating that they did follow the FAQ, and it didn't work, and in __every single case__ it turns out that, after they insist, make sure they do properly select the "also delete driver" checkbox, plug their device to a different port and/or reboot the computer, it eventually works.

So, I'm afraid you will need to try again, and make sure you plug your mouse and desinstall the driver from each port where it may have been plugged, and you should find that it eventually works.

closed 2016-11-01 by pbatard

Kreijstal, 2024-07-21:
For anyone that still has this problem, you can use https://github.com/lostindark/DriverStoreExplorer remove libwdi manually, restart and it should work, what a headache lmao