pbatard/libwdi #69: An issue with USBasp

shturman0000, 2016-11-02:
During an installation of USBasp driver by the Zadig there is a message:"Driver Installation: Failed (Could not allocate resource)".

pbatard, 2016-11-03:
Please try to delete your `C:\Users\<your user name>\usb_driver\` directory and try again.
Make sure that, when you try again, you check _Options_ &rarr; _Advanced Mode_ in Zadig, so that you will get a log that you can copy/paste if it doesn't work.

pbatard, 2017-02-01:
Gonna close this issue now.

closed 2017-02-01 by pbatard

DRSDavidSoft, 2017-07-23:
This should be either:
1) Mentioned in the app directly, or
2) Be automatically done by Zadig,
instead of us Googling it, finding the issue on GitHub, trying to delete the folder, finding out that `installer-x64.exe` is open and thus locking the folder, then doing `taskkill /f /t /im installer-x64.exe`, then removing the folder.

MariwanJ, 2021-01-15:
DRSDavidSoft 
Thanks for your hlep. I couldn't remove the folder .. 
and that commad didn't work either but I used Process Explorer and killed the process. 
Again thanks for all of you for your help.

Littlejd1997, 2021-04-14:
Adding more details. For me it was `installer_x64.exe` (underscore not hyphen). This also happened to me after a "Failed" installation for a different device. zadig claimed that the install had failed, but the drivers were loaded in for me. It looks like the installer hung and prevented zadig for the next run

Muhammad-Hazimi-Yusri, 2023-03-23:
> DRSDavidSoft Thanks for your help. I couldn't remove the folder .. and that command didn't work either but I used Process Explorer and killed the process. Again thanks for all of you for your help.

make sure you have C:\Windows\System32 in your system PATH, or you can try to specify the full path of the taskkill file, which is usually C:\Windows\System32\taskkill.exe.