pbatard/libwdi #70: Bluesceen after Zadig Usage on Win XP

roby1111, 2016-11-20:
Hello, 

i used Zadig, XP-Version, which was part of a Freeware tool to extend possibilities of low-cost-scopes like Hantek and added the WINUsb-Driver for Hantek1008C with its help.

After a reboot (i made to ensure stable function) the System did not boot anymore! Even in safe mode (f8) or "last known good" config it did not start. It showed the following message on the Bluescreen: 

Stop: C000021a {Daten wurden nicht akzeptiert}
Der TDI-Client konnte die empfangenen Daten nicht verarbeiten.

It is XP SP3 prof. with german localisation.

I could not find any similar message on the Web, the only ones showing up with "C000021a" had different text-portions and also did not correspond to the phenomen i got.

So i started investigations...

After searching through all recently changed files and replacing the complete Registration files by an elder, well working copy (did not Change anything!), i finally found an install-log of winusb that showed up with some crazy lines telling that msvcrt.dll was replaced, but also that mfc42u.dll was replaced by a file named mfc42u.dll.tmp so that this already showed not functional...

I replaced those files against their previous Versions (found in the drivercache or Service-Pack Folders (I386) of the Windows Installation, so no backup is needed normally and was able to boot again!

I publish this Information as i did not find any hint on this error-type on the Internet before!

So maybe it is helpful for some others too.

I don t know, what went wrong there, but if this update-script for winusb is meant that way, i would be eager to know for what reason the .tmp file was put to system32, while deleting the functional mfc42u.dll from there?

Maybe somebody can test the Zadig XP functionality against that?

pbatard, 2016-11-20:
> which was part of a Freeware tool

Are you sure this was the official Zadig version? I digitally sign all the Zadig versions I publish. Was the Zadig you downloaded digitally signed? If so, what was the name of the signature?

What you report is very strange. A driver installation should not replace `msvcrt.dll` (that's what malware does!) so it would look to me like you may have been using a "fake" version of Zadig that was modified to do something it shouldn't do. On the other hand, we're talking about XP, which is a very old OS (that NOBODY in their right mind should still be using), so who knows.

Still, my first advice is: Always download Zadig from http:\\zadig.akeo.ie and verify that it is digitally signed and that the signature says "Akeo Consulting".

Second: Do you have the install log and can you please post the section you mention about the driver installation replacing DLLs?

roby1111, 2016-11-20:
Hello Pete,

I tried answering your mail directly, but pop3-client told me that the adress for reply was not correct that way. But maybe that is a wanted effect...

So here and now:

well, please excuse my english.

I tried out a freeware named "sigrok-cli", which worked good so far, but the support for the hantek 1008c was in state "vaporware" there, so unuseful for the moment. But this download contained your Zadig XP software or claimed to do so... (Screens i saw on my machine and here online are identical, so i guess that at least the "frame"-program was original, but maybe not the installer portion...

I will try to find that log-file, it was part of winusb as far as i do remember.

There were some really strange entries which showed up with files from recent, but completely different installations on the same maschine.

This 10year old notebook computer (Tecra A9 from toshiba) i still run with XP for convenience, as i ve got all my lab-tools (i am working as electronics engineer in automotive business) work best on XP and also i need support for PCMCIA/Cardbus peripheral, the toshiba was one of the last maschines which had all of that but also dual-core CPU and modern features too.

Also i love stable relationships,not only between humans ;o)

Ok, back to that problem (that i spent the whole Saturday with...):

I found again this file which brought me on the right track:

It is located in windows\temp and is named: winusb_update.log and has a timestamp of exactly that time, i started the "Zadig XP" for exchanging the driver of the hantek1008C (OEM-driver) with WINUSB like mentioned in that tool.

When i had a closer look at everything now again, it seems to me that really something strange happened, but maybe could not succede in taking over my machine as some server to download was offline already...

The update itself started from a temporary folder in the c: root-dir as you can see from the logfile, this folder did not exist afterwards. But the uninstallfolder named in the logfile does still exist and contains some .exe files also, but looks like normal uninstall folder does...

I have attached that file for your convenience and can also send you the cabinet-file of of sigrok-cli with which "Zadig" and "Zadig XP" were delivered and installed on my machine.

As far as i understood, i ve downloaded also sigrok from: http://sigrok.org/wiki/Downloads

as "sigrok-cli-NIGHTLY-32bit-static-release-installer.exe" (18.11.2016 23:16 munich time)

I also made virus check now again (runs as background service also), nothing so far.

I will start now some anti-Malware tools to see if those will find something suspicious...
I will report to you if it found any suspicious files or activity!

Thanks so far!

best regards,

Robert

roby1111, 2016-11-20:
As i could not email, here is the winusb_update.log in complete:

[winusb_update.log]
0.219: ================================================================================
0.219: 2016/11/18 23:31:57.916 (local)
0.219: c:\8d05fce81552885c8b5f8530a39f\update\update.exe (version 6.3.15.0)
0.234: Hotfix started with following command line: /quiet /o /ER /norestart /log:C:\WINDOWS\temp\winusb_update.log 
0.234: In Function GetReleaseSet, line 1240, RegQueryValueEx failed with error 0x2
0.797: In Function TestVolatileFlag, line 11825, RegOpenKeyEx failed with error 0x2
0.797: In Function TestVolatileFlag, line 11857, RegOpenKeyEx failed with error 0x2
0.797: ---- Old Information In The Registry ------
0.812: Source:C:\DOKUME~1\admin\LOKALE~1\Temp\NGLATempNokia\Nokia Sans Wide Bold v3.1.ttf (0.3.0.10)
0.812: Destination: 
0.812: Source:C:\DOKUME~1\admin\LOKALE~1\Temp\NGLATempNokia\Nokia Sans Wide v3.1.ttf (0.3.0.10)
0.812: Destination: 
0.812: Source:C:\WINDOWS\TEMP\7zS461.tmp\AVG-Secure-Search-Update.exe 
0.812: Destination: 
0.812: Source:C:\WINDOWS\system32\~GLH0079.TMP (6.6.8064.0)
0.812: Destination:C:\WINDOWS\system32\MFC42.DLL (6.2.8081.0)
0.812: Source:C:\WINDOWS\system32\~GLH007a.TMP (7.0.7601.17744)
0.812: Destination:C:\WINDOWS\system32\msvcrt.dll (7.0.2600.5512)
0.812: Source:C:\WINDOWS\system32\@c:\windows\system32\mfc42u.dll.tmp 
0.812: Destination: 
0.812: ---- New Information In The Registry ------
0.812: Source:C:\DOKUME~1\admin\LOKALE~1\Temp\NGLATempNokia\Nokia Sans Wide Bold v3.1.ttf (0.3.0.10)
0.812: Destination: 
0.812: Source:C:\DOKUME~1\admin\LOKALE~1\Temp\NGLATempNokia\Nokia Sans Wide v3.1.ttf (0.3.0.10)
0.812: Destination: 
0.812: Source:C:\WINDOWS\TEMP\7zS461.tmp\AVG-Secure-Search-Update.exe 
0.812: Destination: 
0.812: Source:C:\WINDOWS\system32\~GLH0079.TMP (6.6.8064.0)
0.812: Destination:C:\WINDOWS\system32\MFC42.DLL (6.2.8081.0)
0.812: Source:C:\WINDOWS\system32\~GLH007a.TMP (7.0.7601.17744)
0.812: Destination:C:\WINDOWS\system32\msvcrt.dll (7.0.2600.5512)
0.812: Source:C:\WINDOWS\system32\@c:\windows\system32\mfc42u.dll.tmp 
0.812: Destination: 
0.812: In Function GetReleaseSet, line 1240, RegQueryValueEx failed with error 0x2
0.812: SetProductTypes: InfProductBuildType=BuildType.IP
0.812: SetAltOsLoaderPath: No section uses DirId 65701; done.
1.016: DoInstallation: FetchSourceURL for c:\8d05fce81552885c8b5f8530a39f\update\update_XP.inf failed
1.016: LoadFileQueues: UpdSpGetSourceFileLocation for halmacpi.dll failed: 0xe0000102
1.016: CreateUninstall = 1,Directory = C:\WINDOWS\$NtUninstallwinusb0200$ 
1.016: BuildCabinetManifest: update.url absent
1.016: Starting AnalyzeComponents
1.016: AnalyzePhaseZero used 0 ticks
1.016: No c:\windows\INF\updtblk.inf file.
1.016: User specified '-o'; ignoring any OEM driver files.
1.016: AnalyzePhaseOne: used 0 ticks
1.016: AnalyzeComponents: Hotpatch analysis disabled; skipping.
1.016: AnalyzeComponents: Hotpatching is disabled.
1.531: AnalyzePhaseTwo used 515 ticks
1.531: AnalyzePhaseThree used 0 ticks
1.531: AnalyzePhaseFive used 0 ticks
1.531: AnalyzePhaseSix used 0 ticks
1.531: AnalyzeComponents used 515 ticks
1.531: Downloading 0 files
1.531: bPatchMode = FALSE
1.531: Inventory complete: ReturnStatus=0, 515 ticks
1.531: Num Ticks for invent : 515
1.531: Allocation size of drive C: is 4096 bytes, free space = 33286488064 bytes
1.531: AnalyzeDiskUsage:  Skipping EstimateDiskUsageForUninstall.
1.531: Drive C: free 31744MB req: 6MB w/uninstall: NOT CALCULATED.
1.531: CabinetBuild complete
1.531: Num Ticks for Cabinet build : 0
1.531: DynamicStrings section not defined or empty.
1.531: FileInUse:: Added to Filelist: c:\windows\winusb0200.cat
1.531: FileInUse:: Added to Filelist: c:\windows\system32\drivers\winusb.sys
1.531: FileInUse:: Added to Filelist: c:\windows\system32\winusb.dll
1.594: FileInUse:: IsTaskUsingModule: EnumerateLoadedModules64 failed: 0xc0000022
1.594: FileInUse:: IsTaskUsingModule: EnumerateLoadedModules64 failed: 0xc0000022
[... 46 lines cut ...]
9.172: Source:C:\DOKUME~1\admin\LOKALE~1\Temp\NGLATempNokia\Nokia Sans Wide v3.1.ttf (0.3.0.10)
9.172: Destination: 
9.172: Source:C:\WINDOWS\TEMP\7zS461.tmp\AVG-Secure-Search-Update.exe 
9.172: Destination: 
9.172: Source:C:\WINDOWS\system32\~GLH0079.TMP (6.6.8064.0)
9.172: Destination:C:\WINDOWS\system32\MFC42.DLL (6.2.8081.0)
9.172: Source:C:\WINDOWS\system32\~GLH007a.TMP (7.0.7601.17744)
9.172: Destination:C:\WINDOWS\system32\msvcrt.dll (7.0.2600.5512)
9.172: Source:C:\WINDOWS\system32\@c:\windows\system32\mfc42u.dll.tmp 
9.172: Destination: 
9.172: ---- New Information In The Registry ------
9.187: Source:C:\DOKUME~1\admin\LOKALE~1\Temp\NGLATempNokia\Nokia Sans Wide Bold v3.1.ttf (0.3.0.10)
9.187: Destination: 
9.187: Source:C:\DOKUME~1\admin\LOKALE~1\Temp\NGLATempNokia\Nokia Sans Wide v3.1.ttf (0.3.0.10)
9.187: Destination: 
9.187: Source:C:\WINDOWS\TEMP\7zS461.tmp\AVG-Secure-Search-Update.exe 
9.187: Destination: 
9.187: Source:C:\WINDOWS\system32\~GLH0079.TMP (6.6.8064.0)
9.187: Destination:C:\WINDOWS\system32\MFC42.DLL (6.2.8081.0)
9.187: Source:C:\WINDOWS\system32\~GLH007a.TMP (7.0.7601.17744)
9.187: Destination:C:\WINDOWS\system32\msvcrt.dll (7.0.2600.5512)
9.187: Source:C:\WINDOWS\system32\@c:\windows\system32\mfc42u.dll.tmp 
9.187: Destination: 
14.328: UpdateSpUpdSvcInf: Source [ProcessesToRunAfterReboot] section is empty; nothing to do.
14.344: RebootNecessary = 0,WizardInput = 1 , DontReboot = 1, ForceRestart = 0

roby1111, 2016-11-20:
Well, the file contains information about my anti-virus software (AVG) and also some old Nokia software (or at least one of the nokia fonts left from this). So it seems very mixed up at all, also that "@" Character in the middle of one path: nothing that windows could deal with...

roby1111, 2016-11-20:
malware bytes also found nothing, as AVG did, both with todays definition-updates successfully loaded and some "honey pot" files on the disc to see if they are still useful, so far everythings ok. 

I will wait some days and repeat the scan then.

pbatard, 2016-11-21:
Thanks for the additional details. I looked at the sigrok binary, and it contains the digitally signed version of `zadig.exe` and `zadig_xp.exe`, so I'm positive that whatever happened to your XP computer wasn't due to that application containing malware.

By the looks of it, this could be an issue with the official WinUSB installation files from Microsoft (which is what Zadig invokes behind the scene), which, if Windows XP was still supported, I would advise you to contact Microsoft about, as this is not a Zadig issue. Unfortunately, since XP is no longer supported, you won't be able to do that, and I too don't think there's anything more I'm willing to do to help, as my time is too limited to try to help users of an obsolete OS.

As such, I will close this issue.

closed 2016-11-21 by pbatard

roby1111, 2016-11-21:
Thanks for your time and informations! I think there are alot of users outside who CAN NOT switch from XP to something more new, as old interfaces like PCMCIA are no longer supported, the new OS is not reliable enough for some industries or it has to work without internet-connection in any case for safety- or security-reasons.

So even MS supports XP still if it is "embedded" in industrial applications. So it would be nice to leave my information about a possible error in WINUSB-Upgrade for XP on this site!

I am used to support old applications in my business, working not only for automotive but also railway-industries.  So almost each day i have to deal with Z80, 8031 and similar environments. So don t call XP obsolete, its rather fresh !

roby1111, 2016-11-21:
And please: i did not want any support, i just wanted to post a warning to all XP users, to be aware if they decide to install WINUSB-updates, that this could be fatal. With or without Zadig XP! And to give them a chance to save some time, that i spent on searching for the root-cause as this Error-typ is not explained somewhere else on the web so far!

pbatard, 2016-11-21:
I do understand your point of view. However, please understand that I am providing the program entirely for free, and I am not paid for providing support. As such, and since my time is __EXCEEDINGLY__ limited, I just don't have the luxury of supporting OSes that I have been retired, as I'd rather concentrate my efforts on modern and well supported versions of Windows (so that, if users do encounter issues that require Microsoft's involvement, __as you do__, something can actually be done about these).

I'm sorry, but it's just not worth my time and effort to continue to support XP, and I've already gone beyond what I would normally have done in terms of support (which is usually to tell people who are using XP that _"if it works, great! If it doesn't, you're on your own"_).

The issue will remain in the issue tracker. And it will be seen by people looking for similar error as the one you got. But I can't reasonably keep it open, as this is not something that I will personally support (and, obviously, only the issues that I plan to actively support should be kept open).

roby1111, 2016-11-21:
OK, thanks!