pbatard/libwdi #71: Windows 7 Zidag WinUSB overwrite of Ralink Blutetooth Adapter causes BSOD

NirLevanon, 2016-12-01:
Zadig version: 2.2.689

I've downloaded the latest version of Zadig from http://zadig.akeo.ie/.
Per instructions from the [Bleno node module](https://github.com/sandeepmistry/node-bluetooth-hci-socket#windows) I'm using, I've attempted to replace the Ralink RT3290 Bluetooth driver on my Windows 7 x64 laptop with WinUSB v6.1.7600 which comes with Zadig. Watching the log from Advanced Mode, I saw that the driver was almost completely installed when at the moment Zadig tried to reset the device the system crashed to a BSOD (as of last attempt there was no BSOD but simply a crash with an immediate forwarding to the system recovery screen [warning of unclean shutdown]).

I was unable to figure out how to retrieve the log of the installation attempt as it seems the only moment I can save a log is before or after an installation attempt, which I can never reach because of BSOD. Searching through System Events yeilded no significant message beyond the indication of Kernel-Power critical events, nothing before or during the installation process.

Perhaps worthy of mention is that I was able to install WinUSB on an Intel Bluetooth driver on my other Windows 10 laptop.

I'd appreciate some directions to trying to debug my issue. Alternative solutions to installing WinUSB that would potentially work with Bleno would be appreciated.

Side note - Bleno uses node-usb which works with libusb.c

pbatard, 2016-12-01:
Have a look at `C:\Windows\INF\setupapi.dev.log`. It contains the driver installation details.

For the record, Zadig does NOT issue a reset to devices (which is actually a very difficult task to do for USB devices in Windows). It just asks Windows to install the driver and whatever happens during or after that is between Microsoft and your device.

pbatard, 2017-02-01:
Gonna close this issue now.

closed 2017-02-01 by pbatard