pbatard/libwdi #73: Doesn't install my 3 keys, on my WinXP SP3 laptop - REALTEK_2832U_Device

Toysoft, 2016-12-26:
See below log, the original driver is installed successfully and I can watch TV no problem, but then if I try to upgrade to SDR via your tool, it doesn't work.

Thank you in adance a LOT ! Merci ;o)

Jan

Using inf name: REALTEK_2832U_Device.inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] successfully extracted driver files to C:\Documents and Settings\Jan\usb_driver
libwdi:info [wdi_prepare_driver] successfully created 'C:\Documents and Settings\Jan\usb_driver\REALTEK_2832U_Device.inf'
libwdi:info [wdi_prepare_driver] No .cat file generated (not running Vista or later, or missing elevated privileges)
libwdi:debug [wdi_install_driver] using progress bar mode
libwdi:debug [installer process] got parameter REALTEK_2832U_Device.inf
libwdi:debug [process_message] got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_0BDA&PID_2838\00000001'
libwdi:debug [process_message] got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\Vid_0bda&Pid_2838&Rev_0100'
libwdi:debug [installer process] got user_sid: 'S-1-5-21-16459493-2001771429-3573055957-1005'
libwdi:debug [installer process] using syslog 'C:\WINDOWS\setupapi.log'
libwdi:debug [installer process] sylog reader thread started
libwdi:debug [process_message] switching timeout to infinite
libwdi:debug [installer process] Installing driver for USB\Vid_0bda&Pid_2838&Rev_0100 - please wait...
libwdi:debug [syslog] [2016/12/26 23:06:28 13408.207]
libwdi:debug [syslog] #-198 Command line processed: "C:\Documents and Settings\Jan\usb_driver\installer_x86.exe" "REALTEK_2832U_Device.inf"
libwdi:debug [syslog] #I060 Set selected driver.
libwdi:debug [syslog] #-019 Searching for hardware ID(s): usb\vid_0bda&pid_2838&rev_0100,usb\vid_0bda&pid_2838
libwdi:debug [syslog] #-018 Searching for compatible ID(s): usb\devclass_00&subclass_00&prot_00,usb\devclass_00&subclass_00,usb\devclass_00,usb\composite
libwdi:debug [syslog] #I022 Found "USB\VID_0BDA&PID_2838" in c:\documents and settings\jan\usb_driver\realtek_2832u_device.inf; Device: "REALTEK 2832U Device"; Driver: "REALTEK 2832U Device"; Provider: "libusb.info"; Mfg: "Realtek Semiconductor Corp."; Section name: "USB_Install".
libwdi:debug [syslog] #I087 Driver node not trusted, rank changed from 0x00000001 to 0x00008001.
libwdi:debug [syslog] #I023 Actual install section: [USB_Install]. Rank: 0x00008001. Effective driver date: 06/18/2013.
libwdi:debug [syslog] #-166 Device install function: DIF_SELECTBESTCOMPATDRV.
libwdi:debug [syslog] #I063 Selected driver installs from section [USB_Install] in "c:\documents and settings\jan\usb_driver\realtek_2832u_device.inf".
libwdi:debug [syslog] #I320 Class GUID of device remains: {88BAE032-5A81-49F0-BC3D-A4FF138216D6}.
libwdi:debug [syslog] #I060 Set selected driver.
libwdi:debug [syslog] #I058 Selected best compatible driver.
libwdi:debug [syslog] #-124 Doing copy-only install of "USB\VID_0BDA&PID_2838\00000001".
libwdi:debug [syslog] #E360 An unsigned or incorrectly signed file "c:\documents and settings\jan\usb_driver\realtek_2832u_device.inf" for driver "REALTEK 2832U Device" will be installed (Policy=Ignore). Error 0xe000022f: The third-party INF does not contain digital signature information.
libwdi:debug [syslog] #-024 Copying file "c:\documents and settings\jan\usb_driver\x86\WinUSBCoInstaller2.dll" to "C:\WINDOWS\system32\WinUSBCoInstaller2.dll".
libwdi:debug [syslog] #E360 An unsigned or incorrectly signed file "c:\documents and settings\jan\usb_driver\realtek_2832u_device.inf" for driver "REALTEK 2832U Device" will be installed (Policy=Ignore). Error 0xe000022f: The third-party INF does not contain digital signature information.
libwdi:debug [syslog] #-336 Copying file "c:\documents and settings\jan\usb_driver\x86\WdfCoInstaller01009.dll" to "C:\WINDOWS\system32\WdfCoInstaller01009.dll" via temporary file "C:\WINDOWS\system32\SETA19.tmp".
libwdi:debug [syslog] #E360 An unsigned or incorrectly signed file "c:\documents and settings\jan\usb_driver\realtek_2832u_device.inf" for driver "REALTEK 2832U Device" will be installed (Policy=Ignore). Error 0xe000022f: The third-party INF does not contain digital signature information.
libwdi:debug [syslog] #-166 Device install function: DIF_REGISTER_COINSTALLERS.
libwdi:debug [syslog] #I056 Coinstallers registered.
libwdi:debug [syslog] #-166 Device install function: DIF_INSTALLINTERFACES.
libwdi:debug [syslog] #-011 Installing section [USB_Install.Interfaces] from "c:\documents and settings\jan\usb_driver\realtek_2832u_device.inf".
libwdi:debug [syslog] #I054 Interfaces installed.
libwdi:debug [syslog] #-166 Device install function: DIF_INSTALLDEVICE.
libwdi:debug [syslog] #I123 Doing full install of "USB\VID_0BDA&PID_2838\00000001".
libwdi:debug [syslog] #E360 An unsigned or incorrectly signed file "c:\documents and settings\jan\usb_driver\realtek_2832u_device.inf" for driver "REALTEK 2832U Device" will be installed (Policy=Ignore). Error 0xe000022f: The third-party INF does not contain digital signature information.
libwdi:debug [syslog] #-035 Processing service Add/Delete section [USB_Install.Services].
libwdi:debug [syslog] #E338 Add Service: Binary "C:\WINDOWS\system32\DRIVERS\WinUSB.sys" for service "WinUSB" is not compatible.
libwdi:debug [syslog] #E032 Error 0xe0000234: Driver is not intended for this platform.
libwdi:debug [syslog] #E275 Error while installing services. Error 0xe0000234: Driver is not intended for this platform.
libwdi:debug [syslog] #E122 Device install failed. Error 0xe0000234: Driver is not intended for this platform.
libwdi:debug [syslog] #E157 Default installer failed. Error 0xe0000234: Driver is not intended for this platform.
libwdi:debug [syslog] #W422 Coinstaller 1 (Post Processing) modified status. Error 87: The parameter is incorrect.
libwdi:debug [syslog] #I060 Set selected driver.
libwdi:debug [syslog] #I125 Installing NULL driver for "USB\VID_0BDA&PID_2838\00000001".
libwdi:debug [process_message] switching timeout back to finite
libwdi:debug [installer process] invalid path or hardware ID (C:\Documents and Settings\Jan\usb_driver\REALTEK_2832U_Device.inf)
libwdi:debug [process_message] installer process completed
Driver Installation: FAILED (Invalid parameter)
3 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\Vid_0483&Pid_2016&Rev_0001
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 1.8.1.55
libwdi:debug [wdi_create_list] TcUsb USB device (5): USB\VID_0483&PID_2016\5&20DE6A2D&0&2
libwdi:debug [wdi_create_list] Device description: 'TouchChip Fingerprint Coprocessor'
libwdi:debug [wdi_create_list] Hardware ID: USB\Vid_08e6&Pid_3437&Rev_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_0b&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 4.1.2.7
libwdi:debug [wdi_create_list] GemCCID USB device (6): USB\VID_08E6&PID_3437\6&2D1B5382&0&3
libwdi:debug [wdi_create_list] Device description: 'USB Smart Card Reader'
libwdi:debug [wdi_create_list] Hardware ID: USB\Vid_0bda&Pid_2838&Rev_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\DevClass_00&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driverless USB device (7): USB\VID_0BDA&PID_2838\00000001
libwdi:debug [wdi_create_list] Device description: 'REALTEK 2832U Device'

Toysoft, 2016-12-26:
RTL2841UHIDIR
Dexatek Technology Ltd.
1D19:8202 DK DVBC/T DONGLE

RTL2838UHIDIR
Realtek Semiconductor Corp.
0BDA:2838	ezcap USB 2.0 DVB-T/DAB/FM dongle	E4000

0BDA:2838
RTL2832UUSB USB device (7): USB\VID_0BDA&PID_2838\00000001
Device description: 'REALTEK 2832U Device'

Toysoft, 2016-12-26:
REALTEK 2832U Device

Driver Date : 21.05.2012
Driver Version : 86.1.521.2012
Digital Signer : Microsoft Windows Hardware Compatibility Publ

pbatard, 2016-12-26:
>  but then if I try to upgrade to SDR via your tool, it doesn't work.

Of course it doesn't. What Zadig/libwdi does is replace a __custom and proprietary driver__, specially designed by Realtek or Realtek OEMs to work with their proprietary software, with a __generic driver__ that only provides a limited set of functionality (Microsoft WinUSB)

You should only replace your driver with the WinUSB one if you are planning to use software that advises you to do so, and knowing that, once you do that, you will lose compatibility with any software that requires the proprietary Realtek driver.

Unfortunately, there's no easy solution: you will have to make a choice between being able to watch TV with the proprietary driver, or use the software that relies on the WinUSb driver.

Since this is not a libwdi/Zadig issue, I will close this issue now.

closed 2016-12-26 by pbatard

Toysoft, 2016-12-26:
Sorry, but perhapy I didn't explained me properly, the installation of
the driver doesn't work and doesn't then recognize the dongle anymore
for SDR use, your software tells it failed to install.

Toysoft, 2016-12-26:
I have installed the TV, then your tool find the dongle.
I select the replacement/upgrade of the driver and it fails with the log I posted. So the SDR doesn't work as there is no dongle installed for it, it failed.

Using inf name: REALTEK_2832U_Device.inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] successfully extracted driver files to C:\Documents and Settings\Jan\usb_driver
libwdi:info [wdi_prepare_driver] successfully created 'C:\Documents and Settings\Jan\usb_driver\REALTEK_2832U_Device.inf'
libwdi:info [wdi_prepare_driver] No .cat file generated (not running Vista or later, or missing elevated privileges)
libwdi:debug [wdi_install_driver] using progress bar mode
libwdi:debug [installer process] got parameter REALTEK_2832U_Device.inf
libwdi:debug [process_message] got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_0BDA&PID_2838\00000001'
libwdi:debug [process_message] got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\Vid_0bda&Pid_2838&Rev_0100'
libwdi:debug [installer process] got user_sid: 'S-1-5-21-16459493-2001771429-3573055957-1005'
libwdi:debug [installer process] using syslog 'C:\WINDOWS\setupapi.log'
libwdi:debug [installer process] sylog reader thread started
libwdi:debug [process_message] switching timeout to infinite
libwdi:debug [installer process] Installing driver for USB\Vid_0bda&Pid_2838&Rev_0100 - please wait...
libwdi:debug [syslog] [2016/12/26 23:06:28 13408.207]
libwdi:debug [syslog] #-198 Command line processed: "C:\Documents and Settings\Jan\usb_driver\installer_x86.exe" "REALTEK_2832U_Device.inf"
libwdi:debug [syslog] #I060 Set selected driver.
libwdi:debug [syslog] #-019 Searching for hardware ID(s): usb\vid_0bda&pid_2838&rev_0100,usb\vid_0bda&pid_2838
libwdi:debug [syslog] #-018 Searching for compatible ID(s): usb\devclass_00&subclass_00&prot_00,usb\devclass_00&subclass_00,usb\devclass_00,usb\composite
libwdi:debug [syslog] #I022 Found "USB\VID_0BDA&PID_2838" in c:\documents and settings\jan\usb_driver\realtek_2832u_device.inf; Device: "REALTEK 2832U Device"; Driver: "REALTEK 2832U Device"; Provider: "libusb.info"; Mfg: "Realtek Semiconductor Corp."; Section name: "USB_Install".
libwdi:debug [syslog] #I087 Driver node not trusted, rank changed from 0x00000001 to 0x00008001.
libwdi:debug [syslog] #I023 Actual install section: [USB_Install]. Rank: 0x00008001. Effective driver date: 06/18/2013.
libwdi:debug [syslog] #-166 Device install function: DIF_SELECTBESTCOMPATDRV.
libwdi:debug [syslog] #I063 Selected driver installs from section [USB_Install] in "c:\documents and settings\jan\usb_driver\realtek_2832u_device.inf".
libwdi:debug [syslog] #I320 Class GUID of device remains: {88BAE032-5A81-49F0-BC3D-A4FF138216D6}.
libwdi:debug [syslog] #I060 Set selected driver.
libwdi:debug [syslog] #I058 Selected best compatible driver.
libwdi:debug [syslog] #-124 Doing copy-only install of "USB\VID_0BDA&PID_2838\00000001".
libwdi:debug [syslog] #E360 An unsigned or incorrectly signed file "c:\documents and settings\jan\usb_driver\realtek_2832u_device.inf" for driver "REALTEK 2832U Device" will be installed (Policy=Ignore). Error 0xe000022f: The third-party INF does not contain digital signature information.
libwdi:debug [syslog] #-024 Copying file "c:\documents and settings\jan\usb_driver\x86\WinUSBCoInstaller2.dll" to "C:\WINDOWS\system32\WinUSBCoInstaller2.dll".
libwdi:debug [syslog] #E360 An unsigned or incorrectly signed file "c:\documents and settings\jan\usb_driver\realtek_2832u_device.inf" for driver "REALTEK 2832U Device" will be installed (Policy=Ignore). Error 0xe000022f: The third-party INF does not contain digital signature information.
libwdi:debug [syslog] #-336 Copying file "c:\documents and settings\jan\usb_driver\x86\WdfCoInstaller01009.dll" to "C:\WINDOWS\system32\WdfCoInstaller01009.dll" via temporary file "C:\WINDOWS\system32\SETA19.tmp".
libwdi:debug [syslog] #E360 An unsigned or incorrectly signed file "c:\documents and settings\jan\usb_driver\realtek_2832u_device.inf" for driver "REALTEK 2832U Device" will be installed (Policy=Ignore). Error 0xe000022f: The third-party INF does not contain digital signature information.
libwdi:debug [syslog] #-166 Device install function: DIF_REGISTER_COINSTALLERS.
libwdi:debug [syslog] #I056 Coinstallers registered.
libwdi:debug [syslog] #-166 Device install function: DIF_INSTALLINTERFACES.
libwdi:debug [syslog] #-011 Installing section [USB_Install.Interfaces] from "c:\documents and settings\jan\usb_driver\realtek_2832u_device.inf".
libwdi:debug [syslog] #I054 Interfaces installed.
libwdi:debug [syslog] #-166 Device install function: DIF_INSTALLDEVICE.
libwdi:debug [syslog] #I123 Doing full install of "USB\VID_0BDA&PID_2838\00000001".
libwdi:debug [syslog] #E360 An unsigned or incorrectly signed file "c:\documents and settings\jan\usb_driver\realtek_2832u_device.inf" for driver "REALTEK 2832U Device" will be installed (Policy=Ignore). Error 0xe000022f: The third-party INF does not contain digital signature information.
libwdi:debug [syslog] #-035 Processing service Add/Delete section [USB_Install.Services].
libwdi:debug [syslog] #E338 Add Service: Binary "C:\WINDOWS\system32\DRIVERS\WinUSB.sys" for service "WinUSB" is not compatible.
libwdi:debug [syslog] #E032 Error 0xe0000234: Driver is not intended for this platform.
[... cut ...]
libwdi:debug [wdi_create_list] Driver version: 4.1.2.7
libwdi:debug [wdi_create_list] GemCCID USB device (6): USB\VID_08E6&PID_3437\6&2D1B5382&0&3
libwdi:debug [wdi_create_list] Device description: 'USB Smart Card Reader'
libwdi:debug [wdi_create_list] Hardware ID: USB\Vid_0bda&Pid_2838&Rev_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\DevClass_00&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driverless USB device (7): USB\VID_0BDA&PID_2838\00000001
libwdi:debug [wdi_create_list] Device description: 'REALTEK 2832U Device'

Toysoft, 2016-12-26:
so I will explain step by step.

- I use your new zadig_xp_2.2.exe

- I select all devices, so I can see the REALTEK 2832U Device, with ID
: 0BDA:2838

- It shows RTL2832UUSB (v86.1.521.2

- I press "Replace Driver"
it tells installing driver

then it tells "The driver Installation failed."

Log is below,

Zadig 2.2.689
Windows XP SP3 32 bit
ini file 'zadig.ini' not found - default parameters will be used
default driver set to 'WinUSB'
0 devices found.
3 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\Vid_0483&Pid_2016&Rev_0001
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 1.8.1.55
libwdi:debug [wdi_create_list] TcUsb USB device (5):
USB\VID_0483&PID_2016\5&20DE6A2D&0&2
libwdi:debug [wdi_create_list] Device description: 'TouchChip
Fingerprint Coprocessor'
libwdi:debug [wdi_create_list] Hardware ID: USB\Vid_08e6&Pid_3437&Rev_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_0b&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 4.1.2.7
libwdi:debug [wdi_create_list] GemCCID USB device (6):
USB\VID_08E6&PID_3437\6&2D1B5382&0&3
libwdi:debug [wdi_create_list] Device description: 'USB Smart Card Reader'
libwdi:debug [wdi_create_list] Hardware ID: USB\Vid_0bda&Pid_2838&Rev_0100
libwdi:debug [wdi_create_list] Compatible ID:
USB\DevClass_00&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Upper filter: RTL2832UBDA
libwdi:debug [wdi_create_list] Driver version: 86.1.521.2012
libwdi:debug [wdi_create_list] RTL2832UUSB USB device (7):
USB\VID_0BDA&PID_2838\00000001
libwdi:debug [wdi_create_list] Device description: 'REALTEK 2832U Device'
Using inf name: REALTEK_2832U_Device.inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] successfully extracted driver files to
C:\Documents and Settings\Jan\usb_driver
libwdi:info [wdi_prepare_driver] successfully created 'C:\Documents
and Settings\Jan\usb_driver\REALTEK_2832U_Device.inf'
libwdi:info [wdi_prepare_driver] No .cat file generated (not running
Vista or later, or missing elevated privileges)
libwdi:debug [wdi_install_driver] using progress bar mode
libwdi:debug [installer process] got parameter REALTEK_2832U_Device.inf
libwdi:debug [process_message] got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_0BDA&PID_2838\00000001'
libwdi:debug [process_message] got request for hardware_id
libwdi:debug [installer process] got hardware_id:
'USB\Vid_0bda&Pid_2838&Rev_0100'
libwdi:debug [installer process] got user_sid:
'S-1-5-21-16459493-2001771429-3573055957-1005'
libwdi:debug [installer process] using syslog 'C:\WINDOWS\setupapi.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [process_message] switching timeout to infinite
libwdi:debug [installer process] Installing driver for
USB\Vid_0bda&Pid_2838&Rev_0100 - please wait...
libwdi:debug [syslog] [2016/12/27 00:46:27 13916.208]
libwdi:debug [syslog] #-198 Command line processed: "C:\Documents and
Settings\Jan\usb_driver\installer_x86.exe" "REALTEK_2832U_Device.inf"
libwdi:debug [syslog] #I060 Set selected driver.
[... 80 lines cut ...]
(C:\Documents and Settings\Jan\usb_driver\REALTEK_2832U_Device.inf)
libwdi:debug [process_message] installer process completed
Driver Installation: FAILED (Invalid parameter)
3 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\Vid_0483&Pid_2016&Rev_0001
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 1.8.1.55
libwdi:debug [wdi_create_list] TcUsb USB device (5):
USB\VID_0483&PID_2016\5&20DE6A2D&0&2
libwdi:debug [wdi_create_list] Device description: 'TouchChip
Fingerprint Coprocessor'
libwdi:debug [wdi_create_list] Hardware ID: USB\Vid_08e6&Pid_3437&Rev_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_0b&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 4.1.2.7
libwdi:debug [wdi_create_list] GemCCID USB device (6):
USB\VID_08E6&PID_3437\6&2D1B5382&0&3
libwdi:debug [wdi_create_list] Device description: 'USB Smart Card Reader'
libwdi:debug [wdi_create_list] Hardware ID: USB\Vid_0bda&Pid_2838&Rev_0100
libwdi:debug [wdi_create_list] Compatible ID:
USB\DevClass_00&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driverless USB device (7):
USB\VID_0BDA&PID_2838\00000001
libwdi:debug [wdi_create_list] Device description: 'REALTEK 2832U Device'

2016-12-27 0:41 UTC+01:00, Toysoft <toysofts@gmail.com>:

Toysoft, 2016-12-26:
What do you need, to help you to know what is going wrong ? Thank you very much.

pbatard, 2016-12-27:
`0xe0000234: Driver is not intended for this platform.` seems to be fairly explicit.

I'm not sure how you configured your platform, but it's just not compatible with the WinUSB driver.

Unfortunately, ever since Microsoft ceased to support Windows XP __more than 2 years ago__, I too have ceased to support anything related to that OS. As such, the XP version of Zadig is provided as is. If it works, great, but if it doesn't, I'm afraid that you're on your own, as the only advice I have is to __STRONGLY__ encourage you to upgrade to a version of Windows that is supported by Microsoft.

Toysoft, 2016-12-27:
Dear Pete,

Your software at what I can see is a Must and really good, so at this
point, as I perfectly understand that you don't want to put more time
on an old platform, but as I have only that laptop and that's my only
computer,... and I cannot buy a new one... and it will not support
higher windows...

- Do you have working drivers for the Chipset that I can install that
are compatible ? If the problem is the drivers.

But at what I can read from you, it seems that the WinUSB Driver is
not good, so do you have one that is compatible and working with your
software so I can replace it (keep copy of the old one naturally), to
test if this fix the issue ? It would be really handy and I would be
really thankful to you if you can post it here.

Thank you very much and all the best for the End of the Year, and new
year coming.

Merci !

TS

pbatard, 2016-12-27:
> Do you have working drivers for the Chipset that I can install that are compatible ?

I'm afraid not. That's the very first time I see that error, and I would have expected something like this to only occur on something like Windows XP 64-bit (which was never supported), but it looks like you are using Windows XP 32-bit.

Frankly I have no idea why you are getting the driver incompatibility error, but I'd say (with no ill intention), this is what you get from using a 15+ year old OS, and I have absolutely zero interest in figuring out what is happening.

However, if you do manage to find out why you are getting this error, and how to fix it, feel free to post here, as it may help others (though I will maintain that using XP in this day and age is simply asking for exactly the kind of trouble you are seeing).

Toysoft, 2016-12-27:
Thank you Pete, for sure I will do more testing and FIND the solution.
I perfectly understand your view on this old OS, but that's my task,
to find it.

Also if someone on the group is using WinXP 32bit like me, and you
have that WinUSB file, please post me it so I can swap it and see how
it goes. In the same time, I will do more work and testing to find a
solution, as perhaps replacing the WinUSB could fix the issue (and
could be a good help for all the community using your great and
excellent tool and please keep the WinXP version as you do now, it's
very very nice.)

Merci

TS