pbatard/libwdi #75: Zadig fails to handle Composite IAD device in windows 10

Hoiss, 2017-01-25:
Hi there,

I have a composite device with two interfaces that offers an IAD (Interface Association Descriptor).
For using WinUSB the windows composite parent driver (usbccgp) has to be replaced to get access to all interfaces.
This works as expected using windows 7!!!
But when I use windows 10 with the same USB device I always run into problems: The installation ends successfully, but there is no interface available!
I use zadig_2.2, in windows 10 it looks like this: 
Zadig replaces the usbccgp (v10.0.14393.0) with WinUSB v6.1.7600.16385 --> The driver was installed successfully.
I now have to unplug the device and plug it again, then it's there but I get some warning from windows 10 that the device is not recognized.
In the setupapi.dev.log I also see this warning:
Device not started: Device has problem: 0x0a (CM_PROB_FAILED_START), problem status: 0xc000000e.

When I use libusb I can see the device now, but there's no interface available (Windows 10).
In windows 7 I see both interfaces and can access them as expected.

Does someone know where's the difference from windows 7 to 10 regarding this WinUSB issue?
Thanks for any hints!

Hoiss, 2017-02-01:
* Push * 
Does anyone has ANY idea how to fix this issue?
Is this not the right place to ask about it?

pbatard, 2017-02-01:
Usually `CM_PROB_FAILED_START` indicates an issue with the device firmware (and yeah, the fact that it works with Windows 7 might not be that significant, as I believe Windows 7 is a bit more loose, in terms of how it handles problematic devices compared to Windows 10).

If it were me, I'd use the Windows USB tracing facilities to see what is really happening. I actually [blogged about this](http://pete.akeo.ie/2011/03/troubleshooting-usb-device-development.html) some time ago.

Hoiss, 2017-02-02:
pbatard, thank you for your answer!
I will follow the description in your blog, it seems to be the first hint into the right direction.
If the device really should give a wrong descriptor or similar that was accepted before Win10:
I know there are possible registry entries like "IgnoreHWSerNum" and so on. How could I add such entries to the inf file that's built by zadig? Do you know some good WinUSB examples for such problems?

I'll let you know as soon as I find an issue described in your blog.

Hoiss, 2017-02-06:
Your blog is fantastic !
The device really had an issue with a descriptor value.
It's just so weird because windows7 didn't make any trouble and windows10 simply doesn't accept this device. 
This could break devices out there!!! Is this some deal between MS and HW-vendors?
I also think Windows 10 should make a log entry WHAT exactly was the problem, but it doesn't.
This is really some bad behaviour of Windows10.

pbatard: Thanks! Your blog should be linked on the MS support pages to help.

closed 2017-02-06 by Hoiss

pbatard, 2017-02-06:
Glad to hear you were able to sort it out!