pbatard/libwdi #76: Windows 7 WUP-028 Driver Installation: FAILED (Could not allocate resource)

fundingles, 2017-01-31:
Been trying to get my wiiu gamecube controller adapter to work. Following this guide. https://www.smashladder.com/guides/view/26oz/controller-guide-2-0
I have the official adapter and did everything like it said. 

Zadig 2.2.689
Windows 7 SP1 64-bit
ini file 'zadig.ini' not found - default parameters will be used
default driver set to 'WinUSB'
1 device found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0FD9&PID_005C&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_FF&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driverless USB device (6): USB\VID_0FD9&PID_005C\440D9318C8
libwdi:debug [wdi_create_list] Device description: 'Game Capture HD'
7 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_057E&PID_0337&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_00
libwdi:warning [wdi_create_list] could not read driver version
libwdi:debug [wdi_create_list] HidUsb USB device (2): USB\VID_057E&PID_0337\15/07/2014
libwdi:debug [wdi_create_list] Device description: 'WUP-028'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0C45&PID_0133&REV_0101&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.17514
libwdi:debug [wdi_create_list] HidUsb USB device (4): USB\VID_0C45&PID_0133&MI_00\7&1F7CEABA&0&0000
libwdi:debug [wdi_create_list] Device description: 'USB Keyboard (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0C45&PID_0133&REV_0101&MI_01
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_00
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.17514
libwdi:debug [wdi_create_list] HidUsb USB device (5): USB\VID_0C45&PID_0133&MI_01\7&1F7CEABA&0&0001
libwdi:debug [wdi_create_list] Device description: 'USB Keyboard (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0FD9&PID_005C&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_FF&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driverless USB device (6): USB\VID_0FD9&PID_005C\440D9318C8
libwdi:debug [wdi_create_list] Device description: 'Game Capture HD'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_1AF3&PID_0001&REV_0001
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_02
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.17514
libwdi:debug [wdi_create_list] HidUsb USB device (7): USB\VID_1AF3&PID_0001\6&688A3B8&0&2
libwdi:debug [wdi_create_list] Device description: 'ZOWIE Gaming mouse'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_B58E&PID_9E84&REV_0100&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_01&SubClass_01&Prot_00
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.17514
libwdi:debug [wdi_create_list] usbaudio USB device (11): USB\VID_B58E&PID_9E84&MI_00\7&78CC2BF&0&0000
libwdi:debug [wdi_create_list] Device description: 'Yeti Stereo Microphone (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_B58E&PID_9E84&REV_0100&MI_03
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.17514
libwdi:debug [wdi_create_list] HidUsb USB device (12): USB\VID_B58E&PID_9E84&MI_03\7&78CC2BF&0&0003
libwdi:debug [wdi_create_list] Device description: 'Yeti Stereo Microphone (Interface 3)'
Using inf name: WUP-028.inf
Successfully extracted driver files.
libwdi:info [extract_binaries] successfully extracted driver files to C:\Users\puffs\usb_driver
libwdi:info [wdi_prepare_driver] successfully created 'C:\Users\puffs\usb_driver\WUP-028.inf'
libwdi:info [wdi_prepare_driver] Vista or later detected - creating and self-signing a .cat file...
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\puffs\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\puffs\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'wup-028.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\puffs\usb_driver\wup-028.inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\puffs\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\puffs\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] successfully created file 'C:\Users\puffs\usb_driver\WUP-028.cat'
libwdi:info [RemoveCertFromStore] deleted existing certificate 'CN=USB\VID_057E&PID_0337 (libwdi autogenerated)' from 'Root' store
libwdi:info [RemoveCertFromStore] deleted existing certificate 'CN=USB\VID_057E&PID_0337 (libwdi autogenerated)' from 'TrustedPublisher' store
libwdi:debug [CreateSelfSignedCert] set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] created new key container
libwdi:debug [CreateSelfSignedCert] generated new keypair
libwdi:info [CreateSelfSignedCert] created new self-signed certificate 'CN=USB\VID_057E&PID_0337 (libwdi autogenerated)'
libwdi:debug [SelfSignFile] successfully created certificate 'CN=USB\VID_057E&PID_0337 (libwdi autogenerated)'
[... 546 lines cut ...]
libwdi:debug [wdi_create_list] Driverless USB device (6): USB\VID_0FD9&PID_005C\440D9318C8
libwdi:debug [wdi_create_list] Device description: 'Game Capture HD'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_1AF3&PID_0001&REV_0001
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_02
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.17514
libwdi:debug [wdi_create_list] HidUsb USB device (7): USB\VID_1AF3&PID_0001\6&688A3B8&0&2
libwdi:debug [wdi_create_list] Device description: 'ZOWIE Gaming mouse'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_B58E&PID_9E84&REV_0100&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_01&SubClass_01&Prot_00
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.17514
libwdi:debug [wdi_create_list] usbaudio USB device (11): USB\VID_B58E&PID_9E84&MI_00\7&78CC2BF&0&0000
libwdi:debug [wdi_create_list] Device description: 'Yeti Stereo Microphone (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_B58E&PID_9E84&REV_0100&MI_03
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.17514
libwdi:debug [wdi_create_list] HidUsb USB device (12): USB\VID_B58E&PID_9E84&MI_03\7&78CC2BF&0&0003
libwdi:debug [wdi_create_list] Device description: 'Yeti Stereo Microphone (Interface 3)'
Using inf name: WUP-028.inf
Could not extract files
Driver Installation: FAILED (Could not allocate resource)
libwdi:error [extract_binaries] failed to create file 'C:\Users\puffs\usb_driver\.\installer_x64.exe' ([#00000020] The process cannot access the file because it is being used by another process.)
Using inf name: WUP-028.inf
Could not extract files
Driver Installation: FAILED (Could not allocate resource)
libwdi:error [extract_binaries] failed to create file 'C:\Users\puffs\usb_driver\.\installer_x64.exe' ([#00000020] The process cannot access the file because it is being used by another process.)

pbatard, 2017-02-01:
Looks like something on your machine is preventing access to the driver installer that Zadig executes behind the scenes. Could be a paranoid security solution (which might not like the idea of an application launching another application... which Zadig needs to do, as the installer __must__ have the same bitness as the OS, and Zadig is a 32-bit app, so, if you're running on a 64-bit OS, there's no choice but to invoke a separate 64-bit installer).

But you don't really need the installer since the driver files were extracted anyway. You should be able to just go to Device Manager, right click on your device, select "Update Driver Software" and point to the relevant `.inf` file which you should find under `C:\Users\puffs\usb_driver\`. If you do just that, the driver should install.

fundingles, 2017-02-01:
https://puu.sh/tJRak/62807061dc.png
https://puu.sh/tJRaQ/90a9d6d294.png
https://puu.sh/tJRbF/ae30596568.png

Thanks for the reply, I must be stupid or something its plugged into a working usb port with just the black plug. I can only seem to select the folder and I guess it just detects the .inf file in there? Those pictures are what I see. Maybe it did work but it did said it didnt?

pbatard, 2017-02-01:
Oh yeah, you need to point to the directory rather than the inf.

If Windows didn't manage to install your driver, then the error is not with Zadig but with your Windows platform.

The last section you see in `C:\Windows\INF\setupapi.dev.log` might give you more clues. It contains the most recent driver installation attempt.

fundingles, 2017-02-01:
Wow its an has a lot of text. Ill add the file for the sake of keeping this website readable. Ill add the last three blocks it looks like from today.

The first error thing i see is this.

! sig: Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.

`>>>  [Device Install (DiShowUpdateDevice) - USB\VID_057E&PID_0337\15/07/2014]
>>>  Section start 2017/02/01 12:58:20.332
      cmd: "C:\Windows\system32\mmc.exe" C:\Windows\system32\devmgmt.msc
     dvi: {DIF_UPDATEDRIVER_UI} 12:58:20.333
     dvi:      No class installer for 'WUP-028'
     dvi:      Using exported function 'WdfCoInstaller' in module 'C:\Windows\system32\WdfCoInstaller01011.dll'.
     dvi:      CoInstaller 1 == WdfCoInstaller01011.dll,WdfCoInstaller
     dvi:      Using exported function 'CoDeviceInstall' in module 'C:\Windows\system32\WinUSBCoInstaller2.dll'.
     dvi:      CoInstaller 2 == WinUSBCoInstaller2.dll
     dvi:      CoInstaller 1: Enter 12:58:20.336
     dvi:      CoInstaller 1: Exit
     dvi:      CoInstaller 2: Enter 12:58:20.336
     dvi:      CoInstaller 2: Exit
     dvi:      Default installer: Enter 12:58:20.337
     dvi:      Default installer: Exit
     dvi: {DIF_UPDATEDRIVER_UI - exit(0xe000020e)} 12:58:20.337
     ndv: {Update Driver Software Wizard for USB\VID_057E&PID_0337\15/07/2014}
     ndv: {Update Driver Software Wizard exit(00000000)}
<<<  Section end 2017/02/01 12:58:53.235
<<<  [Exit status: SUCCESS]

>>>  [Device Install (DiShowUpdateDevice) - USB\VID_057E&PID_0337\15/07/2014]
>>>  Section start 2017/02/01 12:58:55.324
      cmd: "C:\Windows\system32\mmc.exe" C:\Windows\system32\devmgmt.msc
     dvi: {DIF_UPDATEDRIVER_UI} 12:58:55.324
     dvi:      No class installer for 'WUP-028'
     dvi:      CoInstaller 1: Enter 12:58:55.325
     dvi:      CoInstaller 1: Exit
     dvi:      CoInstaller 2: Enter 12:58:55.325
     dvi:      CoInstaller 2: Exit
     dvi:      Default installer: Enter 12:58:55.326
     dvi:      Default installer: Exit
     dvi: {DIF_UPDATEDRIVER_UI - exit(0xe000020e)} 12:58:55.326
     ndv: {Update Driver Software Wizard for USB\VID_057E&PID_0337\15/07/2014}
     inf:      Opened INF: 'c:\users\puffs\usb_driver\wup-028.inf' ([strings])
     inf:      {SetupCopyOEMInf: c:\users\puffs\usb_driver\wup-028.inf} 12:59:04.014
!    inf:           Package was already in store: skipping Driver Store add
     inf:           Driver Store location: C:\Windows\System32\DriverStore\FileRepository\wup-028.inf_amd64_neutral_315f96e5f374747f\wup-028.inf
     inf:           Published Inf Path: C:\Windows\INF\oem0.inf
     inf:           OEM source media location: c:\users\puffs\usb_driver\
     inf:      {SetupCopyOEMInf exit (0x00000000)} 12:59:04.025
     dvi:      Searching for hardware ID(s):
     dvi:           usb\vid_057e&pid_0337&rev_0100
     dvi:           usb\vid_057e&pid_0337
     dvi:      Searching for compatible ID(s):
     dvi:           usb\class_03&subclass_00&prot_00
     dvi:           usb\class_03&subclass_00
     dvi:           usb\class_03
     inf:      Opened PNF: 'C:\Windows\System32\DriverStore\FileRepository\wup-028.inf_amd64_neutral_315f96e5f374747f\wup-028.inf' ([strings])
     sig:      {_VERIFY_FILE_SIGNATURE} 12:59:04.029
     sig:           Key      = wup-028.inf
     sig:           FilePath = C:\Windows\System32\DriverStore\FileRepository\wup-028.inf_amd64_neutral_315f96e5f374747f\wup-028.inf
     sig:           Catalog  = C:\Windows\System32\DriverStore\FileRepository\wup-028.inf_amd64_neutral_315f96e5f374747f\WUP-028.cat
!    sig:           Verifying file against specific (valid) catalog failed! (0x800b0109)
!    sig:           Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.
     sig:      {_VERIFY_FILE_SIGNATURE exit(0x800b0109)} 12:59:04.034
     sig:      {_VERIFY_FILE_SIGNATURE} 12:59:04.034
     sig:           Key      = wup-028.inf
     sig:           FilePath = C:\Windows\System32\DriverStore\FileRepository\wup-028.inf_amd64_neutral_315f96e5f374747f\wup-028.inf
     sig:           Catalog  = C:\Windows\System32\DriverStore\FileRepository\wup-028.inf_amd64_neutral_315f96e5f374747f\WUP-028.cat
     sig:           Success: File is signed in Authenticode(tm) catalog.
     sig:           Error 0xe0000242: The publisher of an Authenticode(tm) signed catalog has not yet been established as trusted.
     sig:      {_VERIFY_FILE_SIGNATURE exit(0xe0000242)} 12:59:04.039
     dvi:      Selected driver installs from section [USB_Install] in 'c:\windows\system32\driverstore\filerepository\wup-028.inf_amd64_neutral_315f96e5f374747f\wup-028.inf'.
[... 287 lines cut ...]
!!!  flq:                               Error 2: The system cannot find the file specified.
!    flq:                                    SourceFile   - 'c:\windows\winsxs\amd64\WdfCoInstaller01011.dll'
!    flq:                                    TargetFile   - 'C:\Users\puffs\AppData\Local\Temp\{45b69144-3798-3e4f-5f9f-5d0e98374f3e}\amd64\WdfCoInstaller01011.dll'
     flq:                               {SPFQNOTIFY_COPYERROR}
!!!  sto:                                    Failed to copy file 'c:\windows\winsxs\amd64\WdfCoInstaller01011.dll' to 'C:\Users\puffs\AppData\Local\Temp\{45b69144-3798-3e4f-5f9f-5d0e98374f3e}\amd64\WdfCoInstaller01011.dll'. Error = 0x00000002
     flq:                               {SPFQNOTIFY_COPYERROR - returned 0x00000000}
[... cut ...]
     sto:           {Import Driver Package: exit(0x00000002)} 13:12:45.076
     inf:           Opened INF: 'c:\windows\winsxs\wup-028.inf' ([strings])
!    inf:           Add to Driver Store unsuccessful
!    inf:           Error 2: The system cannot find the file specified.
!!!  inf:           returning failure to SetupCopyOEMInf
     inf:      {SetupCopyOEMInf exit (0x00000002)} 13:12:45.197
     ndv: {Update Driver Software Wizard exit(00000002)}
<<<  Section end 2017/02/01 13:13:02.955
<<<  [Exit status: FAILURE(0x00000002)]

`

fundingles, 2017-02-01:
[asdf.txt](https://github.com/pbatard/libwdi/files/745941/asdf.txt)

sorry forgot to add the file.. i did it again so in the text file its the last block from the log

pbatard, 2017-02-01:
Lines prefixed by a single `!` are warnings (yes, even if it says `Error` in the message - blame Microsoft for that), and the `Error 0x800b0109` is normal (everybody gets that one with Zadig) and in no way prevents driver installation.

The only lines you need to be concerned with are the ones prefixed with 3 `!`.

Seeing that our issue is `Error 2: The system cannot find the file specified.`, I would suggest that you delete the whole `usb_driver` directory and then try Zadig again.

If that still doesn't work, then you may have to contact Microsoft, as the error does not come from Zadig.

pbatard, 2017-02-01:
Hold on, so your actual data is this?:
```
>>>  [Device Install (DiShowUpdateDevice) - USB\VID_057E&PID_0337\15/07/2014]
>>>  Section start 2017/02/01 13:37:37.044
      cmd: "C:\Windows\system32\mmc.exe" C:\Windows\system32\devmgmt.msc
     dvi: {DIF_UPDATEDRIVER_UI} 13:37:37.044
     dvi:      No class installer for 'WUP-028'
     dvi:      CoInstaller 1: Enter 13:37:37.045
     dvi:      CoInstaller 1: Exit
     dvi:      CoInstaller 2: Enter 13:37:37.045
     dvi:      CoInstaller 2: Exit
     dvi:      Default installer: Enter 13:37:37.046
     dvi:      Default installer: Exit
     dvi: {DIF_UPDATEDRIVER_UI - exit(0xe000020e)} 13:37:37.046
     ndv: {Update Driver Software Wizard for USB\VID_057E&PID_0337\15/07/2014}
     inf:      Opened INF: 'c:\users\puffs\usb_driver\wup-028.inf' ([strings])
     inf:      {SetupCopyOEMInf: c:\users\puffs\usb_driver\wup-028.inf} 13:37:38.886
!    inf:           Package was already in store: skipping Driver Store add
     inf:           Driver Store location: C:\Windows\System32\DriverStore\FileRepository\wup-028.inf_amd64_neutral_315f96e5f374747f\wup-028.inf
     inf:           Published Inf Path: C:\Windows\INF\oem0.inf
     inf:           OEM source media location: c:\users\puffs\usb_driver\
     inf:      {SetupCopyOEMInf exit (0x00000000)} 13:37:38.894
     dvi:      Searching for hardware ID(s):
     dvi:           usb\vid_057e&pid_0337&rev_0100
     dvi:           usb\vid_057e&pid_0337
     dvi:      Searching for compatible ID(s):
     dvi:           usb\class_03&subclass_00&prot_00
     dvi:           usb\class_03&subclass_00
     dvi:           usb\class_03
     inf:      Opened PNF: 'C:\Windows\System32\DriverStore\FileRepository\wup-028.inf_amd64_neutral_315f96e5f374747f\wup-028.inf' ([strings])
     sig:      {_VERIFY_FILE_SIGNATURE} 13:37:38.896
     sig:           Key      = wup-028.inf
     sig:           FilePath = C:\Windows\System32\DriverStore\FileRepository\wup-028.inf_amd64_neutral_315f96e5f374747f\wup-028.inf
     sig:           Catalog  = C:\Windows\System32\DriverStore\FileRepository\wup-028.inf_amd64_neutral_315f96e5f374747f\WUP-028.cat
!    sig:           Verifying file against specific (valid) catalog failed! (0x800b0109)
!    sig:           Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.
     sig:      {_VERIFY_FILE_SIGNATURE exit(0x800b0109)} 13:37:38.897
     sig:      {_VERIFY_FILE_SIGNATURE} 13:37:38.897
     sig:           Key      = wup-028.inf
     sig:           FilePath = C:\Windows\System32\DriverStore\FileRepository\wup-028.inf_amd64_neutral_315f96e5f374747f\wup-028.inf
     sig:           Catalog  = C:\Windows\System32\DriverStore\FileRepository\wup-028.inf_amd64_neutral_315f96e5f374747f\WUP-028.cat
     sig:           Success: File is signed in Authenticode(tm) catalog.
     sig:           Error 0xe0000242: The publisher of an Authenticode(tm) signed catalog has not yet been established as trusted.
     sig:      {_VERIFY_FILE_SIGNATURE exit(0xe0000242)} 13:37:38.900
     dvi:      Selected driver installs from section [USB_Install] in 'c:\windows\system32\driverstore\filerepository\wup-028.inf_amd64_neutral_315f96e5f374747f\wup-028.inf'.
     dvi:      Class GUID of device remains: {88bae032-5a81-49f0-bc3d-a4ff138216d6}.
     dvi:      Set selected driver complete.
     dvi:      {Plug and Play Service: Device Install for USB\VID_057E&PID_0337\15/07/2014}
     ump:           Creating Install Process: DrvInst.exe 13:37:38.904
     ndv:           Infpath=C:\Windows\INF\oem0.inf
     ndv:           DriverNodeName=wup-028.inf:libusbDevice_WinUSB.NTamd64:USB_Install:6.1.7600.16385:usb\vid_057e&pid_0337
     ndv:           DriverStorepath=C:\Windows\System32\DriverStore\FileRepository\wup-028.inf_amd64_neutral_315f96e5f374747f\wup-028.inf
     ndv:           Building driver list from driver node strong name...
     dvi:           Searching for hardware ID(s):
     dvi:                usb\vid_057e&pid_0337&rev_0100
     dvi:                usb\vid_057e&pid_0337
     dvi:           Searching for compatible ID(s):
     dvi:                usb\class_03&subclass_00&prot_00
     dvi:                usb\class_03&subclass_00
     dvi:                usb\class_03
     inf:           Opened PNF: 'C:\Windows\System32\DriverStore\FileRepository\wup-028.inf_amd64_neutral_315f96e5f374747f\wup-028.inf' ([strings])
     sig:           {_VERIFY_FILE_SIGNATURE} 13:37:38.913
     sig:                Key      = wup-028.inf
     sig:                FilePath = C:\Windows\System32\DriverStore\FileRepository\wup-028.inf_amd64_neutral_315f96e5f374747f\wup-028.inf
     sig:                Catalog  = C:\Windows\System32\DriverStore\FileRepository\wup-028.inf_amd64_neutral_315f96e5f374747f\WUP-028.cat
!    sig:                Verifying file against specific (valid) catalog failed! (0x800b0109)
!    sig:                Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.
     sig:           {_VERIFY_FILE_SIGNATURE exit(0x800b0109)} 13:37:38.925
     sig:           {_VERIFY_FILE_SIGNATURE} 13:37:38.925
     sig:                Key      = wup-028.inf
[... 181 lines cut ...]
     dvi:                     CoInstaller 1: Enter 13:37:38.981
     inf:                          Opened PNF: 'C:\Windows\INF\oem0.inf' ([strings])
!!!  dvi:                     CoInstaller 1: failed(0x8024200d)!
[... cut ...]
     ump:           Server install process exited with code 0x8024200d 13:37:40.209
     ump:      {Plug and Play Service: Device Install exit(8024200d)}
     ndv: {Update Driver Software Wizard exit(8024200d)}
<<<  Section end 2017/02/01 13:37:41.251
<<<  [Exit status: FAILURE(0x8024200d)]
```

In that case, your issue is the  CoInstaller 1 failure, and you may want to look up ways to fix error `0x8024200d`. Note that the CoInstaller comes from Microsoft, so, again, this is a pure Microsoft issue with your system.

fundingles, 2017-02-01:
Yeah I was messing around with moving the files to other folders and then attempting the driver install. Something from google said winsxs and I tried it from there. That info is from usb_driver like you said to do.

fundingles, 2017-02-01:
FIXED IT I had to reset windows update with this scrip thingy. 
THANK YOU SO MUCH MAN I WOULD OF NEVER FIGURED THIS OUT. 

https://answers.microsoft.com/en-us/windows/forum/windows_10-update/hard-reset-windows-updates/2e26953d-58c4-4704-b627-4a743e87e15c?tm=1447931905306

pbatard, 2017-02-01:
Awesome! I was aware that Windows Updates could play tricks with the driver installation (especially when disabled), but great work on figuring out what needed to be done from the error code.

Hopefully this issue will help others who run into the same problem. I'll also see if there's something I can do in Zadig to make the error and the fix more explicit.

For now, I will close the issue.

closed 2017-02-01 by pbatard