pbatard/libwdi #77: The computer does not see the device!

Nazarych, 2017-02-02:
Problem:
[image]
1
[image]
2
[image]
3
[image]
4
[image]
This happened after I chose this item:
[image]

Nazarych, 2017-02-02:
The device is turned on the USB port

pbatard, 2017-02-02:
You are misunderstanding what you see.

You can __ALWAYS__ install a __WCID__ driver (which is a generic driver part, that is not associated with any device), so the issue is __NOT__ that Zadig is listing your device as a blank entry, against which you can still install the driver, but that you are installing the global WCID driver.

Please be aware that, by default, Zadig does __NOT__ list devices that already have a driver installed, __which is the case for your device__ (because it is listed under the "Ports" category).

So you first need to tell Zadig to show devices that __already__ had a driver installed, by going to _Options_ &rarr; _List All Devices_, and then you will find your USBasp device in the list, and you can install the driver for it.

As I believe I have answered your question, I will close this issue.

closed 2017-02-02 by pbatard

Nazarych, 2017-02-02:
Thanks for the answer! Last question: what driver you need to install flash to Transistir-tester?

pbatard, 2017-02-02:
I don't know. I don't know anything about that device.

Nazarych, 2017-02-02:
Thank you very much!

AlexMaass, 2017-02-07:
Hi, I am trying to get my gamecube controller adapter to work with zadig, I have had the same problem as Nazarych here. I have selected all devices and nothing shows up. This is driving me insane, can you offer me any solutions? Much appreciated!

pbatard, 2017-02-07:
Do you see your device in Windows' Device Manager? What's its name there?

Please post a screenshot from Device Manager showing your device. Unless you see it there, and it's an USB device, it will never show in Zadig.

Nazarych, 2017-02-07:
Options-List all devices  

Choose your programmer and stitch