pbatard/libwdi #78: open a usb with libusb without uninstalling it's Driver from windows

javadhabibi, 2017-02-24:
hi, i am using a FPGA-board with a CY7 (usb communication IC). CY7 has an app(called Control Center) to prog it. after programing, it is ready to transfer data.
in the other hand, i code a Qt app based on libusb. the problem is:
Control Center need windows Driver. but for libusb, it want to uninstall it and replace his Driver(by zadig) .
how can open a usb with libusb without uninstalling it's Driver from windows?
thanks

pbatard, 2017-02-24:
> how can open a usb with libusb without uninstalling it's Driver from windows?

You can't. Because it is a __generic__ library, libusb requires the use of a __generic__ driver, such as WinUSB, libusb0 or libusbK. The driver from Cypress is specific to this device and doesn't offer the generic function calls that libusb requires, so it can't be used with libusb. This means that either you use libusb or you use the Cypress software, but you can't use both without having to switch drivers every time.

The other thing you can do, if you can, and which would help a lot of people, would be to help create an Open Source equivalent for the Cypress proprietary transfer software, so that you can use libusb all the way. Something like this has already been done for Cypress FX chips (see [`ezusb.c`](https://github.com/libusb/libusb/blob/master/examples/ezusb.c#L35-L48) and `fxload.c` in the [libusb example's directory](https://github.com/libusb/libusb/tree/master/examples)).

By the way, this question would be better directed at the libusb project rather than at libwdi, since it relates to libusb usage, so I will close this issue.

closed 2017-02-24 by pbatard

diabolo38, 2017-02-24:
for having work with fx2 and fx3 in the past i remember it was quite  easy to load f/w via control center  that  "re-enum" with a vid/pid that wan't associate to cypress driver (most example  do associate to cypress dev drier) . I guess similar thing could be done with CY7? 
Note that With libusbdk enabled version of libusb you can gain access over cypress native driver and control centre s/w with libusb app without installing any driver but  libdusdk s/w package . You could still used cypress control center after you close the libusdb app (libusdk act as a filter driver when you open the device by libusb). 
Not perfect but it may help  in early dev phase if all you have to load f/w or data is the cypress control centre at first .