pbatard/libwdi #81: Bluetooth problems

Tmax369, 2017-03-16:
so I attempted to replace the drivers on my Bluetooth feature that came with my CPU. everything I see says the same 3 steps. Uninstall, unplug, replug. I can't unplug this, so I'm kinda lost... 

[image]

I tried clicking the Rollback option in properties, and then it went to the driver I had on it before, which still isn't the original one, and I now can not roll back any further. 

[image]

if I could go back one more driver, then I would be back at the original. please help, thx!

pbatard, 2017-03-16:
You need to select uninstall and reboot as many times as necessary. The trick is to insist. You may also try to do it in Safe mode.

There isn't much else you can do I'm afraid and, as [indicated in the FAQ](https://github.com/pbatard/libwdi/wiki/FAQ#Help_Zadig_replaced_the_driver_for_the_wrong_device_How_do_I_restore_it), this is not a really a Zadig/libwdi issue: _If you select the wrong device, and Zadig happen to replace its driver, the responsibility is entirely yours_.

Tmax369, 2017-03-16:
Okay, thank you! Just before clicking the uninstall button, I checked what else I could change. Apparently, i could just update it to the latest version of the original driver and it works. thanks for the advice!

closed 2017-03-16 by Tmax369