pbatard/libwdi #83: Driver sing @ windows 10 anniversary problem.

qquiderr, 2017-04-22:
Hi i have win 10 anniversary @ latest developer prewiev but, zdiag not working, driver signature fail. Here is log from zdiag.

`Using inf name: Bulk-In_Interface_(Interface_0).inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] successfully extracted driver files to C:\Users\Qquiderr Aoiro\usb_driver
libwdi:info [wdi_prepare_driver] successfully created 'C:\Users\Qquiderr Aoiro\usb_driver\Bulk-In_Interface_(Interface_0).inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Qquiderr Aoiro\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Qquiderr Aoiro\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'bulk-in_interface_(interface_0).inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Qquiderr Aoiro\usb_driver\bulk-in_interface_(interface_0).inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Qquiderr Aoiro\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Qquiderr Aoiro\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] successfully created file 'C:\Users\Qquiderr Aoiro\usb_driver\Bulk-In_Interface_(Interface_0).cat'
libwdi:debug [CreateSelfSignedCert] set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] acquired existing key container
libwdi:debug [CreateSelfSignedCert] could not generate keypair: [#80090005] Chybn� data.
libwdi:warning [wdi_prepare_driver] could not sign cat file
libwdi:debug [wdi_install_driver] using progress bar mode
libwdi:debug [installer process] got parameter Bulk-In_Interface_(Interface_0).inf
libwdi:debug [process_message] got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_0BDA&PID_2838&MI_00\6&38A180A5&7&0000'
libwdi:debug [process_message] got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\VID_0BDA&PID_2838&REV_0100&MI_00'
libwdi:debug [installer process] got user_sid: 'S-1-5-21-2581044359-1499900301-3751219286-1001'
libwdi:debug [installer process] using syslog 'C:\WINDOWS\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [process_message] switching timeout to infinite
libwdi:debug [installer process] Installing driver for USB\VID_0BDA&PID_2838&REV_0100&MI_00 - please wait...
libwdi:debug [syslog] <Garbled data>
libwdi:debug [syslog] >>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - USB\VID_0BDA&PID_2838&REV_0100&MI_00]
libwdi:debug [syslog] >>>  Section start 2017/04/22 08:51:00.329
libwdi:debug [syslog]       cmd: "C:\Users\Qquiderr Aoiro\usb_driver\installer_x64.exe" "Bulk-In_Interface_(Interface_0).inf"
libwdi:debug [syslog]      ndv: INF path: C:\Users\Qquiderr Aoiro\usb_driver\Bulk-In_Interface_(Interface_0).inf
libwdi:debug [syslog]      ndv: Install flags: 0x00000001
libwdi:debug [syslog]      ndv: {Update Device Driver - USB\VID_0BDA&PID_2838&MI_00\6&38A180A5&7&0000}
libwdi:debug [syslog]      ndv:      Search options: 0x00000080
libwdi:debug [syslog]      ndv:      Searching single INF 'C:\Users\Qquiderr Aoiro\usb_driver\Bulk-In_Interface_(Interface_0).inf'
libwdi:debug [syslog]      dvi:      {Build Driver List} 08:51:00.361
libwdi:debug [syslog]      dvi:           Searching for hardware ID(s):
libwdi:debug [syslog]      dvi:                usb\vid_0bda&pid_2838&rev_0100&mi_00
libwdi:debug [syslog]      dvi:                usb\vid_0bda&pid_2838&mi_00
libwdi:debug [syslog]      dvi:           Searching for compatible ID(s):
libwdi:debug [syslog]      dvi:                usb\class_ff&subclass_ff&prot_ff
libwdi:debug [syslog]      dvi:                usb\class_ff&subclass_ff
libwdi:debug [syslog]      dvi:                usb\class_ff
libwdi:debug [syslog]      sig:           {_VERIFY_FILE_SIGNATURE} 08:51:00.371
libwdi:debug [syslog]      sig:                Key      = bulk-in_interface_(interface_0).inf
libwdi:debug [syslog]      sig:                FilePath = c:\users\qquiderr aoiro\usb_driver\bulk-in_interface_(interface_0).inf
libwdi:debug [syslog]      sig:                Catalog  = c:\users\qquiderr aoiro\usb_driver\Bulk-In_Interface_(Interface_0).cat
libwdi:debug [syslog] !    sig:                Verifying file against specific (valid) catalog failed! (0x800b0100)
libwdi:debug [syslog]      sig:           {_VERIFY_FILE_SIGNATURE exit(0x800b0100)} 08:51:00.411
libwdi:debug [syslog]      sig:           {_VERIFY_FILE_SIGNATURE} 08:51:00.411
libwdi:debug [syslog]      sig:                Key      = bulk-in_interface_(interface_0).inf
libwdi:debug [syslog]      sig:                FilePath = c:\users\qquiderr aoiro\usb_driver\bulk-in_interface_(interface_0).inf
libwdi:debug [syslog]      sig:                Catalog  = c:\users\qquiderr aoiro\usb_driver\Bulk-In_Interface_(Interface_0).cat
libwdi:debug [syslog] !    sig:                Verifying file against specific Authenticode(tm) catalog failed! (0x800b0100)
libwdi:debug [syslog]      sig:           {_VERIFY_FILE_SIGNATURE exit(0x800b0100)} 08:51:00.417
libwdi:debug [syslog]      dvi:           Created Driver Node:
libwdi:debug [syslog]      dvi:                HardwareID   - USB\VID_0BDA&PID_2838&MI_00
libwdi:debug [syslog]      dvi:                InfName      - c:\users\qquiderr aoiro\usb_driver\bulk-in_interface_(interface_0).inf
libwdi:debug [syslog]      dvi:                DevDesc      - Bulk-In, Interface (Interface 0)
libwdi:debug [syslog]      dvi:                Section      - USB_Install
libwdi:debug [syslog]      dvi:                Rank         - 0x80ff0001
[... 112 lines cut ...]
libwdi:debug [wdi_create_list] Driver version: 10.0.16179.1000
libwdi:debug [wdi_create_list] USBSTOR USB device (19): USB\VID_058F&PID_6362\058F63626476
libwdi:debug [wdi_create_list] Device description: 'Mass Storage Device'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0BDA&PID_58B0&REV_5801&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_0e&SubClass_03&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.16179.1000
libwdi:debug [wdi_create_list] usbvideo USB device (20): USB\VID_0BDA&PID_58B0&MI_00\6&25249C4A&0&0000
libwdi:debug [wdi_create_list] Device description: 'FULL HD 1080P Webcam (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0BDA&PID_2838&REV_0100&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_ff&SubClass_ff&Prot_ff
libwdi:debug [wdi_create_list] Upper filter: RTL2832UBDA
libwdi:debug [wdi_create_list] Driver version: 64.1.802.2010
libwdi:debug [wdi_create_list] RTL2832UUSB USB device (21): USB\VID_0BDA&PID_2838&MI_00\6&38A180A5&7&0000
libwdi:debug [wdi_create_list] Device description: 'Bulk-In, Interface (Interface 0)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_09DA&PID_7143&REV_9900&MI_02
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 10.0.16179.1000
libwdi:debug [wdi_create_list] HidUsb USB device (23): USB\VID_09DA&PID_7143&MI_02\6&3A0EDCFE&0&0002
libwdi:debug [wdi_create_list] Device description: 'USB Device (Interface 2)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_2013&PID_024F&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_FF&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driver version: 5.2010.901.0
libwdi:debug [wdi_create_list] USB28xxBGA USB device (25): USB\VID_2013&PID_024F\00000010VK8U
libwdi:debug [wdi_create_list] Device description: 'PCTV 290e'
`

pbatard, 2017-04-22:
1. Windows Insider builds are __not__ supported, because Microsoft tends to break things in these builds, __A LOT__. If you want support, you need to use a retail release of Windows 10, and not an Insider build.
2. Even then, I'm not seeing any issue with the Windows Insider (or Anniversary, or Creators Update for that matter). See the log below.
3. __DON'T EVER__ truncate the log. You are missing quite a few lines at the start, which means I can't tell what version of Zadig you are using, or which build of Windows. I am therefore assuming that you are using build 16176, which is __unsupported__.
4. Your issue has to do with this:
```
libwdi:debug [CreateSelfSignedCert] acquired existing key container
libwdi:debug [CreateSelfSignedCert] could not generate keypair: [#80090005] Chybn� data.
```
I'm not sure why you still have an existing key container, but by the looks of it, this is something that is specific to your platform or something that Microsoft broke yet again in an Insider build.

Unless you experience the same problem on a non Insider version of Windows, and considering that nobody else is reporting this problem (many people are using Zadig on Windows 10 without issues, including on Anniversary and Creators Update), I'm going to have to dismiss it as specific to your environment.

```
Zadig 2.3.701
Windows 10 64-bit (Build 16176)
ini file 'zadig.ini' not found - default parameters will be used
default driver set to 'WinUSB'
1 device found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_045E&PID_0289&REV_0121
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_58&SubClass_42&Prot_00
libwdi:debug [wdi_create_list] Driverless USB device (2): USB\VID_045E&PID_0289\6&6937646&0&1
libwdi:debug [wdi_create_list] could not read device description for 2: [#0000000D] The data is invalid.
libwdi:debug [wdi_create_list] Device description: 'Unknown Device #1'
Using inf name: XBox_Controller.inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] successfully extracted driver files to C:\Users\nil\usb_driver
libwdi:info [wdi_prepare_driver] successfully created 'C:\Users\nil\usb_driver\XBox_Controller.inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\nil\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\nil\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\nil\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\nil\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'xbox_controller.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\nil\usb_driver\xbox_controller.inf'
libwdi:info [CreateCat] successfully created file 'C:\Users\nil\usb_driver\XBox_Controller.cat'
libwdi:info [RemoveCertFromStore] deleted existing certificate 'CN=USB\VID_045E&PID_0289 (libwdi autogenerated)' from 'Root' store
libwdi:info [RemoveCertFromStore] deleted existing certificate 'CN=USB\VID_045E&PID_0289 (libwdi autogenerated)' from 'TrustedPublisher' store
libwdi:debug [CreateSelfSignedCert] set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] created new key container
libwdi:debug [CreateSelfSignedCert] generated new keypair
libwdi:info [CreateSelfSignedCert] created new self-signed certificate 'CN=USB\VID_045E&PID_0289 (libwdi autogenerated)'
libwdi:debug [SelfSignFile] successfully created certificate 'CN=USB\VID_045E&PID_0289 (libwdi autogenerated)'
libwdi:info [SelfSignFile] added certificate 'CN=USB\VID_045E&PID_0289 (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
libwdi:info [SelfSignFile] successfully signed file 'C:\Users\nil\usb_driver\XBox_Controller.cat'
libwdi:info [SelfSignFile] successfully deleted private key
libwdi:debug [wdi_install_driver] using progress bar mode
libwdi:debug [installer process] got parameter XBox_Controller.inf
libwdi:debug [process_message] got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_045E&PID_0289\6&6937646&0&1'
libwdi:debug [process_message] got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\VID_045E&PID_0289&REV_0121'
libwdi:debug [installer process] got user_sid: 'S-1-5-21-306796962-922541030-1774233152-1001'
libwdi:debug [installer process] using syslog 'C:\WINDOWS\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [process_message] switching timeout to infinite
libwdi:debug [installer process] Installing driver for USB\VID_045E&PID_0289&REV_0121 - please wait...
libwdi:debug [syslog] <Garbled data>
libwdi:debug [syslog] >>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - USB\VID_045E&PID_0289&REV_0121]
libwdi:debug [syslog] >>>  Section start 2017/04/22 17:07:24.871
libwdi:debug [syslog]       cmd: "C:\Users\nil\usb_driver\installer_x64.exe" "XBox_Controller.inf"
libwdi:debug [syslog]      ndv: INF path: C:\Users\nil\usb_driver\XBox_Controller.inf
[... cut ...]
libwdi:debug [wdi_create_list] Device description: 'USB Productivity Option Keyboard( has the hub in # 1 ) (Interface 1)'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_04B3&PID_301B&REV_0100&MI_00
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 10.0.16176.1000
libwdi:debug [wdi_create_list] HidUsb USB device (9): USB\VID_04B3&PID_301B&MI_00\7&28D7CA8&0&0000
libwdi:debug [wdi_create_list] Device description: 'USB Productivity Option Keyboard( has the hub in # 1 ) (Interface 0)'
```

pbatard, 2017-05-04:
Since I cannot replicate the issue, I will close it.

closed 2017-05-04 by pbatard