pbatard/libwdi #84: Missing build64.h

akobberup, 2017-04-28:
I have folowed your guide here (https://github.com/pbatard/libwdi/wiki/Compiling-and-debugging-libwdi-or-Zadig) to try to build libwid, but i cannot build the solution, as the build64.h referred on line 35 in embedder.h cannot be found.
If i search the repository i find the file is in the gitignore (https://github.com/pbatard/libwdi/blob/b18d73f9e9f46b04da51c293d85e0fa2d77c9254/.gitignore) and so it is probably not committed.

I had to retarget the solution to SDK Version 10.0.15063.0 to get many other includes working, but other than that i guess i should be able to get the wheels turning if the build64.h were found?

pbatard, 2017-04-28:
`build64.h` is an autogenerated file (which is why it is in gitignore).

It is generated by the pre-build event of the `detect_64build` project.

If you don't get that file generated, then you probably broke the solution when you altered it, as the `embedder` project has a clear dependency on `detect_64build`.

akobberup, 2017-05-01:
Thank you for the advice - i will revert my changes and try to get it going on the VS2015 components :)

closed 2017-05-01 by akobberup

pbatard, 2017-05-01:
Just for your information, it looks like there can be an issue with the `build64.h` detection if your project path contains a space. This has just been fixed in libwdi.