pbatard/libwdi #88: Driver Installation: FAILED (Resource already exists)

s-williams, 2017-06-23:
Hey I have a problem. I am trying to install a driver with this USB which appears as "Unknown Device". Trying any driver results in the error message "Driver Installation: FAILED (Resource already exists)". Trying to uninstall the device through the device manager and then installing a driver through Zadig does cause the installation to succeed but the USB fails to appear in the device manager or even the Zadig listing again so I'm unsure if it really did work.

pbatard, 2017-06-23:
Please delete your driver (follow the same steps as decribed [here](https://github.com/pbatard/libwdi/wiki/FAQ#Help_Zadig_replaced_the_driver_for_the_wrong_device_How_do_I_restore_it)) and then try to reinstall the driver in Advanced Mode (_Options_ &rarr; _Advanced Mode_) and copy/paste your full log.

closed 2017-07-16 by pbatard

charlhug, 2018-02-26:
I have a similar issue, using a Samsung Galaxy Tab S 10.5 (chagallwifi) but I can't delete the driver because it looks like Windows 10 hasn't assigned one, being unable to recognize the device. The first time I plugged it in it showed up as "Gadget Serial" and I was able to start the flash process, but it seems like the process failed (and changed the USB device ID to be unrecognizable to Windows?) and I'm stuck in this state now.

I tried installing Samsung's USB drivers, restarting the computer, and using a different computer all with the same result. Here's my log:

Using inf name: Unknown_USB_Device_(Device_Descriptor_Request_Failed).inf
Succesfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] successfully extracted driver files to C:\usb_driver
libwdi:info [wdi_prepare_driver] succesfully created 'C:\usb_driver\Unknown_USB_Device_(Device_Descriptor_Request_Failed).inf'
libwdi:info [wdi_prepare_driver] Vista or later detected - creating and self-signing a .cat file...
libwdi:info [ScanDirAndHash] added hash for 'C:\usb_driver\amd64\wdfcoinstaller01009.dll'
libwdi:info [ScanDirAndHash] added hash for 'C:\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:info [ScanDirAndHash] added hash for 'C:\usb_driver\unknown_usb_device_(device_descriptor_request_failed).inf'
libwdi:info [ScanDirAndHash] added hash for 'C:\usb_driver\x86\wdfcoinstaller01009.dll'
libwdi:info [ScanDirAndHash] added hash for 'C:\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] successfully created file 'C:\usb_driver\Unknown_USB_Device_(Device_Descriptor_Request_Failed).cat'
libwdi:info [RemoveCertFromStore] deleted existing certificate 'CN=USB\VID_0000&PID_0002 (libwdi autogenerated)' from 'Root' store
libwdi:info [RemoveCertFromStore] deleted existing certificate 'CN=USB\VID_0000&PID_0002 (libwdi autogenerated)' from 'TrustedPublisher' store
libwdi:info [CreateSelfSignedCert] created new self-signed certificate 'CN=USB\VID_0000&PID_0002 (libwdi autogenerated)'
libwdi:info [SelfSignFile] added certificate 'CN=USB\VID_0000&PID_0002 (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
libwdi:info [SelfSignFile] successfully signed file 'C:\usb_driver\Unknown_USB_Device_(Device_Descriptor_Request_Failed).cat'
libwdi:info [SelfSignFile] successfully deleted private key
Driver Installation: FAILED (Resource already exists)
5 devices found.

charlhug, 2018-02-26:
After leaving download mode (hold home, vol. up, power) and restarting everything looks OK now and zadig.exe was successful in installing the driver on the second try, so I'd recommend that to anyone else who has a similar issue and is led here by searches.

canslgb1, 2025-12-28:
For Snapdragon based devices such as the SurfacePro 11 Elite, follow the instructions here:  https://armsdr.com/drivers.html

I spent hours following every advice on the web before finding this resource and finally solving the problem.  I hope this resource helps save someone else the trouble I went through!