pbatard/libwdi #91: Selected device in the dropdown menu resets when hardware changes

DRSDavidSoft, 2017-07-23:
### Summary
On USB plugged-in or plugged-out, the target device dropdown menu will be reset to reflect the hardware changes, which is useful. However, in doing so the selected device will be set to the first item in the menu, which is undesirable. 

### Description
This is really annoying IMO. Let me explain situation:

1. Device `usb_example` is selected.
2. I plug in my flash-drive to load a program.
3. The target device will be changed to `usb_mouse` instead of the the device I wanted to perform driver installation.
4. I don't realise that the app has changed the target device on its own.
5. I press the `Replace Driver` button, because I want to replace the driver for `usb_example`.
6. The target device won't load the new driver. Instead, my mouse stops working.
7. I realise now that the target USB device was changed. But it's too late now.
8. Proceed to use keyboard shortcuts or a remote management tool to Uninstall the wrong driver from the `usb_mouse` device.

### Proposal
Again, IMO, this is an easy fix. Here's an abstract on how should Zadig handle this situation:
1. On item change, the app should remember the information of the current item.
2. On hardware activity, the app repopulates the list with the current list of devices.
3. After the item has been populated, the item with the matching one in the memory will be searched:

- If the device is still present, the item will be selected (e.g. the dropdown menu's index will be set to the item which matches the VID/PID previously selected by user.)
- If there is no item with the previously matched information, it will THEN set the dropdown menu's selected index to zero.

Thanks for your time developing this amazing usefull app.
I hope this is just as logical to you that is to me.

dnguye86, 2017-07-28:
Agreed with you David. This tool is amazing for what it does and the only thing that I found to be a little bit annoying is this. 

I also accidentally replaced my keyboard driver but I was able to reinstall the driver for it.

Cheers,

pbatard, 2017-07-29:
I don't plan to do that. Installing a device driver is not a benign operation, as such you shouldn't be plugging devices around when you are installing a driver.

If you don't pay attention to what you are doing, then you pay the price. That's all there is to it. And there's probably a valuable lesson to be learned there.

Also, in case you haven't noticed, Zadig does produce an extra warning for HID (keyboards, mice) or Mass Storage (USB HDDs) devices, telling you that you are about to replace a system driver and asking you if you really want to proceed:

[image]

You just don't click yes to every prompt you get, especially if you are replacing a driver.

As such I will reject this suggestion.

closed 2017-07-29 by pbatard