pbatard/libwdi #92: Linux persistent data support

alexfoodforester, 2017-07-25:
Your FAQ answer regarding Linux persistent data support states:-

   "Once you boot the OS, you should have everything you need to create some space for persistent data, so there's no real need for Rufus to do that."

Since Rufus has used the entire USB disk, and gparted only sees a small ~400Kb partition and a lot of 'unallocated' space', how does one create a new partition for any Linux persistent data support?

There should be approx. 4-5GB available on the USB disk I am using, for Linux persistent data, but no way to create the required persistence partition to hold the persistence.conf file and installed apps.

Please would you add some sort of instructions of how to do this from inside the now booted Linux OS, to your very comprehensive FAQ.

Thanking you in anticipation.

Alex.

pbatard, 2017-07-25:
You usually don't create a partition. You create a file that acts as an extended partition. Or you can use Linux tools to resize your partition. Of course, you will complain that FAT32 may limit you, but Rufus does provide __experimental__ support for Linux on NTFS, which most recent distros seem to handle.

I'm afraid that there are a few too many ways to achieve persistent data support, which may vary from one distro to the next, and the last thing I want is have to edit the FAQ with whatever is the latest recommended way to create persistent support for Distro X, as I'm going to have to spend my time maintaining that entry and providing support for Linux beginners.

This is a pure Linux topic, and better in the documentation for each distro, or on Linux forums. As such, I am not planning to provide any more documentation of this topic in the Rufus FAQ, sorry.

pbatard, 2017-07-25:
By the way, why was this issue logged under libwdi? This has nothing to do with libwdi? Please open issues against the actual project you are using.

closed 2017-07-25 by pbatard