pbatard/libwdi #93: WinUSB driver installation from Windows Update not working on Win7

Gamadril, 2017-08-03:
Maybe you should update the WIKI regarding WCID devices:
"On Windows 8 or later, it is native to the system which means there is no need to go online for the driver to be installed, whereas, **for Vista and Windows 7, it will be obtained from the internet, through Windows Update.**"

It seems that Windows Update on Windows 7 does not provide the WinUSB driver any more. It took me a lot of time to find it out. https://social.msdn.microsoft.com/Forums/en-US/47ad1431-7869-4028-9831-f388b8cfbe24/winusb-compatible-device-doesnt-get-drivers-from-windows-update?forum=wdk

pbatard, 2017-08-03:
I think this issue is specific to your environment. I tested on a fresh Win 7 SP1 VM (with all the latest updates) with a WCID device that had never been plugged, and I got no issues with Windows downloading and installing the driver from Windows Update:
[image]
and then:
[image]

Note that it is possible to tell Windows __not__ to download drivers from Windows Update, in which case, obviously, a WCID device will not be able to see a driver installed. But as far as I know, this is not the default setting, and even if you disabled driver lookup from Windows Update, you should get a prompt from Windows 7 asking you if you want to re-enable WU downloads.

What I suspect is that you somehow disabled your Windows Update service, or one of the components that is necessary for Windows 7 to fetch the WinUSB driver. At any rate, this looks like a purely environmental problem, and I don't think there's anything to update on the Wiki.

Gamadril, 2017-08-03:
Thanks for testing. I actually also test on a fresh VM and on real Hardware. Windows is set up to get the drivers from WU automatically. I also see the the WU search in the popup. The only difference is that I have a German Win7 version.

closed 2017-08-03 by Gamadril

Gamadril, 2017-08-04:
Installed a second Win7 VM (English this time) and guess what happened?

[image]

I'm really confused why it doesn't work for German Windows Version..

pbatard, 2017-08-04:
I seriously doubt it has anything to do with the locale. You should look in other differences, else I'm pretty sure you're gonna waste a lot of time for nothing.

Gamadril, 2017-08-04:
compared the list of optional updates on both VMs, installed those that were offered on German version, but not in the list of English version. Some of them were fixes for some usb problems, however that doesn't help. No idea what I should compare else. Both are Win7 x64 professional versions with same settings for windows update and driver update over WU.

pbatard, 2017-08-04:
Compare the network traces when looking for the WinUSB driver as well as the `C:\Windows\INF\setupapi.dev.log` files from two __clean__ Windows 7 SP1 installations, and if you can't spot the differences, then take it to Microsoft, as this would be a pure Windows issue.

Gamadril, 2017-08-04:
Here is the result: 2 VMs parallel installation, same steps, same WU settings. All available updates installed on both VMs.
Connected Device to English VM:
```
     dvi:      Searching for compatible ID(s):
     dvi:           usb\ms_comp_winusb
     dvi:           usb\class_ff&subclass_00&prot_00
     dvi:           usb\class_ff&subclass_00
     dvi:           usb\class_ff
     cpy:      Policy is set to make all digital signatures equal.
     dvi:      Enumerating INFs from path list 'C:\Windows\inf'
     inf:      Searched 0 potential matches in published INF directory
     inf:      Searched 35 INFs in directory: 'C:\Windows\inf'
     dvi: {Build Driver List - exit(0x00000000)} 22:53:31.780
     ndv: Selecting best match from Driver Store (including Device Path)...
     dvi: {DIF_SELECTBESTCOMPATDRV} 22:53:31.781
     dvi:      No class installer for 'TEST-DEVICE'
     dvi:      No CoInstallers found
     dvi:      Default installer: Enter 22:53:31.785
     dvi:           {Select Best Driver}
!    dvi:                Selecting driver failed(0xe0000228)
     dvi:           {Select Best Driver - exit(0xe0000228)}
!    dvi:      Default installer: failed!
!    dvi:      Error 0xe0000228: There are no compatible drivers for this device.
     dvi: {DIF_SELECTBESTCOMPATDRV - exit(0xe0000228)} 22:53:31.787
     ndv: Searching Windows Update for drivers... 22:53:31.801
     ndv: Waiting for previous WU search to complete. 22:53:31.802
     ndv: Acquired WU search serialization mutex. 22:53:42.922
     ndv: About to release WU search serialization mutex. 22:54:03.101
     ndv: Found driver on Windows Update, downloading - 0.0 MB... 22:54:03.101
```

=> everything OK, driver found and installed

German VM:
```
     dvi:      Searching for compatible ID(s):
     dvi:           usb\ms_comp_winusb
     dvi:           usb\class_ff&subclass_00&prot_00
     dvi:           usb\class_ff&subclass_00
     dvi:           usb\class_ff
     cpy:      Policy is set to make all digital signatures equal.
     dvi:      Enumerating INFs from path list 'C:\Windows\inf'
     inf:      Searched 0 potential matches in published INF directory
     inf:      Searched 35 INFs in directory: 'C:\Windows\inf'
     dvi: {Build Driver List - exit(0x00000000)} 23:00:35.087
     ndv: Selecting best match from Driver Store (including Device Path)...
     dvi: {DIF_SELECTBESTCOMPATDRV} 23:00:35.087
     dvi:      No class installer for 'TEST-DEVICE'
     dvi:      No CoInstallers found
     dvi:      Default installer: Enter 23:00:35.087
     dvi:           {Select Best Driver}
!    dvi:                Selecting driver failed(0xe0000228)
     dvi:           {Select Best Driver - exit(0xe0000228)}
!    dvi:      Default installer: failed!
     dvi: {DIF_SELECTBESTCOMPATDRV - exit(0xe0000228)} 23:00:35.087
     ndv: Searching Windows Update for drivers... 23:00:35.103
     ndv: Waiting for previous WU search to complete. 23:00:35.103
     ndv: Acquired WU search serialization mutex. 23:00:43.824
     ndv: About to release WU search serialization mutex. 23:01:03.670
     ndv: No driver found on Windows Update. 23:01:03.670
```

Driver not found.
So yes, you are right - Windows issue

dani1024, 2020-12-08:
I had the same problem with a Windows 7 64 bits ultimate edition (with English UI). Windows update didn't detect the suitable driver for my greaseweazle f7+.
Solved it by forcing installation of CDC driver using Zadig