pbatard/libwdi #96: WinUsb per nfc reader?

userDevop123, 2017-08-26:
I use nfcpy to read nfc tag with reader via usb
I need install WinUSB according to  https://nfcpy.readthedocs.io/en/latest/topics/get-started.html#installation

1)On zadig i saw i need to select my device and choose WinUSB
Do i need to install WinUSB per card reader?
Why zadig can't install WinUSB for all the card reader?

2) what is exactly WinUSB ?what is the diffrent between that and libnfc?

pbatard, 2017-08-26:
> Do i need to install WinUSB per card reader?

If you have multiple card readers, then yes, you will need to install the WinUSB driver for each device.

> Why zadig can't install WinUSB for all the card reader?

Because this is how Windows driver installation works. The situation where you want all devices of a specific type to get the same driver is quite rare. In most cases, you only want to replace the drive for a single device.

> what is exactly WinUSB ?what is the diffrent between that and libnfc?

WinUSB is a windows __driver__. This is the part that helps the OS provide an interface between a user application or library and the device (because there are so many different devices that Microsoft cannot provide driver for them all). libnfc is a __library__, which you can consider as a module that is part of a user application. Because user libraries do not have the capabilities to interface directly with a device, a driver must be installed before they can communicate with it.

pbatard, 2017-09-02:
I believe this question has been answered. I will now close the issue.

closed 2017-09-02 by pbatard