pbatard/libwdi #97: Distribute zadiC as well

ddavidebor, 2017-09-02:
It would be nice if in the future you were to distribute zadic as well as zadig, we've found it very useful to distribute inside our application as an easy way to install drivers.

We've forked and updated the appveyor config to upload to a s3 bucket
http://fermiumlabs-libwdi-builds.s3.amazonaws.com/index.html
https://github.com/fermiumlabs/libwdi

pbatard, 2017-09-02:
From the reports I get, I think more people would be interested in wdi-simple rather than zadic, and if I were to distribute something else besides Zadig, I'd rather distribute wdi-simple. As a matter of fact, I'm thinking of removing zadic altogether from the codebase, as I think it duplicates a lot of what wdi-simple now does, and most of the imrpovement requests I receive are for wdi-simple.

Is there something that you are using zadic for that you don't think you'd be able to do with wdi-simple?

> We've forked and updated the appveyor config to upload to a s3 bucket

Sorry if that comes across the wrong way, but please, please, please, __NEVER__ distribute win32 binaries that require elevation __UNLESS__ they are digitally signed (which your S3 binaries aren't).
By implying that it is fine to run unsigned elevated Windows binaries, you are doing a major disservice to Windows users, as you are continuing to propagate the very damaging idea that having a minimum level of accountability and traceability, with regards to software that may easily infect a whole computer, is accessory.

Because your the binaries you put online are not signed, for all I know, you could be a malware author, that's trying to use the (hopefully) good reputation of libwdi and the fact that it's open source to trick people, such as corporate users, who don't want to have to recompile their own binaries, into running __elevated__ software that has been modified to inject software.

With a signed binary, at the very least, security conscious users can have some confidence that the author of the software has had a minimum level of vetting and can hopefully be be tracked (with the ability to run the infected software disabled) in case it turns out that the binaries they distributed contains malware.

Unfortunately, because that would require the continuous build process to have access to the developer's private signing key, which is not something that should ever be shared, it is not possible to have an automated process upload trustworthy signed binaries... which is the main reason why, even though I use appveyor on libwdi, I will not allow generated artefacts from being downloaded, and encourage people who cloned libwdi to do the same.

As a result, I have removed your download links.

But to come back to your original question, I will consider distributing one of zadic or wdi-simple, if I get a few more requests from people, and some clarification about the circumstances where they think Zadig cannot be used instead. Also, please bear in mind that if/when I do that, this will be a very large download, as it'll embed all the drivers, since I have no scope for multiplying versions.

ddavidebor, 2017-09-02:
I understand your point and I aready plan to signing it with our company key.

I didn't see wdi-simple, it looks to me fairly similar to zadic, what's the main difference ?

I don't think you will get many questions, the only way I was able to find zadic was by searching through the source code, and my attention got grabbed by the "c" instead of the "g".

Most CI (virtually all) support functions for encrypting sensitive data. 

Davide

pbatard, 2017-09-02:
> I understand your point and I aready plan to signing it with our company key.

Good.

> I didn't see wdi-simple, it looks to me fairly similar to zadic, what's the main difference ?

Not much. zadic started as a basic commandline utility that mimicked zadig's functionality, whereas wdi-simple started as a very basic example of libwdi usage that got extended to probably include more functionality than zadic, as people seemed to like it better.

> Most CI (virtually all) support functions for encrypting sensitive data.

Yes. But you still need to store your private key on a server that the CI can access, which is very dangerous practice, not to say downright irresponsible as you should never let the private key leave your side, or be used without your presence. I'm not aware of any CI solution having devised a solution to the binary signing problem (and frankly, I can't really see how they can, without first sending the binary back to the user to be signed, in a process that has to be manual and conducted in the user's environment, unless they introduce a major security risk).

closed 2017-11-15 by pbatard