pbatard/libwdi #99: Zadig failed to install driver (other error)

PewPewYoureDead, 2017-10-11:
I am using windows 10, and I recently bought a mayflash Wii U adapter, but I can't seem to install the drivers for it. Log is as follows.

> Using inf name: WUP-028.inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:debug [wdi_create_list] Device description: 'WUP-028'
libwdi:info [extract_binaries] successfully extracted driver files to C:\Users\HOME\Documents\USB
libwdi:info [wdi_prepare_driver] successfully created 'C:\Users\HOME\Documents\USB\WUP-028.inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\HOME\Documents\USB\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\HOME\Documents\USB\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'wup-028.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\HOME\Documents\USB\wup-028.inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\HOME\Documents\USB\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\HOME\Documents\USB\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] successfully created file 'C:\Users\HOME\Documents\USB\WUP-028.cat'
libwdi:info [RemoveCertFromStore] deleted existing certificate 'CN=USB\VID_057E&PID_0337 (libwdi autogenerated)' from 'Root' store
libwdi:info [RemoveCertFromStore] deleted existing certificate 'CN=USB\VID_057E&PID_0337 (libwdi autogenerated)' from 'TrustedPublisher' store
libwdi:debug [CreateSelfSignedCert] set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] created new key container
libwdi:debug [CreateSelfSignedCert] generated new keypair
libwdi:info [CreateSelfSignedCert] created new self-signed certificate 'CN=USB\VID_057E&PID_0337 (libwdi autogenerated)'
libwdi:debug [SelfSignFile] successfully created certificate 'CN=USB\VID_057E&PID_0337 (libwdi autogenerated)'
libwdi:info [SelfSignFile] added certificate 'CN=USB\VID_057E&PID_0337 (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
libwdi:info [SelfSignFile] successfully signed file 'C:\Users\HOME\Documents\USB\WUP-028.cat'
libwdi:info [SelfSignFile] successfully deleted private key
libwdi:debug [wdi_install_driver] using progress bar mode
libwdi:debug [installer process] got parameter WUP-028.inf
libwdi:debug [process_message] got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_057E&PID_0337\15/07/2014'
libwdi:debug [process_message] got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\VID_057E&PID_0337&REV_0100'
libwdi:debug [installer process] got user_sid: 'S-1-5-21-3955069286-1131033007-3689905193-1000'
libwdi:debug [installer process] using syslog 'C:\WINDOWS\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [process_message] switching timeout to infinite
libwdi:debug [installer process] Installing driver for USB\VID_057E&PID_0337&REV_0100 - please wait...
libwdi:debug [syslog] <Garbled data>
libwdi:debug [syslog] >>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - USB\VID_057E&PID_0337&REV_0100]
libwdi:debug [syslog] >>>  Section start 2017/10/11 19:40:37.713
libwdi:debug [syslog]       cmd: "C:\Users\HOME\Documents\USB\installer_x64.exe" "WUP-028.inf"
libwdi:debug [syslog]      ndv: INF path: C:\Users\HOME\Documents\USB\WUP-028.inf
libwdi:debug [syslog]      ndv: Install flags: 0x00000001
libwdi:debug [syslog]      ndv: {Update Device Driver - USB\VID_057E&PID_0337\15/07/2014}
libwdi:debug [syslog]      ndv:      Search options: 0x00000080
libwdi:debug [syslog]      ndv:      Searching single INF 'C:\Users\HOME\Documents\USB\WUP-028.inf'
libwdi:debug [syslog]      dvi:      {Build Driver List} 19:40:37.753
libwdi:debug [syslog]      dvi:           Searching for hardware ID(s):
libwdi:debug [syslog]      dvi:                usb\vid_057e&pid_0337&rev_0100
libwdi:debug [syslog]      dvi:                usb\vid_057e&pid_0337
libwdi:debug [syslog]      dvi:           Searching for compatible ID(s):
libwdi:debug [syslog]      dvi:                usb\class_03&subclass_00&prot_00
libwdi:debug [syslog]      dvi:                usb\class_03&subclass_00
libwdi:debug [syslog]      dvi:                usb\class_03
libwdi:debug [syslog]      sig:           {_VERIFY_FILE_SIGNATURE} 19:40:37.801
libwdi:debug [syslog]      sig:                Key      = wup-028.inf
libwdi:debug [syslog]      sig:                FilePath = c:\users\home\documents\usb\wup-028.inf
libwdi:debug [syslog]      sig:                Catalog  = c:\users\home\documents\usb\WUP-028.cat
libwdi:debug [syslog] !    sig:                Verifying file against specific (valid) catalog failed! (0x800b0109)
libwdi:debug [syslog] !    sig:                Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.
libwdi:debug [syslog]      sig:           {_VERIFY_FILE_SIGNATURE exit(0x800b0109)} 19:40:37.883
libwdi:debug [syslog]      sig:           {_VERIFY_FILE_SIGNATURE} 19:40:37.887
libwdi:debug [syslog]      sig:                Key      = wup-028.inf
libwdi:debug [syslog]      sig:                FilePath = c:\users\home\documents\usb\wup-028.inf
libwdi:debug [syslog]      sig:                Catalog  = c:\users\home\documents\usb\WUP-028.cat
[... 100 lines cut ...]
libwdi:debug [syslog]      dvi:                Default installer: Enter 19:40:38.544
libwdi:debug [syslog]      dvi:                Default installer: Exit
libwdi:debug [syslog]      dvi:           {DIF_ALLOW_INSTALL - exit(0xe000020e)} 19:40:38.545
libwdi:debug [syslog]      dvi:           {DIF_REGISTER_COINSTALLERS} 19:40:38.545
libwdi:debug [syslog]      dvi:                Default installer: Enter 19:40:38.546
libwdi:debug [syslog]      dvi:                Default installer: Exit
libwdi:debug [syslog]      dvi:           {DIF_REGISTER_COINSTALLERS - exit(0x00000000)} 19:40:38.547
libwdi:debug [syslog]      dvi:           {DIF_INSTALLDEVICE} 19:40:38.547
libwdi:debug [syslog]      dvi:                Default installer: Enter 19:40:38.548
libwdi:debug [syslog] !    dvi:                     Installing NULL driver!
libwdi:debug [syslog]      dvi:                     Install Null Driver: Removing device sub-tree. 19:40:38.552
libwdi:debug [syslog]      dvi:                     Install Null Driver: Removing device sub-tree completed. 19:40:38.554
libwdi:debug [syslog]      dvi:                     Install Null Driver: Restarting device. 19:40:38.560
libwdi:debug [syslog]      dvi:                     Install Null Driver: Restarting device completed. 19:40:38.561
libwdi:debug [syslog]      dvi:                     Install Device: Starting device. 19:40:38.562
libwdi:debug [process_message] switching timeout back to finite
libwdi:debug [installer process] unhandled error 0x3F1 (1009)
libwdi:debug [installer process] [1009] The configuration registry database is corrupt.
Driver Installation: FAILED (Other error)
1 device found.
libwdi:debug [process_message] installer process completed
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_057E&PID_0337&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Driverless USB device (3): USB\VID_057E&PID_0337\15/07/2014
What is the problem?

pbatard, 2017-10-11:
> `Error 1009: The configuration registry database is corrupt.`

Looks like a Windows issue to me.

As far as Windows is concerned, it thinks there's something that is corrupted in your OS. This is the first time I am seeing this error, so I don't have any other pointers but to say that this doesn't look like it's related to Zadig, and you will have to look on the internet for ways to fix this.

PewPewYoureDead, 2017-10-11:
Would you just suggest a complete re-install of windows then? I have been experiencing problems with installing new drivers for keyboards and mice before but I thought it was just a problem with the hardware.

pbatard, 2017-10-11:
I'd avoid a complete reinstall if you can. Try to look for the error and see what advice people provide.

pbatard, 2017-11-15:
I'm going to close this issue now.

closed 2017-11-15 by pbatard