pbatard/libwdi #100: Zadig fails to install WinUSB on Win10

dtomtom, 2017-11-01:
Hi 
I cant get Zadig (2.3) to install WINUSB, any help appreciated. Log file below.

Regards
Dave

Using inf name: RTL2838UHIDIR.inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] successfully extracted driver files to C:\Users\Home\usb_driver
libwdi:info [wdi_prepare_driver] successfully created 'C:\Users\Home\usb_driver\RTL2838UHIDIR.inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Home\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Home\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'rtl2838uhidir.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Home\usb_driver\rtl2838uhidir.inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Home\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Home\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] successfully created file 'C:\Users\Home\usb_driver\RTL2838UHIDIR.cat'
libwdi:info [RemoveCertFromStore] deleted existing certificate 'CN=USB\VID_0BDA&PID_2838 (libwdi autogenerated)' from 'Root' store
libwdi:info [RemoveCertFromStore] deleted existing certificate 'CN=USB\VID_0BDA&PID_2838 (libwdi autogenerated)' from 'TrustedPublisher' store
libwdi:debug [CreateSelfSignedCert] set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] created new key container
libwdi:debug [CreateSelfSignedCert] generated new keypair
libwdi:info [CreateSelfSignedCert] created new self-signed certificate 'CN=USB\VID_0BDA&PID_2838 (libwdi autogenerated)'
libwdi:debug [SelfSignFile] successfully created certificate 'CN=USB\VID_0BDA&PID_2838 (libwdi autogenerated)'
libwdi:info [SelfSignFile] added certificate 'CN=USB\VID_0BDA&PID_2838 (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
libwdi:info [SelfSignFile] successfully signed file 'C:\Users\Home\usb_driver\RTL2838UHIDIR.cat'
libwdi:info [SelfSignFile] successfully deleted private key
libwdi:debug [wdi_install_driver] using progress bar mode
libwdi:debug [installer process] got parameter RTL2838UHIDIR.inf
libwdi:debug [process_message] got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_0BDA&PID_2838\00000001'
libwdi:debug [process_message] got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\VID_0BDA&PID_2838&REV_0100'
libwdi:debug [installer process] got user_sid: 'S-1-5-21-3464210001-2128229742-3006012953-1000'
libwdi:debug [installer process] using syslog 'C:\WINDOWS\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [process_message] switching timeout to infinite
libwdi:debug [installer process] Installing driver for USB\VID_0BDA&PID_2838&REV_0100 - please wait...
libwdi:debug [syslog] <Garbled data>
libwdi:debug [syslog] >>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - USB\VID_0BDA&PID_2838&REV_0100]
libwdi:debug [syslog] >>>  Section start 2017/11/01 18:43:02.094
libwdi:debug [syslog]       cmd: "C:\Users\Home\usb_driver\installer_x64.exe" "RTL2838UHIDIR.inf"
libwdi:debug [syslog]      ndv: INF path: C:\Users\Home\usb_driver\RTL2838UHIDIR.inf
libwdi:debug [syslog]      ndv: Install flags: 0x00000001
libwdi:debug [syslog]      ndv: {Update Device Driver - USB\VID_0BDA&PID_2838\00000001}
libwdi:debug [syslog]      ndv:      Search options: 0x00000080
libwdi:debug [syslog]      ndv:      Searching single INF 'C:\Users\Home\usb_driver\RTL2838UHIDIR.inf'
libwdi:debug [syslog]      dvi:      {Build Driver List} 18:43:02.132
libwdi:debug [syslog]      dvi:           Searching for hardware ID(s):
libwdi:debug [syslog]      dvi:                usb\vid_0bda&pid_2838&rev_0100
libwdi:debug [syslog]      dvi:                usb\vid_0bda&pid_2838
libwdi:debug [syslog]      dvi:           Searching for compatible ID(s):
libwdi:debug [syslog]      dvi:                usb\devclass_00&subclass_00&prot_00
libwdi:debug [syslog]      dvi:                usb\devclass_00&subclass_00
libwdi:debug [syslog]      dvi:                usb\devclass_00
libwdi:debug [syslog]      dvi:                usb\composite
libwdi:debug [syslog]      sig:           {_VERIFY_FILE_SIGNATURE} 18:43:02.179
libwdi:debug [syslog]      sig:                Key      = rtl2838uhidir.inf
libwdi:debug [syslog]      sig:                FilePath = c:\users\home\usb_driver\rtl2838uhidir.inf
libwdi:debug [syslog]      sig:                Catalog  = c:\users\home\usb_driver\RTL2838UHIDIR.cat
libwdi:debug [syslog] !    sig:                Verifying file against specific (valid) catalog failed! (0x800b0109)
libwdi:debug [syslog] !    sig:                Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.
libwdi:debug [syslog]      sig:           {_VERIFY_FILE_SIGNATURE exit(0x800b0109)} 18:43:02.235
[... 108 lines cut ...]
libwdi:debug [syslog] !!!  ndv:      Error 1726: The remote procedure call failed.
libwdi:debug [syslog]      ndv: {Update Device Driver - exit(000006be)}
libwdi:debug [syslog] !!!  ndv: Failed to install device instance 'USB\VID_0BDA&PID_2838\00000001'. Error = 0x000006be
libwdi:debug [process_message] switching timeout back to finite
libwdi:debug [installer process] unhandled error 0x6BE (1726)
libwdi:debug [installer process] [1726] The remote procedure call failed.
libwdi:debug [process_message] installer process completed
Driver Installation: FAILED (Other error)
3 devices found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0BDA&PID_2838&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\DevClass_00&SubClass_00&Prot_00
libwdi:debug [wdi_create_list] Upper filter: RTL2832UBDA
libwdi:debug [wdi_create_list] Driver version: 64.1.304.2009
libwdi:debug [wdi_create_list] RTL2832UUSB USB device (3): USB\VID_0BDA&PID_2838\00000001
libwdi:debug [wdi_create_list] Device description: 'RTL2838UHIDIR'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_413C&PID_3010&REV_0220
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_02
libwdi:debug [wdi_create_list] Driver version: 10.0.15063.0
libwdi:debug [wdi_create_list] HidUsb USB device (4): USB\VID_413C&PID_3010\5&18B83DD8&0&2
libwdi:debug [wdi_create_list] Device description: 'USB Input Device'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_413C&PID_2003&REV_0306
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_01
libwdi:debug [wdi_create_list] Driver version: 10.0.15063.0
libwdi:debug [wdi_create_list] HidUsb USB device (8): USB\VID_413C&PID_2003\5&18B83DD8&0&1
libwdi:debug [wdi_create_list] Device description: 'Dell USB Keyboard'

pbatard, 2017-11-01:
> `Failed to install catalog 'RTL2838UHIDIR.cat' as 'oem8.cat'. Error = 0x000006BE`

This is a Windows internal error that has nothing to do with Zadig and that I'm pretty sure you will get even if you weren't using Zadig to install the driver (you can do that test yourself by right clicking on your device in device manager, selecting install driver and then picking up the driver from `C:\Users\Home\usb_driver\rtl2838uhidir.inf`.

Clearly, the issue is with your environment, so you will have to fix whatever it is with your environment that is preventing this driver installation. Maybe [this](https://www.rarst.net/software/failed-install-catalog-files/) will help, but if not, you will have to google around for possible solutions.

dtomtom, 2017-11-04:
Thanks for the advice, I will Google it further

closed 2017-11-04 by dtomtom