pbatard/libwdi #102: Use zadig to install/switch between other drivers than WinUSB and libusb

MCUdude, 2017-11-15:
Hi!
I have an issue where I'm using an Atmel programmer (avrisp mkii) with a windows computer. I need to switch between the old driver (Windriver6) and the "new" driver (libusb) due to some batch programming of AVR microcontrollers. I can switch to libusb using Zadig, but it's impossible to switch back to Windriver6 again. WinUSB does not work either, so it _has_ to be Windriver6. I know where Windriver6 is located on the computer, so how can I make Zadig switch between libusb and Windriver6? is it even possible?
Thanks!

pbatard, 2017-11-15:
I'm afraid Zadig does not have the capability to "switch" drivers, and I have non plans to add one at the moment. You'll have to use Device Manager to add/remove drivers.

closed 2017-11-15 by pbatard

MCUdude, 2017-11-15:
OK, so there's no way to make Windriver6 appear in the in the right window?

[image]

pbatard, 2017-11-15:
I'm afraid not. The only drivers that will ever appear there are the __generic__ ones that have been __embedded__ in libwdi at compilation time, and these are WinUSB, libusb-win32, libusbK or CDC. The whole goal of Zadig/libwdi is to help with the installation of one of these __generic__ drivers. If that's not one of the drivers you want to use, then Zadig/libwdi cannot help.