pbatard/libwdi #103: Option to install driver for a single device

geovie, 2017-11-17:
Hi,
i have a special use case, where I want to install winUSB drivers for a single device only.
Specifically it should be possible that one USB device gets installed with the libwdi driver and devices which are inserted later will not get the libwdi driver but the generic windows driver.

Is this already possible with libwdi? (I tried setting use_wcid_driver to FALSE, but that didn't worked)

Are there known workarounds in cases like this? (For now I'm testing if I could uninstall the driver package via [DriverPackageUninstall](https://msdn.microsoft.com/en-us/library/windows/hardware/ff544822(v=vs.85).aspx), right after the driver has been installed by libwdi)

Thanks!

pbatard, 2017-11-17:
WCID is unrelated to what you are trying to achieve. And your issue has to do purely with the Windows driver mechanism, not libwdi. I would suggest you contact Microsoft or Microsoft forums, as this question is out of the scope of what I can provide support for.

closed 2017-11-17 by pbatard

geovie, 2017-11-17:
Thank you anyway!