pbatard/libwdi #105: WinUSB driver detection

mondalaci, 2017-11-27:
My application must be able to check whether the WinUSB driver has been installed by zadig/zadic/wdi-simple for a given VID:PID:interface. How is it possible?

I read that zadic will likely be deprecated, so it'd be great if this was possible with wdi-simple.

pbatard, 2017-11-27:
This is not something that I am planning to ever add or support, especially as the whole point of libwdi is to automate what one should be able to do manually when it comes to driver installation.

What this means is that, ultimately, there shouldn't be much of anything that indicates whether a libwdi based application was used for the driver installation, or something else.

This being said, if you used the default installer, because libwdi will eliminated user prompts, you should get a certificate installed for each device under Trusted Publishers in your cert store:

[image]

So you could build an application that checks the cert store for these certificates, which would tell you whether the drive was installed by libwdi. But once more, you will have to do that yourself, as I have no plans to add that feature to libwdi, especially as you definitely appear to be trying to solve a __custom__ problem that applies only to __you__. If that is the case, then you can understand that, considering that the library is fully open source, the expectation is that __you__ will be the one to bring the code that adds your specific feature requirement, rather than the developers of the library themselves.

mondalaci, 2017-11-27:
Thanks for the idea, and the blazingly fast response!

Based on your answer, [our application](https://github.com/UltimateHackingKeyboard/agent) (also a Free Software project) will query Windows certificates to figure out whether the WinUSB driver has been installed for the raw HID interfaces of our keyboard. Hopefully, it'll work out.

closed 2017-11-27 by mondalaci