pbatard/libwdi #107: Driver Installation FAILED, (OTHER ERROR)

TheKangaskhanKid, 2017-12-15:
When I try to convert my gamecube controller adapter to WinUSB, it fails. When I try again it says in the debug: 
libwdi:debug [installer process] [1618] Another installation is already in progress. Complete that installation before proceeding with this install.
Help please!

TheKangaskhanKid, 2017-12-15:
Here is code:
Using inf name: WUP-028.inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] successfully extracted driver files to C:\Users\Michael\usb_driver
libwdi:info [wdi_prepare_driver] successfully created 'C:\Users\Michael\usb_driver\WUP-028.inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Michael\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Michael\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'wup-028.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Michael\usb_driver\wup-028.inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Michael\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Michael\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] successfully created file 'C:\Users\Michael\usb_driver\WUP-028.cat'
libwdi:info [RemoveCertFromStore] deleted existing certificate 'CN=USB\VID_057E&PID_0337 (libwdi autogenerated)' from 'Root' store
libwdi:info [RemoveCertFromStore] deleted existing certificate 'CN=USB\VID_057E&PID_0337 (libwdi autogenerated)' from 'TrustedPublisher' store
libwdi:debug [CreateSelfSignedCert] set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] created new key container
libwdi:debug [CreateSelfSignedCert] generated new keypair
libwdi:info [CreateSelfSignedCert] created new self-signed certificate 'CN=USB\VID_057E&PID_0337 (libwdi autogenerated)'
libwdi:debug [SelfSignFile] successfully created certificate 'CN=USB\VID_057E&PID_0337 (libwdi autogenerated)'
libwdi:info [SelfSignFile] added certificate 'CN=USB\VID_057E&PID_0337 (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
libwdi:info [SelfSignFile] successfully signed file 'C:\Users\Michael\usb_driver\WUP-028.cat'
libwdi:info [SelfSignFile] successfully deleted private key
libwdi:debug [wdi_install_driver] using progress bar mode
libwdi:debug [installer process] got parameter WUP-028.inf
libwdi:debug [process_message] got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_057E&PID_0337\15/07/2014'
libwdi:debug [process_message] got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\VID_057E&PID_0337&REV_0100'
libwdi:debug [installer process] got user_sid: 'S-1-5-21-3123326749-3470722530-428927036-1000'
libwdi:debug [installer process] using syslog 'C:\Windows\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [process_message] switching timeout to infinite
libwdi:debug [installer process] Installing driver for USB\VID_057E&PID_0337&REV_0100 - please wait...
libwdi:debug [syslog] <Garbled data>
libwdi:debug [syslog] >>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - USB\VID_057E&PID_0337&REV_0100]
libwdi:debug [syslog] >>>  Section start 2017/12/14 18:25:29.522
libwdi:debug [syslog]       cmd: "C:\Users\Michael\usb_driver\installer_x64.exe" "WUP-028.inf"
libwdi:debug [syslog]      dvi: Set selected driver complete.
libwdi:debug [syslog]      dvi: {Build Driver List} 18:25:29.542
libwdi:debug [syslog]      dvi:      Searching for hardware ID(s):
libwdi:debug [syslog]      dvi:           usb\vid_057e&pid_0337&rev_0100
libwdi:debug [syslog]      dvi:           usb\vid_057e&pid_0337
libwdi:debug [syslog]      dvi:      Searching for compatible ID(s):
libwdi:debug [syslog]      dvi:           usb\class_03&subclass_00&prot_00
libwdi:debug [syslog]      dvi:           usb\class_03&subclass_00
libwdi:debug [syslog]      dvi:           usb\class_03
libwdi:debug [syslog]      cpy:      Policy is set to make all digital signatures equal.
libwdi:debug [syslog]      dvi:      Processing a single INF: 'c:\users\michael\usb_driver\wup-028.inf'
libwdi:debug [syslog]      inf:      Opened INF: 'c:\users\michael\usb_driver\wup-028.inf' ([strings])
libwdi:debug [syslog]      sig:      {_VERIFY_FILE_SIGNATURE} 18:25:29.548
libwdi:debug [syslog]      sig:           Key      = wup-028.inf
libwdi:debug [syslog]      sig:           FilePath = c:\users\michael\usb_driver\wup-028.inf
libwdi:debug [syslog]      sig:           Catalog  = c:\users\michael\usb_driver\WUP-028.cat
libwdi:debug [syslog] !    sig:           Verifying file against specific (valid) catalog failed! (0x800b0109)
libwdi:debug [syslog] !    sig:           Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.
libwdi:debug [syslog]      sig:      {_VERIFY_FILE_SIGNATURE exit(0x800b0109)} 18:25:29.597
libwdi:debug [syslog]      sig:      {_VERIFY_FILE_SIGNATURE} 18:25:29.597
libwdi:debug [syslog]      sig:           Key      = wup-028.inf
libwdi:debug [syslog]      sig:           FilePath = c:\users\michael\usb_driver\wup-028.inf
libwdi:debug [syslog]      sig:           Catalog  = c:\users\michael\usb_driver\WUP-028.cat
libwdi:debug [syslog]      sig:           Success: File is signed in Authenticode(tm) catalog.
[... cut ...]
libwdi:warning [wdi_create_list] could not read driver version
libwdi:debug [wdi_create_list] libusbK USB device (12): USB\VID_057E&PID_0337\15/07/2014
libwdi:debug [wdi_create_list] Device description: 'WUP-028'
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0C45&PID_6701&REV_0100
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_03&SubClass_01&Prot_02
libwdi:debug [wdi_create_list] Driver version: 6.1.7601.17514
libwdi:debug [wdi_create_list] HidUsb USB device (13): USB\VID_0C45&PID_6701\5&739AAB7&0&5

pbatard, 2017-12-16:
1. Try to remove the `usb_driver` directory under `C:\Users\Michael`
2. Reboot your computer
3. Try again

closed 2018-01-05 by pbatard