pbatard/libwdi #110: How to run intaller_x64.exe using Zadig

zinnya, 2018-01-10:
Hello all, 
I am trying to install USB driver based on libusb. I have an installer_x64.exe file and a .inf  file. both fail to install the driver. Running installer_x64.exe says 

> This application can not be run from the command line.
Please use your initial installer application if you want to install the driver. 
Googling the issue revealed to use Zadig to install using .exe file. but in Zadig there is no option to run that .exe or install driver using .inf file. Any idea how to do that?

pbatard, 2018-01-10:
You __DON'T__ run `installer_x64.exe` yourself. It's what Zadig calls behind the scenes and you're not supposed to touch it, ever.

Just run Zadig, select your device, and press the _Install Driver_ button.

closed 2018-01-10 by pbatard

zinnya, 2018-01-10:
I did so, an issue is bothering. When I try to install driver with Zadig an error occurs that 

> driver installation failed (resource already exists)

When I uninstall driver from Device Manager it disappears from device list in Zadig. Any idea whats wrong?

pbatard, 2018-01-10:
Try to remove the `usb_driver` directory under `C:\users\<your_user_name>\` and then reboot.

Then make sure that, in Device Manager, your device is really listed with no driver installed.
Then Run Zadig, go to _Options_ &rarr; _List All Devices_ and try again.

Note that, if you want to install the driver manually from the files that exist under `C:\users\<your_user_name>\usb_driver`, you would __NOT__ use `installer_x64.exe` but point to the `.inf` file in Device Manager.

zinnya, 2018-01-10:
I tried it as well again same behavior. When I try installing with .inf file it says 

> The folder you specified doesn't contain a compatible software driver for your device. If the folder contains a driver, make sure it is designed to work with Windows for x64-based systems.

pbatard, 2018-01-10:
Then chances are you are using Zadig wrong. Please post your complete log from Zadig.

zinnya, 2018-01-10:
Here is the Log

> Using inf name: Unknown_USB_Device_(Device_Descriptor_Request_Failed).inf
> Successfully extracted driver files.
> Installing driver. Please wait...
> libwdi:info [extract_binaries] successfully extracted driver files to C:\Users\dsaadmin\usb_driver
> libwdi:info [wdi_prepare_driver] successfully created 'C:\Users\dsaadmin\usb_driver\Unknown_USB_Device_(Device_Descriptor_Request_Failed).inf'
> libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
> libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
> libwdi:info [ScanDirAndHash] added hash for 'C:\Users\dsaadmin\usb_driver\amd64\wdfcoinstaller01011.dll'
> libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
> libwdi:info [ScanDirAndHash] added hash for 'C:\Users\dsaadmin\usb_driver\amd64\winusbcoinstaller2.dll'
> libwdi:debug [AddFileHash] 'unknown_usb_device_(device_descriptor_request_failed).inf': INF type
> libwdi:info [ScanDirAndHash] added hash for 'C:\Users\dsaadmin\usb_driver\unknown_usb_device_(device_descriptor_request_failed).inf'
> libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
> libwdi:info [ScanDirAndHash] added hash for 'C:\Users\dsaadmin\usb_driver\x86\wdfcoinstaller01011.dll'
> libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
> libwdi:info [ScanDirAndHash] added hash for 'C:\Users\dsaadmin\usb_driver\x86\winusbcoinstaller2.dll'
> libwdi:info [CreateCat] successfully created file 'C:\Users\dsaadmin\usb_driver\Unknown_USB_Device_(Device_Descriptor_Request_Failed).cat'
> libwdi:info [RemoveCertFromStore] deleted existing certificate 'CN=USB\VID_0000&PID_0002 (libwdi autogenerated)' from 'Root' store
> libwdi:info [RemoveCertFromStore] deleted existing certificate 'CN=USB\VID_0000&PID_0002 (libwdi autogenerated)' from 'TrustedPublisher' store
> libwdi:debug [CreateSelfSignedCert] set Enhanced Key Usage, URL and CPS
> libwdi:debug [CreateSelfSignedCert] created new key container
> libwdi:debug [CreateSelfSignedCert] generated new keypair
> libwdi:info [CreateSelfSignedCert] created new self-signed certificate 'CN=USB\VID_0000&PID_0002 (libwdi autogenerated)'
> libwdi:debug [SelfSignFile] successfully created certificate 'CN=USB\VID_0000&PID_0002 (libwdi autogenerated)'
> libwdi:info [SelfSignFile] added certificate 'CN=USB\VID_0000&PID_0002 (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
> libwdi:info [SelfSignFile] successfully signed file 'C:\Users\dsaadmin\usb_driver\Unknown_USB_Device_(Device_Descriptor_Request_Failed).cat'
> libwdi:info [SelfSignFile] successfully deleted private key
> libwdi:debug [wdi_install_driver] using progress bar mode
> libwdi:debug [installer process] got parameter Unknown_USB_Device_(Device_Descriptor_Request_Failed).inf
> libwdi:debug [process_message] got request for device_id
> libwdi:debug [installer process] got device_id: 'USB\VID_0000&PID_0002\5&1548B049&1&4'
> libwdi:debug [process_message] got request for hardware_id
> libwdi:debug [installer process] got hardware_id: 'USB\DEVICE_DESCRIPTOR_FAILURE'
> libwdi:debug [installer process] got user_sid: 'S-1-5-21-1196300794-3732143510-2954861690-1001'
> libwdi:debug [installer process] using syslog 'C:\WINDOWS\inf\setupapi.dev.log'
> libwdi:debug [installer process] syslog reader thread started
> libwdi:debug [installer process] successfully disabled the system restore point creation setting
> libwdi:debug [process_message] switching timeout to infinite
> libwdi:debug [installer process] Installing driver for USB\DEVICE_DESCRIPTOR_FAILURE - please wait...
> libwdi:debug [process_message] switching timeout back to finite
> libwdi:debug [installer process] more recent driver was found (force option required)
> libwdi:debug [process_message] installer process completed
> Driver Installation: FAILED (Resource already exists)
> 3 devices found.
> libwdi:debug [wdi_create_list] Hardware ID: USB\VID_8087&PID_0A2B&REV_0001
> libwdi:debug [wdi_create_list] Compatible ID: USB\Class_E0&SubClass_01&Prot_01
> libwdi:debug [wdi_create_list] Driver version: 18.1.1611.3223
> libwdi:debug [wdi_create_list] BTHUSB USB device (0): USB\VID_8087&PID_0A2B\5&1548B049&1&8
> libwdi:debug [wdi_create_list] Device description: 'Intel(R) Wireless Bluetooth(R)'
> libwdi:debug [wdi_create_list] Hardware ID: USB\VID_0BDA&PID_5686&REV_5729&MI_00
> libwdi:debug [wdi_create_list] Compatible ID: USB\Class_0e&SubClass_03&Prot_00
> libwdi:debug [wdi_create_list] Driver version: 10.0.14393.82
> libwdi:debug [wdi_create_list] usbvideo USB device (2): USB\VID_0BDA&PID_5686&MI_00\6&153A3DF0&1&0000
> libwdi:debug [wdi_create_list] Device description: 'Integrated Webcam (Interface 0)'
> libwdi:debug [wdi_create_list] Hardware ID: USB\DEVICE_DESCRIPTOR_FAILURE
> libwdi:debug [wdi_create_list] Compatible ID: USB\DEVICE_DESCRIPTOR_FAILURE
> libwdi:debug [wdi_create_list] Driver version: 10.0.14393.0
> libwdi:debug [wdi_create_list] Driverless USB device (4): USB\VID_0000&PID_0002\5&1548B049&1&4
> libwdi:debug [wdi_create_list] Device description: 'Unknown USB Device (Device Descriptor Request Failed)'
>

pbatard, 2018-01-10:
> `libwdi:debug [installer process] more recent driver was found (force option required)`

There's your problem.

If the log says that a more recent driver was found, it means that you did __NOT__ properly uninstall the driver in Device Manager, which is also consistent with the fact that your device wasn't listed, as a truly driverless device should have be listed automatically in Zadig.

You need to find a way to make sure that, before you run Zadig, your device really doesn't have a driver, and remains that way.

zinnya, 2018-01-10:
This is my issue when I uninstall driver from device manager it uninstalls the driver but never show that usb device in Zadig as well.

pbatard, 2018-01-10:
No, the log is pretty clear that it actually does NOT really uninstall the driver. You need to make sure that the driver __files__ from that other driver are deleted as well, else Windows will will refuse to install the one from Zadig.

I would suggest that you reinstall that original driver (the one that the log says is more recent, because, clearly, Windows says it is available for installation), and then properly uninstall it by making sure you also check "Delete the driver software for this device" as per the second screenshot [here](https://github.com/pbatard/libwdi/wiki/FAQ#Help_Zadig_replaced_the_driver_for_the_wrong_device_How_do_I_restore_it).

You may have to plug and replug your device on various ports and do the uninstallation again, until the driver is truly deleted from your system.

Then Zadig should be able to install the WinUSB driver without getting the `more recent driver was found` message.

But again, as long as you get that message from Zadig, it means you do have a conflicting driver file on your system, which you need to remove one way or another.

zinnya, 2018-01-10:
[image]
This is how I am uninstalling the driver.

zinnya, 2018-01-10:
let me try changing ports and uninstall it.

pbatard, 2018-01-10:
Yeah. You clearly have an issue with this device. If it says `Device Descriptor Request Failed`, there's something fundamentally wrong with it, and the least of your worries should be try to install a driver coz as far as Windows is concerned, you already have an issue with the hardware itself.

zinnya, 2018-01-10:
No luck, Really stuck.

zinnya, 2018-01-10:
[image]
This shows up when I click on driver details in device manager.

zinnya, 2018-01-10:
Dear pbatard, I tried  installation on other PCs as well, Windows7 and WindowsXp. Its failing everywhere.

pbatard, 2018-01-10:
Then the problem is a hardware issue with your device. If it can't reply to Device Descriptor requests, it's not something Zadig can help with.