pbatard/libwdi #111: SNES Driver installation Failed (other error)

meggsy, 2018-01-12:
Hi, 
I have tried to install the winusb driver for my SNES. It keeps failing, and the log is saying this:
1 device found.
Using 'C:\Users\Megan\usb_driver' as extraction directory.
Using inf name: Unknown_Device_#1.inf
Successfully extracted driver files.
Installing driver. Please wait...
Driver Installation: FAILED (Other error)
1 device found.

pbatard, 2018-01-12:
Post the __full__ log. Thanks. Nobody can't help you if you don't post the complete log from Zadig.

meggsy, 2018-01-13:
Zadig 2.3.701
Windows 7 SP1 64-bit
ini file 'zadig.ini' not found - default parameters will be used
default driver set to 'WinUSB'
1 device found.
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_1F3A&PID_EFE8&REV_02;3
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_FF&SubClass_FF&Prot_FF
libwdi:debug [wdi_create_list] Driverless USB device (3): USB\VID_1F3A&PID_EFE8\6&155590E0&0&2
libwdi:debug [wdi_create_list] could not read device description for 3: [#0000000D] The data is invalid.
libwdi:debug [wdi_create_list] Device description: 'Unknown Device #1'
Using inf name: Unknown_Device_#1.inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] successfully extracted driver files to C:\Users\Megan\usb_driver
libwdi:info [wdi_prepare_driver] successfully created 'C:\Users\Megan\usb_driver\Unknown_Device_#1.inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Megan\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Megan\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'unknown_device_#1.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Megan\usb_driver\unknown_device_#1.inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Megan\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\Megan\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] successfully created file 'C:\Users\Megan\usb_driver\Unknown_Device_#1.cat'
libwdi:info [RemoveCertFromStore] deleted existing certificate 'CN=USB\VID_1F3A&PID_EFE8 (libwdi autogenerated)' from 'Root' store
libwdi:info [RemoveCertFromStore] deleted existing certificate 'CN=USB\VID_1F3A&PID_EFE8 (libwdi autogenerated)' from 'TrustedPublisher' store
libwdi:debug [CreateSelfSignedCert] set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] created new key container
libwdi:debug [CreateSelfSignedCert] generated new keypair
libwdi:info [CreateSelfSignedCert] created new self-signed certificate 'CN=USB\VID_1F3A&PID_EFE8 (libwdi autogenerated)'
libwdi:debug [SelfSignFile] successfully created certificate 'CN=USB\VID_1F3A&PID_EFE8 (libwdi autogenerated)'
libwdi:info [SelfSignFile] added certificate 'CN=USB\VID_1F3A&PID_EFE8 (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
libwdi:info [SelfSignFile] successfully signed file 'C:\Users\Megan\usb_driver\Unknown_Device_#1.cat'
libwdi:info [SelfSignFile] successfully deleted private key
libwdi:debug [wdi_install_driver] using progress bar mode
libwdi:debug [installer process] got parameter Unknown_Device_#1.inf
libwdi:debug [process_message] got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_1F3A&PID_EFE8\6&155590E0&0&2'
libwdi:debug [process_message] got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\VID_1F3A&PID_EFE8&REV_02;3'
libwdi:debug [installer process] got user_sid: 'S-1-5-21-858701222-3170018690-3397541490-1001'
libwdi:debug [installer process] using syslog 'C:\windows\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [process_message] switching timeout to infinite
libwdi:debug [installer process] Installing driver for USB\VID_1F3A&PID_EFE8&REV_02;3 - please wait...
libwdi:debug [syslog] <Garbled data>
libwdi:debug [syslog] >>>  [Device Install (UpdateDriverForPlugAndPlayDevices) - USB\VID_1F3A&PID_EFE8&REV_02;3]
libwdi:debug [syslog] >>>  Section start 2018/01/13 18:09:46.839
libwdi:debug [syslog]       cmd: "C:\Users\Megan\usb_driver\installer_x64.exe" "Unknown_Device_#1.inf"
libwdi:debug [syslog]      dvi: Set selected driver complete.
libwdi:debug [syslog]      dvi: {Build Driver List} 18:09:46.855
libwdi:debug [syslog]      cpy:      Policy is set to make all digital signatures equal.
libwdi:debug [syslog]      dvi:      Processing a single INF: 'c:\users\megan\usb_driver\unknown_device_#1.inf'
libwdi:debug [syslog]      inf:      Opened INF: 'c:\users\megan\usb_driver\unknown_device_#1.inf' ([strings])
libwdi:debug [syslog]      sig:      {_VERIFY_FILE_SIGNATURE} 18:09:46.855
libwdi:debug [syslog]      sig:           Key      = unknown_device_#1.inf
libwdi:debug [syslog]      sig:           FilePath = c:\users\megan\usb_driver\unknown_device_#1.inf
libwdi:debug [syslog]      sig:           Catalog  = c:\users\megan\usb_driver\Unknown_Device_#1.cat
libwdi:debug [syslog] !    sig:           Verifying file against specific (valid) catalog failed! (0x800b0109)
libwdi:debug [syslog] !    sig:           Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.
libwdi:debug [syslog]      sig:      {_VERIFY_FILE_SIGNATURE exit(0x800b0109)} 18:09:46.917
libwdi:debug [syslog]      sig:      {_VERIFY_FILE_SIGNATURE} 18:09:46.917
libwdi:debug [syslog]      sig:           Key      = unknown_device_#1.inf
libwdi:debug [syslog]      sig:           FilePath = c:\users\megan\usb_driver\unknown_device_#1.inf
[... cut ...]
libwdi:debug [wdi_create_list] Hardware ID: USB\VID_1F3A&PID_EFE8&REV_02;3
libwdi:debug [wdi_create_list] Compatible ID: USB\Class_FF&SubClass_FF&Prot_FF
libwdi:debug [wdi_create_list] Driverless USB device (3): USB\VID_1F3A&PID_EFE8\6&155590E0&0&2
libwdi:debug [wdi_create_list] could not read device description for 3: [#0000000D] The data is invalid.
libwdi:debug [wdi_create_list] Device description: 'Unknown Device #1'
Checking for Zadig updates...

Checking release channel...

No new release version found.

pbatard, 2018-01-13:
Thanks for the log.

By the looks of it, there is something locked down about the Windows system you are trying to install the driver on that is preventing the driver installation. Especially:
```
libwdi:debug [syslog] !!!  cpy:                          Drp can't overwrite protected file.
libwdi:debug [syslog] !!!  cpy:                          Error 0: The operation completed successfully.
libwdi:debug [syslog]      flq:                          Caller requested to abort install of file 'C:\windows\system32\DRIVERS\winusb.sys' with error 0x00000005.
libwdi:debug [syslog] !!!  flq:                          Error installing file (0x00000005)
libwdi:debug [syslog] !!!  flq:                          Error 5: Access is denied.
```
seems to indicate does not allow all the operations that Zadig requires for driver installation, such as upgrading WinUSB driver components.
Unfortunately, since this appears to be an environmental issue, I don't have any idea why that might be the case. I suspect that your system might have been configured to enforce local security policies to prevent some driver updates, but it's only a wild guess, and I also don't know where these policies may be set.

Also: 
```
libwdi:debug [syslog] !!!  flq:                          Error 1459: This operation requires an interactive window station.
```
does seem to corroborate that your system is not configured like one would expect Windows to be set, as it appears the driver installation process is under the impression that your system has no graphical interface.

The only thing I can advise is that, since the driver files should have been already extracted in `C:\Users\Megan\usb_driver\` is to try to use Windows' Device Manager, right click on your device there, select driver installation, and point to that directory to see if that works better. But that's about it.

pbatard, 2018-02-01:
Since there hasn't been any update, I'm going to close this issue.

closed 2018-02-01 by pbatard