pbatard/libwdi #113: Zadig installation (WinUSB) corrupted after installing Silicon Labs CP210x driver

sunk818, 2018-02-07:
Is there a way to update specific devices with Zadig WinUSB drivers?

I used Zadig to install GekkoScience 2Pac Bitcoin miners (qty 10). All working fine.

I installed Silicon Labs (CP210x_Universal_Windows_Driver) so I could use qty 10 of the Moonlander 2, Litecoin miners.

Silicon Labs Universal Virtual COM Port (VCP) driver for Windows 10
Package version: 10.1.1
Release date:    2017-11-15

Based on Issue #106, I tried to rollback the driver to the previous version. I selected a specific device (the Bitcoin miner that should use Zadig), but it seems to have rolled back the entire driver installation. Its as if, now all 20 devices are registered with Silicon Labs and am unable to update the drivers for specific devices.

pbatard, 2018-02-07:
Looks to me like an issue with the Silicon Labs driver installer. Have you contacted them?
I hope you can understand that it's very difficult for me to try to support an issue that occurred once you started using or installing software that is unrelated to Zadig/libwdi.

The only thing I can venture is that I suspect Silicon Lab may have forgotten that WinUSB is meant to be used as a __generic__ driver for more than one device (just not theirs), and have done something in their software that tries to alter any WinUSB driver they can see. In that case, it should be up to them to sort the issue.

Still you may want to have a look at your `C:\Windows\INF\setupapi.dev.log` to understand what happened there. That files logs the details of every single driver installation that happened on your system, and will tell you what really happened during the the Silicon Lab driver installation.

sunk818, 2018-02-07:
If I install zadig all drivers get updated as well. Was trying to figure out how to apply winusb to specific USB devices not carpet bomb. That is what I asked because I realize it's not your role to help troubleshoot  regarding third party.

pbatard, 2018-02-07:
Well, I guess the issue then is that there is only one actual WinUSB driver binary (as well as library dependencies) for the whole system, and, what's more, Windows may actually decide to update them as part of the driver installation, depending on what Microsoft sees fit (so you don't actually always end up with the actual binaries that were embedded in Zadig/libwdi, but the ones that comes from the system, or that were downloaded from Windows Update).

So, if your question is: _"Can I use WinUSB driver x.y.z along side WinUSB drive a.b.c on the same Windows system?"_, then, unless you want to fight __against__ Windows, the answer is no.

If you install a specific version of the WinUSB driver (or rather if Windows, through it's somewhat convoluted and hidden actual WinUSB driver update process decides that a specific version should be installed) then this driver binary will be used for the whole system, because it's closer to a system file that gets superseded on update, than anything else.

pbatard, 2018-02-07:
And again, if you know how to read it, `C:\Windows\INF\setupapi.dev.log` will always tell you exactly what happened during the driver installation process.

sunk818, 2018-02-07:
I'll take a look at `setupapi.dev.log`. Worst case, I'll just have to run them on separate systems. I appreciate your clarification on how the driver installation may have worked.

closed 2018-02-07 by sunk818