pbatard/libwdi #116: Batch installing with Zadig

dylanede, 2018-02-23:
It would be useful if Zadig could be run in a silent mode to install a list of devices. This could be done as an extension to the device config file format (to support multiple devices, and the ability to specify the driver to use) in combination with a command line switch.

pbatard, 2018-02-23:
I think you may want to look into `wdi-simple`, which is a commandline equivalent to what Zadig is doing.
This is part of the [samples provided by libwdi](https://github.com/pbatard/libwdi/tree/master/examples) (of which Zadig is only one example).

Many people have been using wdi-simple to achieve the kind of thing that you want.
Of course, you will have to recompile it yourself.

closed 2018-02-23 by pbatard

PetteriAimonen, 2022-02-10:
Is there a particular reason why `wdi-simple` is not included in the precompiled releases?

Anutrix, 2022-09-04:
Same here. Adding cmd support to zadig for advanced users seems like a really useful option.

pbatard, 2022-09-04:
This would take time. Which I don't have for this specific request.