pbatard/libwdi #117: Help needed installing the RTL-SDR driver using Zadig

shrishailayya, 2018-02-28:
Dear All, 

I m trying to install the driver for RTL-SDR in windows 10 OS using zadig.  After taking admin control and installing driver also it is failing. Pls if anybody can give some suggestion how to overcome this problem. Please find the log file for
[zadig.log](https://github.com/pbatard/libwdi/files/1767166/zadig.log)
 installation.

pbatard, 2018-02-28:
This looks like a Windows internal error. Error `0x000006BE` (11726) means that a remote procedure call failed, which I suspect means that Windows was trying to call into one of the services required for driver installation, and that call failed. I am pretty sure that you will get the same error if you try to install the driver manually, without using Zadig, so I don't think there's much the application can do to help.

Since this appear to be a purely environmental problem, and googling error `0x000006BE` doesn't seem to provide much insight into it, you may have to check your Windows environment for corruption or missing services. Especially, since it is the `.cat` install that seems to fail, you should check that you haven't disabled or restricted default services or policies that have to do security, as a vanilla Windows platform should be able to import a `.cat` file without trouble. You may also want to try installing the driver on a different Windows 10 platform to see if you have the same error.

Unfortunately, this is probably the most help you'll be able to get...

shrishailayya, 2018-03-01:
Dear Pete Batard 
Thank you reply and suggestion. I will check them out. 

With Best Regards 
Shrishail Hiremath, 
Department of Electronics and Communication Engineering 
National Institute of Technology 
Rourkela, PIN - 769008 
ORISSA, INDIA 
hiremaths@nitrkl.ac.in 
Mob:9438503621 
Office:0661-2462459

closed 2018-03-08 by pbatard