pbatard/libwdi #119: Can't update WinUSB driver on Samsung device ("Driver Installation Failed [Resource already exists]")

dswalen, 2018-03-15:
Hi, I'm trying to updated WinUSB for a Samsung Galaxy S6 Android device driver. The Samsung version of WinUSB is 2.12.5.0 and I'm trying to update it to 6.1.7600.16385. Every time I try to get Zadig to do this it fails. The log file seems to suggest a more recent driver was found and that a force option is required. However I know of no such option or switch in Zadig (nor anywhere else) that will facilitage this. I'm including the log here. This was observed on two different Win 10 systems. Can you tell me what's going on because I'm really not sure (or how to proceed).

Using inf name: SAMSUNG_Android.inf
Successfully extracted driver files.
Installing driver. Please wait...
libwdi:info [extract_binaries] successfully extracted driver files to C:\Users\dswalen\usb_driver
libwdi:info [wdi_prepare_driver] using Android Device Interface GUID
libwdi:info [wdi_prepare_driver] successfully created 'C:\Users\dswalen\usb_driver\SAMSUNG_Android.inf'
libwdi:info [wdi_prepare_driver] Creating and self-signing a .cat file...
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\dswalen\usb_driver\amd64\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\dswalen\usb_driver\amd64\winusbcoinstaller2.dll'
libwdi:debug [AddFileHash] 'samsung_android.inf': INF type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\dswalen\usb_driver\samsung_android.inf'
libwdi:debug [AddFileHash] 'wdfcoinstaller01011.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\dswalen\usb_driver\x86\wdfcoinstaller01011.dll'
libwdi:debug [AddFileHash] 'winusbcoinstaller2.dll': PE type
libwdi:info [ScanDirAndHash] added hash for 'C:\Users\dswalen\usb_driver\x86\winusbcoinstaller2.dll'
libwdi:info [CreateCat] successfully created file 'C:\Users\dswalen\usb_driver\SAMSUNG_Android.cat'
libwdi:info [RemoveCertFromStore] deleted existing certificate 'CN=USB\VID_04E8&PID_6860 (libwdi autogenerated)' from 'Root' store
libwdi:info [RemoveCertFromStore] deleted existing certificate 'CN=USB\VID_04E8&PID_6860 (libwdi autogenerated)' from 'TrustedPublisher' store
libwdi:debug [CreateSelfSignedCert] set Enhanced Key Usage, URL and CPS
libwdi:debug [CreateSelfSignedCert] created new key container
libwdi:debug [CreateSelfSignedCert] generated new keypair
libwdi:info [CreateSelfSignedCert] created new self-signed certificate 'CN=USB\VID_04E8&PID_6860 (libwdi autogenerated)'
libwdi:debug [SelfSignFile] successfully created certificate 'CN=USB\VID_04E8&PID_6860 (libwdi autogenerated)'
libwdi:info [SelfSignFile] added certificate 'CN=USB\VID_04E8&PID_6860 (libwdi autogenerated)' to 'Root' and 'TrustedPublisher' stores
libwdi:info [SelfSignFile] successfully signed file 'C:\Users\dswalen\usb_driver\SAMSUNG_Android.cat'
libwdi:info [SelfSignFile] successfully deleted private key
libwdi:debug [wdi_install_driver] using progress bar mode
libwdi:debug [installer process] got parameter SAMSUNG_Android.inf
libwdi:debug [process_message] got request for device_id
libwdi:debug [installer process] got device_id: 'USB\VID_04E8&PID_6860&ADB\8&25509774&6&0003'
libwdi:debug [process_message] got request for hardware_id
libwdi:debug [installer process] got hardware_id: 'USB\VID_04E8&PID_6860&REV_0400&ADB'
libwdi:debug [installer process] got user_sid: 'S-1-5-21-32718380-1033319993-1322207217-1488'
libwdi:debug [installer process] using syslog 'C:\WINDOWS\inf\setupapi.dev.log'
libwdi:debug [installer process] syslog reader thread started
libwdi:debug [installer process] successfully disabled the system restore point creation setting
libwdi:debug [process_message] switching timeout to infinite
libwdi:debug [installer process] Installing driver for USB\VID_04E8&PID_6860&REV_0400&ADB - please wait...
libwdi:debug [process_message] switching timeout back to finite
libwdi:debug [installer process] more recent driver was found (force option required)
libwdi:debug [process_message] installer process completed
Driver Installation: FAILED (Resource already exists)

pbatard, 2018-03-16:
>  The Samsung version of WinUSB is 2.12.5.0 and I'm trying to update it to 6.1.7600.16385

Be mindful that the version reporting from Zadig is not that accurate, because the WinUSB driver is a weird beast (it is part DLL, part driver) that ultimately gets updated behind the scenes by Windows Update rather than an external application like Zadig.

Therefore, if Windows says that your current WinUSB driver is more recent than the one provided by Zadig, then Windows is correct.

I would strongly advise __against__ trying to replace the existing WinUSB driver. This is pretty much a Windows system file, which you should avoid modifying (which is pretty much the reason why the system won't let you). Besides, I have yet to see a single case where updating/rolling back the WinUSB driver actually solved a perceived issue, so I am pretty confident that, whatever issue you believe might be solved by "upgrading" the WinUSB driver, is not actually going to be solved that way.

pbatard, 2018-03-24:
I'm going to close this issue now.

closed 2018-03-24 by pbatard