pbatard/libwdi #120: zadig is able to trash my usb hid devices

commandline-be, 2018-03-16:
Having endless trouble with getting my Samsung T800 detected in "download mode" i ended up with Zadig. Which did not change much.

Eventually i tried out installing wcid and reinstall driver to no avail.

Then i extracted drivers and reinstalled drivers from this folder.

First i made my Logitech keyboard unusable. It took an install of MSW10 over MSW10 to fix this, i tried al other possible option besides editing the registry.

Now it killed my Logitech Mouse which is detected but appears to no longer function. I assume the solution may be similar, however the mouse appears dead.

If there is anything i should now to not do let me know. I'd love zadig to have a 'remove changes' button so it returns the system to a clean state.

pbatard, 2018-03-16:
Please see [this FAQ entry](https://github.com/pbatard/libwdi/wiki/FAQ#Help_Zadig_replaced_the_driver_for_the_wrong_device_How_do_I_restore_it).

Also, __PLEASE__ pay attention to the big warning Zadig shows when you are about to replace an HID driver:
[image]

If you choose to ignore these warnings such as these, and try things at random (such as installing the WCID driver, which is unlikely to help), then it's up to you to fix the issues you created. A driver installation is __NOT__ a benign operation, and as much as some people may think otherwise, Zadig is a powerful tool, that you can use to shoot yourself in the foot, so you are expected to have a minimum knowledge about driver installation operation.

>  I'd love zadig to have a 'remove changes' button so it returns the system to a clean state.

As indicated [here](https://github.com/pbatard/libwdi/issues/8#issuecomment-10312046), the time I can spend on Zadig/libwdi is very limited (this is only one of the many Open Source projects I am involved with), and adding extensive hand holding and reverting features to an application such as Zadig would be very time consuming (a lot more than people imagine). Considering that Zadig/libwdi is targeted at people who have a minimum knowledge of what they are doing, I have no plans to add such a feature.

Then again, this project is Open Source, so if anybody wants to invest the time required to do so, they are very welcome to send a patch.

commandline-be, 2018-03-16:
With all due respect for spending time writing open source software. Sadly i'm not a dev and most likely cannot help much with further improving zadig.

What is did was follow what was at not one but few times recommended approach for using zadig.

Installing USB drivers (even HID) particularly should be a trivial thing. My point is the way zadig handles drivers appears to break how MS Windows handles drivers properly. When even removing the device+installed drivers leaves a mess there is something wrong.

Not arguing i may have clicked one too many clicks in frustration.

> Zadig is a Windows application that installs generic USB drivers, such as WinUSB, libusb-win32/libusb0.sys or libusbK, to help you access USB devices.

Which is exactly what i was told to use it for with ODIN and Heimdall. I gave up on the idea of rooting this tablet since none of the tools and procedures, including zadig, or works as advertised or has adequate documentation to understand well how to operate it.

pbatard, 2018-03-16:
> What is did was follow what was at not one but few times recommended approach for using zadig.

Yes, unfortunately, Zadig has been victim of its own success, and is now being used by people who don't understand its interface (you are supposed to be able to recognize and select the one device you want to install the driver for in the dropdown list, and only ever apply changes for that device!), and therefore end up doing things they shouldn't be doing.

I'd really like to have the opportunity to provide a dumbed down version of Zadig, that adds a lot of hand holding and driver restoration, but that's just too much effort. Besides, if manufacturers really wanted to help their users, they would be using WCID in their devices to make the need to manually install the WinUSB driver obsolete, be it through Zadig or other means. Considering how easy it is to create a device that will automatically install the WinUSB driver as soon as it is plugged on a Windows computer (which is what WCID is for), it's almost criminal in my book that phone manufacturers are still not doing so for their recent models, because it only results in user frustration.

For the time being, if you insist, you should be able to manually restore your HID devices to their default drivers using Device Manager. You may have to plug another mouse and keyboard to be able to do so, and you may also try the various advice provided in #8. But that's about all the help I can provide...

hedgepiggy, 2020-03-25:
Two years down the line, I just got absolutely destroyed by your software I was directed to install to install a driver. I don't know if I navigated your UI wrong or something (maybe I accidentally pressed enter?), but my mouse and keyboard drivers got replaced randomly and my screen stuttered and turned black. I spent about 30 minutes trying to find which driver got replaced as I had a lot of stuff listed in device manager, and had to figure out how to get on screen keyboard working in a bind.

It would be really nice if you added a giant caps screen upon first execution or input device replacement WARNING users to be careful what they are doing. If filtering for input devices is too much effort, please just add a nag screen upon first execution, since I (and probably plenty of other unsuspecting users) don't actually know what the risks of using this software improperly. 

If you seek for inspiration on the issue, maybe use a prompt similar to disk partitioning?

Thanks.

pbatard, 2020-03-25:
> but my mouse and keyboard drivers got replaced

Then you will have seen __this EXTRA prompt__:
[image]

When replacing HID drivers, which is what keyboard and mouse use, I made sure to add an __EXTRA__prompt.

So I'm afraid that you've just made the point that, it doesn't matter how many prominent warnings one tries to add to an app, users still ignore them... Doesn't matter if you place them on first app use or when executing a potentially dangerous operation or make sure to use a big red WARNING in a large font because it's been my experience that folks still choose to see them as an __annoyance__ that they will ignore because they think they're in a rush and don't want to spend any time having to try to understand the risks associated with the operation they are trying to carry out.

Sorry, but the onus is on __YOU__ to pay attention to what the prompts tell you. The extra prompt you ask for has been there all along. Zadig is not designed to be a hand holding application for people who have no clue what a driver is. The thing is I went out of my way to provide a library (libwdi) that application developers can use to create foolproof installation software for specific hardware, whereas, on the other hand, because Zadig has no idea what specific device you are trying to install a driver for, it is designed to be generic and allow you to shoot yourself in the foot if that's what you choose.

commandline-be, 2020-03-25:
Two years and you still deflect responsibility instead of addressing issues

Verzonden met ProtonMail Mobile

\-------- Oorspronkelijk bericht --------

[view it on GitHub]: https://github.com/pbatard/libwdi/issues/120#issuecomment-603801901
[unsubscribe]: https://github.com/
[AID7C3LS4G4V2YPBPXMBEFLRJHXIJA5CNFSM4EVVQFW2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEP6USLI.gif]: https://github.com/notifications/beacon/AID7C3LS4G4V2YPBPXMBEFLRJHXIJA5CNFSM4EVVQFW2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEP6USLI.gif

pbatard, 2020-03-25:
Two years, and I still haven't seen anything to make me think that users who choose to dismiss important __EXTRA__ warnings, that I made sure the application provides in case they are about to do something that they probably shouldn't do, might not actually be better off getting a valuable lesson about the importance of not ignoring said warnings, instead of never realizing that, yes, there is an associated risk to some operations they are trying to accomplish (even more so when the "risk" in this case is rather easily recoverable and doesn't incur any loss of data), and that they should not be dismissing warnings a bit too casually.

Also, you seem to believe that my time is worth as much as the price of the application you are using (i.e. 0), and therefore that I should be able to drop all the other stuff I'm doing development-wise (which I personally see as endeavours that are going to benefit a lot more users than trying to dumb down Zadig) to add whatever request you wish for. Well, the application might be free but my time isn't, so I get to choose what I consider as important to spend it on. And time-wise, fostering reckless behaviours by promoting the idea that software should always do all the thinking for you as well as consider that users are never going to pay enough attention to explicit warnings, does not seem like a good investment, sorry.

hedgepiggy, 2020-03-26:
Okay, let's be real here, I made a simple, polite request for you to add a screen (I didn't see anything show up on your shoddy UI, and perhaps my understanding of drivers is flawed but aren't all drivers under the system ("system driver is quite ambiguous here)? Maybe a pointer to the fact that it's an HID driver that's getting replaced, since I never got any PROMPT as to what was happening, before my computer just spazzed out? I'd be happy to spend the time to reproduce this issue, though I'll be installing a virtual machine since the last time around all I did was move drop downs and all of a sudden my mouse and keyboard died.

Regardless, given that you probably spent as much time typing just about 500 words in two responses as it would have taken you to do this feature, you could have saved more time simply stating that you didn't want to. Heck, I would've sent a PR if you were open to it.

The thing about software is that it should be intuitive (OR AT LEAST SUPER INFORMATIVE IF SOMETHING DANGEROUS IS ABOUT TO HAPPEN, AND SPECIFIC ABOUT WHAT'S GONNA HAPPEN AND NOT VAGUE WITH SOMETHING LIKE "SYSTEM DRIVER"), and while I might not be the cream of the crop, I have had my fair share of experiences with software including setting up a hypervisor, flashing firmware, developing front and back end web code. Never have I nuked low level system drivers without at least 2 visible descriptive warnings, and I counted zero on your application.

Anyways, best of luck to you on whatever your project is (I again have no idea nor care what this software is to be used for, but presumably it's useful since it was listed as an external dependency which is why I wound up installing it anyways), janky as its front end GUI might be. While I understand that open source development might seem like a thankless give and not take process, at least from the user (my perspective), all I see is that I lost an hour of my time trying to figure out why my drivers died, fix them, take the time out of my day to post an issue, then get aggressively shot down trying to suggest an improvement to an **EXISTING ISSUE CLEARLY INDICATING THAT I'M NOT THE ONLY ONE ENCOUNTERING PROBLEMS** with this application's lack of basic HCI principles.

pbatard, 2020-03-26:
>  I'd be happy to spend the time to reproduce this issue

Please do that. If you do you will see the prompt that you allegedly never got. Unless you have some weird hardware, your mouse and keyboard will use the Windows native HID driver, which Zadig will detect as a system driver, and you will get the extra prompt I showed above. I am 100% positive here that, as oppose to what you relate, you __did__ see the extra warning and that you simply chose to ignore it.

Now about the amping up the warning, there's nothing fundamentally dangerous about replacing a system driver, since you can always revert that change, which is why, as opposed to what you seem to be adamant I should do, I'm not going to replace the current supplementary warning that tells you that you are about to modify a system driver with a big red skull.

I'm afraid that your issue is that you are assuming that, because you perceive things a certain way everybody must be wanting the same thing as you. Unfortunately for you, and for the handful of people who might agree that Zadig should produce a warning similar to the one you'd get from a nuclear reactor going critical, this is far from the reality I see from the feedback I have gotten as a developer, especially as the notice that is allegedly not present to warn you is __already__ there.

Oh, and so far I'm only seeing two people complaining about this, and, since both these people appear to have replaced mice and keyboard drivers, from which I made sure to __already add__ an extra warning, yet don't seem to be willing to admit that they got that extra warning indeed, you may perhaps understand why I'm not that sympathetic to these pleas.

Of course, if you can demonstrate that you can replace an HID mouse or keyboard without seeing that extra warning, I'll be happy to stand corrected and revise my current theory that you are simply unwilling to admit that you made a mistake...

hedgepiggy, 2020-03-26:
> Also, you seem to believe that my time is worth as much as the price of the application you are using (i.e. 0), and therefore that I should be able to drop all the other stuff I'm doing

Along these lines, couldn't I make the argument that my time is worth too much to go around screwing up my laptop again and getting to use onscreen keyboard navigation to fix the problem? Given that you're the developer with the most experience with others' issues, _I acknowledge that I probably did click through the warning_. But that was the point that I think everyone in this thread (note that that thumbs up wasn't from either one of us...) was trying to make, it's not a clear warning because it's written in the most useless, uninformative way possible. You might as well replace that warning with the body text "YUR DOING THINGGSSS". To me, this is the equivalent of running diskpart to format a drive on windows and getting the prompt "You're modifying a disk, are you _sure_ you want to do this"? Also, I honestly find it hard to believe that there aren't a ton more users getting absolutely destroyed than eventually figuring out the problem and but just bother going on github to find some libwdi respository instead of something titled "Zadig". If you're developing for Windows, and you offer those users compiled binaries, how many of them are really gonna know or care about the Github repository where the source code is located on when they're busy trying to fix their input drivers? Come on.

I actually used Rufus for a while, and never had any issues with the UI (it was descriptive and provided appropriate warnings when things were getting changed), so I don't get why this Zadig's is so bad and why you're so touchy about trying to make things more descriptive. Whatever.

Thanks for the time that you spent developing Rufus, it was really helpful to me.

Screw this project though, at least from the UI side of things. "It's meant for developers or people that know what they are doing" is clearly not a good excuse because you wind up with so many people that don't know what they're doing (I didn't open this issue, and this isn't the only issue of the nature, and you see tired of people complaining, so I'm sure there's been plenty) that somehow get involved with running the project.

Granted, I wouldn't be surprised if fixing it would be a waste of your time since you can probably do incur more impact on other projects like Rufus anyways.

pbatard, 2020-03-26:
> _I acknowledge that I probably did click through the warning_

Thank you. I mean it.

And yes, you pretty much figured out why Zadig's UI is a lot more clunky than Rufus:
1. It was never developed to be used by everybody and their mothers. libwdi (the parent project of Zadig) originated as a very specific side project of libusb, to help __application developers__ integrate the installation of the driver they needed into their application, and the idea what that software applications that needed a driver installed would integrate libwdi and provide their own specific UI, while also making sure that the driver would only be installed for a specific set of devices.
2. Of course, a library alone doesn't get you far, so I decided to create a sample "generic" application, with a GUI, that demonstrated how one could go about doing the above. That's the reason why Zadig is found in the libwdi samples, and not a standalone project. And also why it's far from being as polished as can be, especially compared to Rufus, which was designed from the get go to be (relatively) user friendly.

At this stage two things happened:

1. Providers of software that requires driver installation decided that they didn't need to sort out their own driver installation and could just point to Zadig, often leaving __unfamiliar users__ have to figure out what the heck it was all about.
2. I got busy with other stuff (mostly Rufus, but not only), and even as Zadig became a lot more popular than I'd liked, its overall download count did not justify the level of investment that would be required to make it the kind of user-friendly application that people unfamiliar with drivers and USB could use.

For instance, the UI redesign that happened between Rufus 2.x and 3.x is something that took me about 3 months of full time work, which most people find surprising when they look at how minimal the changes seem to be. So, yeah, making something more digesteable to a bunch of users with completely different views of what "intuitive" should be is not just about adding an extra red dialog with big letters here and there.

Long story short, __properly__ redesigning an UI is not the 5 mins job that non developers seem to think it should boil down to, and for a free application that actually sees low usage, and that, from the very beginning, was actually designed to help its users shoot themselves in the foot if that's what they wanted (by replacing a system driver, because there are circumstances where you do want to do that) and not be over obnoxious about it for its targeted audience of developers and power users, something that's hard to justify investing resources in.

That's not to say that I can't understand your frustration, but I'm afraid nothing in life is cost-less, especially with free applications. Microsoft, Google, Apple and others can afford to gingerly spend their resources on UI redesign but I simply don't have that luxury because properly doing it __genuinely__ takes a lot more time than adding a dialog or changing a message here and there.

hedgepiggy, 2020-04-01:
I was able to reproduce the issue, and I know exactly what happened, and let me preface this with my sincere apology, I am a real fool.

What actually happened, was I had Zadig OPEN while I plugged in the device that I needed the driver on, but I also had VirtualBox running a VM in the background. I stupidly set a catch all filter for USB devices since I wasn't able to capture the device I needed to, and what happened is that for some reason _VirtualBox_ has a bug where it crashed my keyboard driver. Naturally, expecting it to be your application that crashed it, I googled the result and found it.

I'm really sorry for misplacing blame on your application, and now I think it makes sense why I never got a confirmation message and things seemed like they were broken - I didn't get one, and things WERE broken, I just made a stupid mistake and broke them myself (along with Oracle's bad software) while your poor application took collateral damage.

Thank you again for working tirelessly on open source projects. Hopefully you don't get too bogged down by fools like me. Best wishes!

pbatard, 2020-04-01:
Not many people are willing to admit when they make a mistake on the internet, which is a shame really because everybody makes mistakes and, as long as you are honest with them, I found that nobody will ever judge you negatively for it. On the contrary, because reporting one's mistakes actually help __others__. So I can only commend you for following up and reporting on your initial error. Thank you!

And I will readily admit that I made a mistake too in assuming that you had seen the extra Zadig warning and ignored it, because, clearly, even the problem was completely orthogonal to Zadig, you certainly did not get that extra popup. So it wasn't a case of _"user just dismissed a dialog a bit too casually"_, as I wrongly thought it was.

Your report about this VM mishap certainly gives me pause with regards to not striving to keep an open mind with regards to scenarios where I think I have a pretty good idea what "must" have happened, because even if your conclusions about Zadig being responsible for the issue weren't correct, your report that you did not see the extra dialog certainly was. And as you know, some of the points you made about the clunkiness of the UI are very valid too.

So you really have my thanks for following up on this, as, even as one has to relate on their own mistakes in it, this is the kind of follow up that is most helpful, as it may both help others running into a similar situation and help me realize that, even if I can't fathom how this might be possible, I shouldn't be too quick in dismissing the idea that something that a user say didn't happen, but that I can't see not happening ever, might genuinely not have happened...

shizoor, 2020-09-16:
I'd just like to add that if you've done this, then panicked, like I did and relentlessly used Zadig to try to rectify the issue, cycling through all available drivers to see if one would work, then use device manager to remove the mouse.   In my case it was a bit difficult to find the mouse as Windows was recognising it, failing to communicate with it, dropping it and finding it again which meant that expanding the menu was difficult.  I found USB mouse, under USB devices, instead of under "Mice and other pointing devices", I uninstalled the device, checking delete driver for this device.   Then it appeared as a WLAN adapter on COM7 under LibUSB devices, or similar.   Same routine, uninstall and delete driver, repeatedly until "Windows is setting up your USB mouse", and it worked.  

Note that the fix supplied did not work for me as once the damage has been done, the mouse no longer appeared under Mice and other pointing devices.   Driver Talent reported a normal mouse that was appearing and disappearing and didn't suggest any action, and wanted payment for any driver updates as well.   

I did try some other things, I went into c:\Windows\system32\drivers and deleted the relevant sys files, no avail, but I'm sure I didn't do it completely.  (If you're reading this and you've got the problem stick to device manager, this is very dangerous).  I also deleted some USB related keys from the registry under HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Enum\USB\  having backed up first.   Again if you're reading this because it's happened, don't do that because the registry failed to restore properly afterwards, no lasting damage in my case but I cannot speak for your system, my system is back up and running with mouse though.   

Also, if you own a Hewlett Packard, downloading the driver packs again and running them didn't help in my case either.  

I don't really know why I ignored the warning, it was only after I'd continued that I looked back and USB Mouse was selected, not  the SDR.   I've just run Zadig now and removed the SDR while Zadig was running without doing anything and Zadig will roll the menu option to the next device away from Bulk In 0 if the target device is unplugged while running Zadig, something to be wary of if your USB ports are worn.  Possible that I gave the desk a knock.  

Finally, thanks for writing Zadig!   Nothing else does what Zadig does and it's great that I can flash FPGAs and listen to radio stations on my laptop.  Thank you!

shizoor, 2020-09-16:
PS - I hereby submit myself to the Zadig hall of shame.

njmnbb, 2022-02-15:
> FWIW, there is a simple, really simple way of restoring device functionality after this gets mangled by erroneous operations.
> 
> Open 'Device Manager' enable 'Show hidden devices' under 'View' and remove any and all greyedout Mouse and USB devices, specifically 'unknown ...'
> 
> I guess we don't know all we need to know all the time don't we.
> 
> Here is a very helpful page regarding, this may be of use to add to the FAQ for zadig for referral.
> 
> https://www.drivethelife.com/windows-10/fix-hid-compliant-mouse-not-working-on-windows-10.html

This solved my issue with overwriting the driver of my Razer Chroma mouse mat. Thanks a ton!

xan1242, 2023-08-13:
FYI - I managed to do this too.

By having the device unplug itself while the installation was running. This extra dialog box did not appear and the selected device changed mid installation.

The registry is now corrupted and that device cannot install any other driver anymore, Windows helpfully trying to translate its error codes into a message also doesn't help matters (FYI - it just says "Access is denied").

This _really_ should be taken a look at again. Whatever this is doing to set its driver up, it's doing it very, very wrong.

over-fitted, 2023-08-23:
Adding myself to the wall of shame - overwrote my CH552G device's HID firmware and had it stuck as libusb-win32. Fixed it in the end though, anyone with the same issue can try the following:

1) go to device manager
2) right click on CH552x-<whatever the suffix is>
3) click update driver
4) Browse my computer for drivers
5) let me pick from a list of available drivers...
6) USB input device
7) next

Good luck

pushnextbutton, 2023-09-07:
OMG finally a fix now dam keyboard is working again thanks guys now for my next curiosity