pbatard/libwdi #121: [help request] Build wdi-simple in Docker, for webusb devices only

karel-3d, 2018-03-26:
Hello.

I am asking here because I am a little lost in the complexity of this project.

I want to build wcid-simple - simple instalator of WinUSB driver for one specific device - in a reproducible and easy way, that is, from docker. (I think this will be useful for more people, so that is why I am asking publicly here.)

I need specifically

* Driver for one concrete device with vid/pid combination
* The device is WinUSB device
* I want the resulting command-line installator to work on as many windows as possible - including 32bit versions - but if it isn't possible, then win7+
* I want the build of everything to be in docker (so cross-compiled on Linux)

If I understand correctly, we need to
* install mingw64 in the docker
* ... maybe add some DLL files (this is where I am lost. Do I need to copy the WinUSB DLL driver from somewhere to somewhere? Is that what is included in MS WDK? Or is it present in Windows7 by default and it will be used from there? Can I put that file in a github repo?)
* compile with ./configure and make

Is that all?

karel-3d, 2018-03-26:
This will mostly be needed for win7 users - a lot of users have Windows Update disabled and the automatic WCID installation fails - but some percentage of win10 users still have problems that are currently fixed with zadig.

karel-3d, 2018-03-26:
So far, I got stuck on the WDK/DDK installation.

The DDK ISO file linked in some of the wiki pages seems to be outdated and does not seem to work with the autoconf.

The new WDK seems to be un-installable in docker. I will probably need to install it in Windows and copy the files over?

pbatard, 2018-03-26:
Please see [what we do for Appveyor builds](https://github.com/pbatard/libwdi/blob/master/appveyor.yml). It should be fairly similar to what you want to achieve in Docker (PS: why don't you just use Appveyor since it's already there? All you should need is to for the github project). And yes, you need to provide the WinUSB redistributables from the WDK.

Now, I don't know anything about Docker, so that's about as far as I'm going to help you. I'll just mention however that is is __VERY BAD PRACTICE__ to provide applications that need to run elevated on Windows and that aren't digitally signed (which is not something you can automate, as it would mean sharing your private signing key), as all it does is __diminish__ the good security practice reflexes Windows users should have. Even hinting that it is okay to run an unsigned elevated application, just because it was built through Docker or Appveyor, is asking for trouble. But then, this is just my advice.

karel-3d, 2018-03-26:
Thanks.

> Why don't you just use Appveyor since it's already there?
That is just because I have no experience with AppVeyor. I will try it.

> I'll just mention however that is is VERY BAD PRACTICE to provide applications that need to run elevated on Windows and that aren't digitally signed (which is not something you can automate, as it [...]

Thanks for the tip. Signing can be semi-automated at least. (That is, sign only if the private key is present, which won't be in the repo.)

I am looking at the appveyor project. Can I directly use the wdk10-redist.7z file from files.akeo.ie, as you are using? (What is the license, do I need to first agree to the WDK10 EULA?)

pbatard, 2018-03-26:
> Can I directly use the wdk10-redist.7z file from files.akeo.ie

As long as you're not planning to run hundreds of builds each day, and put a strain on my server, that's fine by me 😃 

> What is the license

These files are redistributable. They come from `C:\Program Files (x86)\Windows Kits\10\Redist\wdf\` when you install the WDK and you should be able to access the license (`license.rtf`) if you have WDK installed.

pbatard, 2018-03-26:
No idea if that still applies. You'll have to try by yourself.

karel-3d, 2018-03-27:
This is working for me

```Dockerfile
FROM debian:stretch

RUN apt-get update && \
  apt-get install -y 
  autoconf \
  autotools-dev \
  binutils-mingw-w64-i686 \
  build-essential \
  dos2unix \
  g++-mingw-w64-i686 \
  gcc-multilib \
  git \
  libtool \
  mingw-w64-common \
  p7zip-full
RUN git clone https://github.com/pbatard/libwdi.git
RUN mkdir /wdk
ADD wdk10-redist.7z /wdk/wdk.7z
RUN cd /wdk/ && 7z x wdk.7z
RUN apt-get install -y 
WORKDIR /libwdi 

RUN bash ./autogen.sh --host=i686-w64-mingw32 --with-wdkdir=/wdk/wdk/ --with-libusb0="" --with-libusbk=""

#add missing COINSTALLER_DIR because of some bug in m4 with AC_CHECK_FILES and cross-compilation
RUN echo '#define COINSTALLER_DIR "wdf"' >> config.h

#patching ancient mingw in debian
ADD setupapi.patch .
RUN patch /usr/share/mingw-w64/include/setupapi.h < setupapi.patch

# actual make
RUN cd libwdi && make all-local
RUN cd examples && make wdi-simple.exe
```

where setupapi.patch looks like this
```diff
--- setupapi.h  2018-03-26 20:58:51.412792188 +0000
+++ /usr/i686-w64-mingw32/include/setupapi.h    2018-03-26 20:57:12.450203083 +0000
@@ -7,6 +7,7 @@
 #define _INC_SETUPAPI
 
 #include <_mingw_unicode.h>
+#include <devpropdef.h>
 
 #ifndef _SETUPAPI_
 #define WINSETUPAPI DECLSPEC_IMPORT
@@ -1875,6 +1876,9 @@
   WINSETUPAPI WINBOOL WINAPI SetupConfigureWmiFromInfSectionA(HINF InfHandle,PCSTR SectionName,DWORD Flags);
   WINSETUPAPI WINBOOL WINAPI SetupConfigureWmiFromInfSectionW(HINF InfHandle,PCWSTR SectionName,DWORD Flags);
 
+  WINSETUPAPI WINBOOL WINAPI SetupDiGetDevicePropertyW(HDEVINFO DeviceInfoSet, PSP_DEVINFO_DATA DeviceInfoData, const DEVPROPKEY *PropertyKey, DEVPROPTYPE *PropertyType, PBYTE PropertyBuffer, DWORD PropertyBufferSize, PDWORD RequiredSize, DWORD Flags);
+  WINSETUPAPI WINBOOL WINAPI SetupDiGetDevicePropertyKeys(HDEVINFO DeviceInfoSet, PSP_DEVINFO_DATA DeviceInfoData, DEVPROPKEY *PropertyKeyArray, DWORD PropertyKeyCount, PDWORD RequiredPropertyKeyCount, DWORD Flags);
+
 #ifdef __cplusplus
 }
 #endif
```

karel-3d, 2018-03-27:
Closing this, I guess :)

(This produces 32b binary, 64b is similar)

closed 2018-03-27 by karel-3d

karelbilek, 2024-06-07:
> 64b sample of above:
>   binutils-mingw-w64-x86-64 \
>   g++-mingw-w64-x86-64 \
> RUN bash ./autogen.sh --host=x86_64-w64-mingw32 --disable-32bit --with-wdkdir=/wdk/wdk/ --with-libusb0="" --with-libusbk=""
> See also
> For what it's worth, I'm compiling on debian:sid-20230612 without any
> patches:
> FROM debian:sid-20230612 AS builder
> COPY . /app
> WORKDIR /app
> RUN bash ./autogen.sh --host=x86_64-w64-mingw32 --disable-32bit --with-wdkdir="" --with-libusb0="" --with-libusbk="" --with-userdir="/app/userdir"

Note that trezor (where I don't work anymore) is not using libwdi anymore
as they have decided to deprecate support for win7, and libwdi was used
mainly to install winusb devides on win7.

so this dockerfile was later removed from repo.