pbatard/libwdi #123: How to replace driver of existing, but unplugged device?

sqrt, 2018-03-26:
We'd like to replace the driver for a device that is already installed with a different driver, but it may not be present at the time of replacement.

This does not work using the "Create New Device" option, since the original driver will be loaded when the device gets attached.

It does work when manually removing the device from device manager, but we'd like to do it without (manually) having to remove the device.

Is there any way to do that? Thanks!

pbatard, 2018-03-26:
Sorry, but that's not something libwdi supports. You should ask this question on more appropriate forums such as Windows driver development forums.

closed 2018-03-26 by pbatard

karel-3d, 2018-03-27:
@sqrt try to read about devcon.exe

For example

`devcon.exe remove =media "*VID_045e&PID_02a0*"`