pbatard/libwdi #125: wdi-simple exe download

alexdresko, 2018-05-09:
I don't have all of the prerequisites installed to compile wdi-simple, and it would be really helpful if someone could simply provide a compiled version of the application. I need to automate the work being done by zadig, and I believe that's what wdi-simple does, right?

pbatard, 2018-05-09:
Sorry, but I'm not planning to compile and upload `wdi-simple.exe` anywhere. It's painful enough to upload Zadig whenever there's a major libwdi update, and I'd rather not add extra work for wdi-simple. What's more, the people who use `wdi-simple.exe` tend to want only part of the drivers there, or want a 64-bit version ans do on. If I start to give in on uploading one additional executable, I'm pretty sure it will just open the floor to more custom upload requests.

My best advice is, if you don't want to set up your own dev env, figure our how to build `wdi-simple.exe` in your own github clone of libwdi with AppVeyor (see https://github.com/pbatard/libwdi/blob/master/appveyor.yml). Then you will be able to download the `wdi-simple.exe` artefact straight from AppVeyor.

Or you can also see the [relevant issues](https://github.com/pbatard/libwdi/issues/121) and [pull requests](https://github.com/pbatard/libwdi/pull/68).

Note, the reason why we don't do that in the main project is that we don't want to officially make available an executable that is not digitally signed (which cannot be automated through AppVeyor on account that we would have to share the private key), as this creates a huge security issue by giving users the expectation that it's okay to share unsigned elevated apps.

closed 2018-05-09 by pbatard

alexdresko, 2018-05-09:
Thanks for the quick, clear explanation. It makes perfect sense and I have no complaints. 

Building on appveyor is a good idea. I'll give that a go and report back for posterity.

andrewleech, 2023-05-28:
I see now that both `zadig.exe` and `wdi-simple.exe` are built in github actions and can be downloaded in the `VS2022` Artifacts eg. 
* https://github.com/pbatard/libwdi/actions/runs/4306435296 
* https://github.com/pbatard/libwdi/suites/11286925065/artifacts/578976155

These exe's aren't signed which is understandable, but I'd also love it if the default wdi_simple was also published for use in automated installers!